oliexdev/openScale #262: Issues, for new Install and User

PetRose, 2018-05-09:
Hi
I wanted my Beurer BF700 scale data to be picked up by openScale.
Once installed, I cannot see my scale anywhere - don't I need to identify it - like any other BT device ?
And, my scale I used shortnames, ie Initials of two letters, so I assume the reason for not seeing my data, could be that I cannot add a UserName in openScale with less than 3 letters ! Hence I cannot pick up my measurements !
See attached screenshot, when I click the BT symbol in openScale
[image]

erijo, 2018-05-09:
You need to go to settings, Bluetooth, device type to find your scale.

PetRose, 2018-05-11:
OK, Tried to do that - but the search for device did not find my BF700 scale.
Tried to search using builtin Android BT search, and it recognizes it as BF700 and can pair with it, even using the SmartConnect.
Tried to stand on the scale while it was active, but only quite few devices pops up in the Bluetooth search after some minutes, but not the BF700 scale.
Any thoughts ?

oliexdev, 2018-05-11:
Try to unpair the device with the other app restart your smartphone and try again in openScale. If you device is already paired to something different it can't be seen.

Please write also down the Bluetooth device name of your scale.

PetRose, 2018-05-12:
That did work !
Removed the Android pairing, Rebooted Android, and used openScale to search for device, and it responded rather quick. 
However, my previous scale measurements (on the scale) was not recoqnized, even my weight matches only for me (+/- 2 kg) using the scale.
I probably reset the scale, and re-define users on it, to match the openScale username (truncated to 3 chars), and have it a go.
And btw, the MAC address of my BF700 scale is: 00:13:04:19:40:7E if that can be of any use.
Thanks for the guidance

closed 2018-05-12 by PetRose