oliexdev/openScale #266: Add support for an activity level

oliexdev, 2018-05-13:
Add an user option to support an activity level (e.g. 0-none; 1-average: 2-high). With this option we can  refine the BMR calculation.

erijo, 2018-05-13:
Are three levels enough? I think I've seen that some have five (but I might be mistaken).

oliexdev, 2018-05-13:
yeah some have five but some have three. No matter what we decide we have to convert some activity level ranges.

erijo, 2018-05-15:
[This formula](http://www.globalrph.com/harris-benedict-equation.htm) has five levels. There are also a couple of other equations listed there that seem to use 4-5 levels.

oliexdev, 2018-05-19:
added activity level option with commit https://github.com/oliexdev/openScale/commit/6cbdfbbe05bc6e0cfc0d4d503961024e679e7dfc

closed 2018-05-19 by oliexdev

erijo, 2018-05-19:
I think it would be easier to understand if the user activity had the levels listed by name in a spinner.

oliexdev, 2018-05-19:
Yeah if you want you can change it to a spinner.