oliexdev/openScale #267: Add support to take a picture on a measurement

oliexdev, 2018-05-13:
Add support to take a picture of yourself on a measurement. I don't know if it useful or not!? One user asked this feature already...Feedbacks are welcome!
If yes, is it better if we store the picture or is it better to store the file links in the SQL database. The advantage is the easy backup process the [disadvantage](https://stackoverflow.com/questions/50124747/is-it-a-good-practice-to-use-room-to-store-links-to-binary-files-alongside-with) is that the database size can be very large.

erijo, 2018-05-19:
I think it would complicate the interface with little gain. Isn't it better to store the pictures using a software made for that? One that can generate index, thumbnails etc.

oliexdev, 2018-05-19:
I think the interface won't be the problem but the backup of the whole images if we don't store the images in the database.

Anyway, currently I also don't see a huge advantage of this feature.
I will close this issue. Maybe in the future we can implement it or somebody create a PR with a example implementation.

closed 2018-05-19 by oliexdev