oliexdev/openScale #268: Support for data other than weight from Xiaomi Mi Scale 2

Pastafarianist, 2018-05-17:
Currently measurements from Xiaomi Mi Scale 2 only contain weight data. However, the official app (Mi Fit) provides a lot more information, including body water, muscle, body fat, bone mass, and visceral fat. It would be great to have all these measurements in openScale too.

Pastafarianist, 2018-05-17:
Sorry, I've just noticed [here](https://github.com/oliexdev/openScale/wiki/Supported-scales-in-openScale) that the lack of support is due to "not transparent impedance calculation by the company". Are there any additional details on this?

erijo, 2018-05-17:
I think the Mi app uses the same native library as in #159 to convert impedance measurement to fat, water etc. You can see in that other issue how far we've come with the reverse engineering.

closed 2018-05-19 by oliexdev

Pastafarianist, 2018-08-29:
Perhaps this should have been tagged as a feature request and not a question (and not have been closed). Any chance this could be resolved when #159 is done?

oliexdev, 2018-08-29:
@Pastafarianist I should had marked as duplicated because all information was available on the project page at the time you have asked. We are aware of that problem, blame the company for the intransparent calculation.
And @erijo already answered your question. It could be possibly resolved when issue #159 is resolved.