oliexdev/openScale #269: Support for Medisana BS 440

YoSiJo, 2018-05-18:
I tried to follow the instructions here (https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale), but I could not find my wipe at point 3.2. I provided you with the data and hope you can continue to do so.

Logs and Files: https://nc.waimanu.io/s/Z8R4fZDNYTtZQa6

erijo, 2018-05-18:
I just made openScale treat BS440 in the same way as BS444. It looked similar enough. Could you please try the [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) and see if it works? Please enable debug logging under settings, about and attach the log here.

YoSiJo, 2018-05-18:
There is no end to the scanning process and the device is still not found.

```
2018-05-18 22:44:55.087 Debug AboutPreferences: Debug log enabled, openScale (dev) v1.7.3-dev_6ca941e3 (1526673840), SDK 22, nubia NX512J
2018-05-18 22:45:24.234 Debug BluetoothPreferences: Start discovery
2018-05-18 22:46:12.793 Debug BluetoothPreferences: Stop discovery
```

erijo, 2018-05-19:
Please use the latest dev version when testing. I added some more logging.

YoSiJo, 2018-05-19:
I started two attempts and the Bluetooth snooping was off, but still no success.

```
2018-05-19 08:37:17.531 Debug AboutPreferences: Debug log enabled, openScale (dev) v1.7.3-dev_1652a37e (1526706865), SDK 22, nubia NX512J
2018-05-19 08:37:54.733 Debug BluetoothPreferences: Start discovery
2018-05-19 08:38:55.599 Debug BluetoothPreferences: Stop discovery
2018-05-19 08:38:59.946 Debug BluetoothPreferences: Start discovery
2018-05-19 08:39:01.286 Debug BluetoothPreferences: Stop discovery
2018-05-19 08:53:45.350 Debug BluetoothPreferences: Start discovery
2018-05-19 08:54:06.744 Debug BluetoothPreferences: Stop discovery
2018-05-19 08:54:07.081 Debug BluetoothPreferences: Start discovery
2018-05-19 08:54:12.885 Debug BluetoothPreferences: Stop discovery
2018-05-19 08:54:13.079 Debug BluetoothPreferences: Start discovery
2018-05-19 08:54:31.439 Debug BluetoothPreferences: Stop discovery
```
Ps. He seems to restart the discover when I rotate the screen.

YoSiJo, 2018-05-19:
Restart always works wonders, here's the data after the reboot, the balance was found, but it didn't seem to be able to pull the data.

```
2018-05-19 09:03:20.723 Debug AboutPreferences: Debug log enabled, openScale (dev) v1.7.3-dev_1652a37e (1526706865), SDK 22, nubia NX512J
2018-05-19 09:03:48.618 Debug BluetoothPreferences: Start discovery
2018-05-19 09:03:48.800 Debug BluetoothPreferences: Discovery: android.bluetooth.adapter.action.DISCOVERY_STARTED
2018-05-19 09:03:49.985 Debug BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-05-19 09:03:50.011 Debug BluetoothPreferences: Found supported device 0202B6101AFE149CFE [FE:9C:14:FE:1A:10] (driver: Medisana BS44x, type: 2)
2018-05-19 09:03:52.221 Debug BluetoothPreferences: Stop discovery
2018-05-19 09:03:52.327 Debug OpenScale: Trying to connect to bluetooth device [FE:9C:14:FE:1A:10] in debug mode
2018-05-19 09:03:52.331 Info BluetoothCommunication: Connecting to [FE:9C:14:FE:1A:10] (driver: Debug)
2018-05-19 09:03:52.337 Debug BluetoothCommunication: Starting LE scan
2018-05-19 09:03:52.404 Debug OpenScale: Disconnecting from bluetooth device
2018-05-19 09:03:52.411 Info BluetoothCommunication: Disconnecting (with cleanup)
```
He got stuck with the message with fetch data.

erijo, 2018-05-19:
What happens if you try to connect from the main screen (by clicking the Bluetooth icon)?

YoSiJo, 2018-05-19:
No, I went through the settings and selected Scale in the Bluetooth section.

erijo, 2018-05-19:
Yes, I guessed that. But to get readings you need to click the Bluetooth icon in the main view.

YoSiJo, 2018-05-19:
```
Build version: 1.7.3-dev_1652a37e 
Current date: 2018-05-19 13:38:43 
Device: Nubia NX512J 
 
Stack trace:  
java.util.UnknownFormatConversionException: Conversion: 2.
	at java.util.Formatter$FormatSpecifierParser.unknownFormatConversionException(Formatter.java:2321)
	at java.util.Formatter$FormatSpecifierParser.parsePrecision(Formatter.java:2389)
	at java.util.Formatter$FormatSpecifierParser.parseWidth(Formatter.java:2375)
	at java.util.Formatter$FormatSpecifierParser.parseArgumentIndexAndFlags(Formatter.java:2345)
	at java.util.Formatter$FormatSpecifierParser.parseFormatToken(Formatter.java:2298)
	at java.util.Formatter.doFormat(Formatter.java:1071)
	at java.util.Formatter.format(Formatter.java:1042)
	at java.util.Formatter.format(Formatter.java:1011)
	at java.lang.String.format(String.java:1803)
	at java.lang.String.format(String.java:1777)
	at com.health.openscale.core.datatypes.ScaleMeasurement.toString(ScaleMeasurement.java:423)
	at java.lang.StringBuilder.append(StringBuilder.java:202)
	at android.os.Message.toString(Message.java:509)
	at android.os.Looper.loop(Looper.java:154)
	at android.app.ActivityThread.main(ActivityThread.java:5463)
	at java.lang.reflect.Method.invoke(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:372)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:909)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:704)

User actions: 
2018-05-19 13:38:15: MainActivity resumed
2018-05-19 13:38:16: SettingsActivity destroyed
```

YoSiJo, 2018-05-19:
The data itself can be seen in the app, should I add the error log from the app?

oliexdev, 2018-05-19:
@YoSiJo use the latest dev version. The error should be fixed.

erijo, 2018-05-19:
Also please verify that all measurements are correct, including time.

Does the original app support user management?

oliexdev, 2018-05-19:
you have to reinstall the app since it is the development version we made some database changes.

YoSiJo, 2018-05-19:
All values were correct.
The following values Records my Scale, but do not appear in the app:
* Bone
* BMR
With the original app, user management is also carried out via the scale and not via the app.

erijo, 2018-05-19:
Have you enabled the measurements in settings, measurements?

What info do you enter on the scale and which values did you enter?

erijo, 2018-05-19:
Once you have verified the bone measurement (BMR is calculated by openScale and may differ compared to what the scale reports due to using different formulas) I would like to get some more information regarding the problem you had after scanning.

The idea is that if you have the debug log enabled and click on your scale in the search dialog a popup dialog should appear informing you that we're trying to fetch some more info from the scale. Did you see this dialog? Can you try again?

YoSiJo, 2018-05-20:
Okay, I hadn't turned on the bone mass yet.
The scale only asked about age and size at that time (it's been over a year since I set it up.).
When scanning for scales, I see the scale listed. But no more dialogue. If I select the scale, I return to the Bluetooth settings and the scale is selected.
Screenshots: https://nc.waimanu.io/s/Z8R4fZDNYTtZQa6

erijo, 2018-05-20:
Did you enable debug logging? Otherwise the extra dialogue isn't shown.

YoSiJo, 2018-05-20:
Sorry, I forgot to log. The new screenshots are in NextCloud. It all worked out this time.
I assume it's been on my phone since the reboot.

erijo, 2018-05-20:
Nice! Could you please attach the log as well?

Finally, could you take a picture of the scale for [our gallery] 
(https://github.com/oliexdev/openScale)?/wiki/Supported-scales-in-openScale

YoSiJo, 2018-05-20:
Here is the log:

```
2018-05-20 19:51:32.228 Debug AboutPreferences: Debug log enabled, openScale (dev) v1.7.3-dev_e13119ae (1526758005), SDK 22, nubia NX512J
2018-05-20 19:51:48.430 Debug BluetoothPreferences: Start discovery
2018-05-20 19:51:48.640 Debug BluetoothPreferences: Discovery: android.bluetooth.adapter.action.DISCOVERY_STARTED
2018-05-20 19:51:49.920 Debug BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-05-20 19:51:49.943 Debug BluetoothPreferences: Found unsupported device [LG] webOS TV UJ6519 [61:D0:86:A6:0E:4D] (type: 2)
2018-05-20 19:51:56.948 Debug BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-05-20 19:51:56.975 Debug BluetoothPreferences: Found supported device 0202B6101AFE149CFE [FE:9C:14:FE:1A:10] (driver: Medisana BS44x, type: 2)
2018-05-20 19:51:58.002 Debug BluetoothPreferences: Stop discovery
2018-05-20 19:51:58.120 Debug OpenScale: Trying to connect to bluetooth device [FE:9C:14:FE:1A:10] in debug mode
2018-05-20 19:51:58.123 Info BluetoothCommunication: Connecting to [FE:9C:14:FE:1A:10] (driver: Debug)
2018-05-20 19:51:58.128 Debug BluetoothCommunication: Starting LE scan
2018-05-20 19:51:58.354 Debug BluetoothCommunication$: onConnectionStateChange: status=0, newState=2
2018-05-20 19:51:58.566 Debug BluetoothCommunication$: onServicesDiscovered: status=0
2018-05-20 19:51:59.575 Debug BluetoothCommunication: INIT STATE: 0
2018-05-20 19:51:59.713 Debug BluetoothCommunication$: onCharacteristicRead 00002a00-0000-1000-8000-00805f9b34fb (status=0): 30 32 30 32 42 36 31 30 31 41 46 45 31 34 39 43 46 45 
2018-05-20 19:51:59.714 Debug BluetoothCommunication: INIT STATE: 1
2018-05-20 19:51:59.810 Debug BluetoothCommunication$: onCharacteristicRead 00002a01-0000-1000-8000-00805f9b34fb (status=0): 90 03 
2018-05-20 19:51:59.812 Debug BluetoothCommunication: INIT STATE: 2
2018-05-20 19:52:00.006 Debug BluetoothCommunication$: onCharacteristicRead 00002a04-0000-1000-8000-00805f9b34fb (status=0): 06 00 20 00 00 00 FF FF 
2018-05-20 19:52:00.009 Debug BluetoothCommunication: INIT STATE: 3
2018-05-20 19:52:00.102 Debug BluetoothCommunication$: onCharacteristicRead 00002a25-0000-1000-8000-00805f9b34fb (status=0): 31 35 38 37 00 00 00 00 00 00 00 00 
2018-05-20 19:52:00.103 Debug BluetoothCommunication: INIT STATE: 4
2018-05-20 19:52:00.248 Debug BluetoothCommunication$: onDescriptorRead 00002904-0000-1000-8000-00805f9b34fb (status=0): 19 00 00 00 01 00 00 
2018-05-20 19:52:00.250 Debug BluetoothCommunication: INIT STATE: 5
2018-05-20 19:52:00.346 Debug BluetoothCommunication$: onCharacteristicRead 00002a27-0000-1000-8000-00805f9b34fb (status=0): 42 32 
2018-05-20 19:52:00.347 Debug BluetoothCommunication: INIT STATE: 6
2018-05-20 19:52:00.591 Debug BluetoothCommunication$: onDescriptorRead 00002904-0000-1000-8000-00805f9b34fb (status=0): 19 00 00 00 01 00 00 
2018-05-20 19:52:00.593 Debug BluetoothCommunication: INIT STATE: 7
2018-05-20 19:52:00.837 Debug BluetoothCommunication$: onCharacteristicRead 00002a26-0000-1000-8000-00805f9b34fb (status=0): 42 2E 38 
2018-05-20 19:52:00.839 Debug BluetoothCommunication: INIT STATE: 8
2018-05-20 19:52:00.932 Debug BluetoothCommunication$: onDescriptorRead 00002904-0000-1000-8000-00805f9b34fb (status=0): 19 00 00 00 01 00 00 
2018-05-20 19:52:00.934 Debug BluetoothCommunication: INIT STATE: 9
2018-05-20 19:52:01.128 Debug BluetoothCommunication$: onCharacteristicRead 00002a29-0000-1000-8000-00805f9b34fb (status=0): 54 72 61 6E 73 74 65 6B 00 00 00 00 00 00 00 00 00 00 00 00 
2018-05-20 19:52:01.129 Debug BluetoothCommunication: INIT STATE: 10
2018-05-20 19:52:01.420 Debug BluetoothCommunication$: onDescriptorRead 00002904-0000-1000-8000-00805f9b34fb (status=0): 19 00 00 00 01 00 00 
2018-05-20 19:52:01.421 Debug BluetoothCommunication: INIT STATE: 11
2018-05-20 19:52:01.567 Debug BluetoothCommunication$: onCharacteristicRead 00002a28-0000-1000-8000-00805f9b34fb (status=0): 42 4C 45 5F 42 46 
2018-05-20 19:52:01.569 Debug BluetoothCommunication: INIT STATE: 12
2018-05-20 19:52:01.664 Debug BluetoothCommunication$: onDescriptorRead 00002904-0000-1000-8000-00805f9b34fb (status=0): 19 00 00 00 01 00 00 
2018-05-20 19:52:01.665 Debug BluetoothCommunication: INIT STATE: 13
2018-05-20 19:52:01.809 Debug BluetoothCommunication$: onDescriptorRead 00002902-0000-1000-8000-00805f9b34fb (status=0): 00 00 
2018-05-20 19:52:01.814 Debug BluetoothCommunication: INIT STATE: 14
2018-05-20 19:52:01.907 Debug BluetoothCommunication$: onDescriptorRead 00002902-0000-1000-8000-00805f9b34fb (status=0): 00 00 
2018-05-20 19:52:01.908 Debug BluetoothCommunication: INIT STATE: 15
2018-05-20 19:52:02.103 Debug BluetoothCommunication$: onCharacteristicRead 00008a20-0000-1000-8000-00805f9b34fb (status=0): 37 FB 
2018-05-20 19:52:02.105 Debug BluetoothCommunication: INIT STATE: 16
2018-05-20 19:52:02.248 Debug BluetoothCommunication$: onDescriptorRead 00002902-0000-1000-8000-00805f9b34fb (status=0): 00 00 
2018-05-20 19:52:02.250 Debug BluetoothCommunication: INIT STATE: 17
2018-05-20 19:52:02.252 Debug BluetoothCommunication: CMD STATE: 0
2018-05-20 19:52:02.253 Debug BluetoothDebug: Service 00001800-0000-1000-8000-00805f9b34fb
2018-05-20 19:52:02.255 Debug BluetoothDebug: |- characteristic 00002a00-0000-1000-8000-00805f9b34fb (instance 0): WRITE, WRITE_NO_RESPONSE, READ (permissions=0x0)
[... cut ...]
2018-05-20 19:52:15.148 Debug BluetoothCommunication$: onConnectionStateChange: status=0, newState=2
2018-05-20 19:52:15.362 Debug BluetoothCommunication$: onConnectionStateChange: status=62, newState=0
2018-05-20 19:52:15.364 Debug OpenScale: Disconnecting from bluetooth device
2018-005-20 19:52:15.364 Info BluetoothCommunication: Disconnecting
2018-05-20 19:52:38.997 Debug AboutPreferences: Debug log disabled
```

Pictures from scale do you found here: https://nc.waimanu.io/s/Z8R4fZDNYTtZQa6
I hope it's ok, because I put the pictures under GNU AGPL v3 license.

erijo, 2018-05-20:
Thanks for all help @YoSiJo! BS440 will now be supported in the next release.

closed 2018-05-20 by erijo