oliexdev/openScale #270: Issues editing measurements

Disgustus, 2018-05-22:
Running v1.8(27) when manually editing the weight in an entry, after clicking save, and returning to the list it's been reset to what it originally was. Is this working as intended?

erijo, 2018-05-22:
The problem seems to be that the weight view no longer saves the converted value. I can fix it later tonight unless @oliexdev beats me to it.

I've found another issue with navigating between measurements that seems to include disabled measurements that I would like to fix as well.

oliexdev, 2018-05-22:
@Disgustus thanks for the hint. That was my fault :(
fixed with commit https://github.com/oliexdev/openScale/commit/0479c31d6f442cadab3a81cbbadc38be76d8b1c4

I have stopped the introduction of version 1.8. I will wait @erijo if you have fixed the other bug then I am gonna publish v1.8.1

Maybe we should add more Use-Case tests... openScale getting step by step more complex.

erijo, 2018-05-22:
Did a quick fix @oliexdev. Please go ahead with 1.8.1.

oliexdev, 2018-05-22:
done v1.8.1 is on air ;)