oliexdev/openScale #272: Incorrect display/calculation of goal weight and weight difference in statistics screen v1.8.1

Forger125, 2018-05-23:
In version 1.8.x (1.8 and 1.8.1), the goal weight and weight difference are not displayed correctly in the Statistics tab. The first problem is that when using the Imperial system units (lbs, in), regardless of what is entered for user's starting weight and goal weight, the goal weight does not match what is entered in the user profile. This happens regardless of other user profile information (gender, height, etc.). See example below:

Data from user profile:
Initial weight: 200.0 lb
Goal weight: 175.0 lb

Data displayed in statistics tab (no additional measurements)
Goal weight: 79.4 lb
Weight difference: 79.4 lb

Data displayed in statistics tab (measurement at 198 lb added)
Goal weight: 79.4 lb
Weight difference: 118.6 lb

It correctly calculates goal weight but does so with the incorrect goal weight. After putting in a couple different values for goal weights, it looks like what may be happening is that the values may be getting converted to SI units (kg) while still retaining the English label (lbs), i.e. 175.0 lb = 79.4 kg.

The second (minor) issue is that regardless of which unit system is used, the weight difference is not correctly calculated when there are not any measurements entered, i.e. immediately after the user is created but before the first set of measurements is entered. In other words, the app currently displays **weight difference = goal weight** (see above), but instead it should be calculated it as **weight difference = initial weight (from user profile) - goal weight**. Then when a users has entered their first weight value in the app, the formula would change to **weight difference = most recent measured weight - goal weight**.

Also, as an aside, thank you for all the work you are doing on this app - it is a great resource and a very helpful contribution to the open source ecosystem on Android! :) keep up the great work!

erijo, 2018-05-23:
Should be fixed now. You can verify it by testing the [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build).

closed 2018-05-23 by erijo

Forger125, 2018-05-23:
@erijo Just installed Dev APK, both issues are resolved - thank you for the quick turnaround!

commit 7875bf4c (2018-05-23): Don't need to convert goal weight as it is already in kg

Fix #272.