oliexdev/openScale #278: Bluetooth unstable for Medisana BS 444

hirnsushi, 2018-05-29:
I'm experiencing problems with the bluetooth connection to my Medisana BS 444 since upgrading to 1.8.
The connection only gets established if I readd the scale through settings. About 2/3 of the time that doesn't work either or the connection drops immediately. Once successfully connected the measurements get synced as always, but reconnecting the scale at a later point simply by pressing the bluetooth icon doesn't work.
I've had similar problems with 1.7.3 but it only happened every other week or so, the releases with hardcoded MAC-Adresses worked without problems.
The attached log is from unsuccessfully trying to establish a connection, then readding the scale twice until it finally worked.
[openScale_2018-05-28_12-52.txt](https://github.com/oliexdev/openScale/files/2047159/openScale_2018-05-28_12-52.txt)

erijo, 2018-05-29:
Could you please try the [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) and see if it works better?

hirnsushi, 2018-05-29:
The device not found error seems fixed now. I always get a connection but ~ 9 out of 10 times it's dropped immediately. Tried multiple times, sometimes it worked on the first or second try, sometimes it didn't before my scale shut off.
[openScale_2018-05-29_12-42.txt](https://github.com/oliexdev/openScale/files/2047894/openScale_2018-05-29_12-42.txt)

Disgustus, 2018-05-30:
I have the same issue, using the Dev build does seem to work. Will run it some more.

erijo, 2018-05-31:
Please try the latest dev version and report how it works. Please also include the debug log.

It would also be interesting to see a btsnoop log from the original app to see if/how it deletes measurements.

hirnsushi, 2018-06-01:
The latest dev version (2018-06-01 07:41:37 UTC) doesn't install for me. I'll wait for a newer CI-Build and report back to you.
I've never used the original app, I'm running microG without any Gapps, but I'll give it a try.

oliexdev, 2018-06-01:
@hirnsushi why you can't install the latest dev version?

hirnsushi, 2018-06-01:
I'm not sure, the Package Installer just throws a nondescript error ("Package wasn't installed") back at me. Hasn't happened with any of the dev versions before, so it might just be some problem on my side, will try on a different device later.

hirnsushi, 2018-06-01:
Reliability has gone up for me with the new version, I'd say 1 in 5 attempts successfully syncs data, that's absolutely good enough for me 😃
I did manage to hit a new bug where the bluetooth icon on the overview page disappeared until I restarted the app. That was before I enabled logging and I wasn't able to reproduce it though.
[openScale_2018-06-01_17-59.txt](https://github.com/oliexdev/openScale/files/2063367/openScale_2018-06-01_17-59.txt)

erijo, 2018-06-01:
I've seen the problem with the Bluetooth icon as well. It's still there and you can click it, it's just not visible.

Teyro, 2018-06-08:
@erijo 

It helps to add permission to storage inside the app settings

erijo, 2018-06-11:
@Teyro: in what way does that help?

erijo, 2018-06-11:
@hirnsushi: the log seems to indicate that openScale downloads a bunch of old measurements. If we knew how to delete them it might improve the connectivity.

hirnsushi, 2018-06-12:
Sorry for the delay, I almost forgot about that, here you go [btsnoop_hci.log](https://github.com/oliexdev/openScale/files/2092149/btsnoop_hci.log).
I hope there's something useful in there. Let me know if you need any more logs.

erijo, 2018-06-14:
It seems like the original app doesn't delete any measurements either.

Could you provide a btsnoop log when openScale tries to connect and one where it succeeds? I would like to compare with the one from the original app to see if I can figure out why the original app doesn't have the connectivity problems that openScale does.

hirnsushi, 2018-06-14:
Sure, unsuccessful sync [btsnoop_hci.log](https://github.com/oliexdev/openScale/files/2103702/btsnoop_hci.log), succesful sync (after a few tries, at the end) [btsnoop_hci.log](https://github.com/oliexdev/openScale/files/2103713/btsnoop_hci.log).

erijo, 2018-06-14:
Thanks @hirnsushi.

Please try the latest dev version and let me know how it works.

hirnsushi, 2018-06-14:
Syncs immediately every time I tried, perfect, great work! 👍

erijo, 2018-06-15:
Glad to hear!

I will leave this issue open for a while since I need to clean up the fix a bit (e.g. add a scan timeout).