oliexdev/openScale #280: Visceral fat and medisana bs444

Disgustus, 2018-05-29:
I noticed that visceral fat doesn't receive data when syncing with a medisana bs444. Is this a bug, or should visceral fat be entered manually?
If it's the latter, can anyone recommend me a way to measure this?

erijo, 2018-05-29:
As far as I can tell, bs444 doesn't support visceral fat measurement. Do you get that measurement in the original app?

Btw, do you have any problems connecting to the scale (as reported in #278)?

Disgustus, 2018-05-29:
Will download the original app later today to check.

Also, yeah, bluetooth connection is rather flaky. Will try the dev build.

Disgustus, 2018-05-29:
Just checked in the stock app. It indeed doesn't show visceral fat along the other measurements.
Will continue the bluetooth issues in the other ticket.

Disgustus, 2018-05-30:
You can close this issue, I'll post on #278 about the Bluetooth issues.

closed 2018-05-31 by erijo