oliexdev/openScale #284: Incorrect Units on Yunmai SE

mrsolitonwave, 2018-06-06:
Im using the Android App on a Pixel 2XL (Android 8.1) with the Yunmai SE scale and when lbs is selected as the preferred weight unit, the openscale app erroneously displays the weight in kg (but labels it as lbs).

For example, the Yunmai SE will display 175.4 lbs on the scale, but openscale will list my weight as "79.54lbs" (which appears to be the kg conversion but inappropriately labeled as lbs).

If I change my user preferences to kg, the Yunmai scale now displays my weight in kg and the openscale app also stores the correct weight also in kg. Thus, the issue seems to  be only with lbs for some reason.

erijo, 2018-06-06:
Can you enable debug log under settings, about and then do a measurement with lbs selected and attach the log here?

oliexdev, 2018-06-06:
@mrsolitonwave fixed with https://github.com/oliexdev/openScale/commit/7c2573f813504329ed7e86315a207679eee68dbd please try out the latest [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build).

closed 2018-06-06 by oliexdev