oliexdev/openScale #292: Goal weight not displayed properly in Chart and statistics.

faisal-shah, 2018-06-21:
v1.8.1 (28)

In user configuration I have my scale unit set to lb, and my goal weight set to the appropriate value in lbs.
When I view Chart and Statistics, it shows my goal weight as the number I have in my profile converted to the numerical equivalent in kgs. As a result, derived values are also incorrect, such as Weight Difference. 

For example, I put a goal weight of 150lb in my user profile. On Chart and Statistics it shows a goal weight of 68.0 lbs.

erijo, 2018-06-21:
The statistics bug is already fixed (see #272), but the graph bug remains.

erijo, 2018-06-21:
Should be fixed now. If you want to can verify it by testing the [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build).

Thanks for the report!

closed 2018-06-21 by erijo

faisal-shah, 2018-06-22:
Thanks for the app! Glad to be able to contribute in a small way.

commit a8ff04b5 (2018-06-21): Convert goal weight to user's unit before plotting in graph

Fix #292