oliexdev/openScale #295: Support for the official BLE standard

Shigawire, 2018-06-25:
Hi!

In the course of my master thesis I am developing a BLE simulator. We simulate the Bluetooth Low Energy standard and cross-test various mobile health applications to check for their compliance with the BLE standard.

I am pretty sure the simulator is mimicking a default BLE scale as defined per the BLE standardization (https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.service.weight_scale.xml) but still, the openScale application reports it to be not supported.

My question now is: Is this a not yet implemented device, or is my simulator not working properly?
All my tests and cross-checks seem to indicate that the standard is sufficiently simulated, though.

Bests

Fabian

erijo, 2018-06-26:
We don't have any generic support for a standard ble scale (that I know of). But I'd you're willing to add that it might be interesting to have.

oliexdev, 2018-07-04:
@Shigawire no your simulator is properly working but we identify the scale if it supported or not by the device name. 
Unfortunately, none of our currently supported scales are not complying with the BLE standard. Every company does their own implementation.
@Dododappere wrote in #219 that the Beurer BF600 use the BLE standard. But we don't own that scale but if you could implement with your BLE simulator a BLE standard scale in openScale that would be awesome.
Is your BLE simulator available, so we can test a BLE standard scale protocol?

Shigawire, 2018-07-07:
Hi @oliexdev I am developing the simulator in conjunction with the university clinic in Münster (Germany): https://www.ukm.de.
To date, the project is subject to an NDA, but I will talk to my supervisors and emphasize the importance of FOSS, especially in a field where standardization matters.
I am pretty sure we will find a solution for making the simulator available to you.

The Beurer devices do employ the BLE standard, but do not adhere to it to 100%, IIRC. We have a blood pressure monitor (BC57) here. One of the key issues with the vendor's apps is, that they implement additional BLE characteristics used for device-specific identification which we cannot (and, more importantly, do not want to simulate). BLE is supposed to be a standard (even if not open to public, looking at you, ISO), and manufacturers fiddling around adding custom layers to prevent third-party usage is not in the interests of the customers, patients and clinics all alike.

For implementation I almost exclusively relied on the information provided by the Bluetooth SIG, e.g. https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.service.weight_scale.xml.

Shigawire, 2018-07-16:
Hi @oliexdev,
we've discussed that we will publicly disclose the implementation particular for the weight scale services as soon as the master thesis is finished. I'll let you know then.

Bests

oliexdev, 2018-07-18:
it is a shame that the vendors don't fully comply to the BLE standard, isn't it?
Thanks for your effort to support the open source community and good luck with your master thesis.
I am looking forward to see your implementation of your BLE simulator. I will close this issue. Please open a new issue or send me an email if your are able to publish your weight scale simulator.

Best regards from Germany

oliexdev, 2018-11-27:
@shigawire did you finished your master thesis? Could you provide the BLE implementation?

dror3go, 2020-11-18:
Hi @Shigawire, a couple of years forward - did you finished your master thesis? Any chance for an update regarding the BLE implementation?
Thanks a lot!

Shigawire, 2020-11-19:
@oliexdev @dror3go Hey! The thesis has been finished, a while back I did ask my supervisor if my work can be publicly disclosed and the faculty did agree!

I would love to share the work done. However, please do keep in mind that this is a piece of academic work and it may not conform to the best practices and engineering standards we would impose on ourselves in a "real-world" engineering scenario. Say: It likely contains spaghetti code and "it just works"™ quirks.

However, I'll forward this GitHub issue to my supervisor of back then and we'll work on making the code publicly available.
Also, as time has passed, the Web Bluetooth standards seems to have emerged, which is pretty awesome: https://googlechrome.github.io/samples/web-bluetooth/device-info.html

Maybe we can port the app to the browser with native Web Bluetooth, as the current App is really just an Electron App with additional system-only Node.js modules included.

dror3go, 2020-12-27:
Hey @Shigawire, I hope I'm not nagging (too much).
Any news? When can we expect this to see the light of day? :) I for one am eager to see this topic moves forward.

Shigawire, 2020-12-27:
Hey @dror3go! Not nagging at all, the public interest on that matter is highly appreciated. I'll once again knock at the faculty and ask for a publication of the source code :) 

Expecting a feedback by end of next week.

Shigawire, 2021-01-20:
@dror3go Unfortunately, it does look like that the faculty is not providing the project's source code for now. I'll keep this project in the back of my head to let you know once this changes. Sorry!