oliexdev/openScale #299: Diagram view: Increase month selector minimum height

hirnsushi, 2018-07-02:
It's very hard to select a month with very few measurements, would be great if you could set a minimum height for those bars.
Examples:
[image]
1 measurement, nearly impossible to select.
[image]
2 measurements, still very hard to hit.

oliexdev, 2018-07-04:
Unfortunately, the library that we are using doesn't support that and it is not maintain anymore.
I am not sure how we can best solve this issue... any ideas?

hirnsushi, 2018-07-04:
Is the value shown on the bars linked to the bar height somehow? If not, would it be possible to increase the number of measurements in the background by x and use the real amount of measurements minus x for the value on the bars? I've obviously no idea how this is implemented, so just a thought 🙂

breversa, 2018-07-23:
Time to refactor the graph view then, and switch to a maintained library (if there's any...) ?

oliexdev, 2018-07-23:
@hirnsushi could be solution. More a hack but could work :wink: 
@breversa not for that one reason but if you have a good alternate you could suggest another library which doesn't lack at design.

oliexdev, 2018-08-05:
@hirnsushi please try out the latest dev build. Now the minimum height should be 20% of the maximum value.

closed 2018-08-05 by oliexdev

hirnsushi, 2018-08-05:
Perfect! 👍