oliexdev/openScale #300: Question about BF700 history data

postolache, 2018-07-03:
I have BF700 and it is working fine if the application is open when stepping on the scale. But it doesn't seems to download previous weight it measured while app was close.
In the supported scales page, BF700 is listed on History Data.
Should "Merge with last measurement" be checked? Tried both ways, still can't get history data from the scale. Any suggestions?
Thanks.

erijo, 2018-07-03:
"Merge with last measurement" only merges values from a previous measurement with the current in cases where a measurement isn't sent by the scale (i.e. waist circumference).

Can you please enable debug logging under settings -> about, connect to the scale and then attach the log here? If possible, please use the [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build).

postolache, 2018-07-03:
It works with the device version.
I'm attaching 2 logs one from Google play version (release) and 2nd from device version that is working:
[openScale_release.txt](https://github.com/oliexdev/openScale/files/2161249/openScale_release.txt)
[openScale_2018-07-04_00-10.txt](https://github.com/oliexdev/openScale/files/2161250/openScale_2018-07-04_00-10.txt)

erijo, 2018-07-03:
Glad to hear it's working with the latest version. Please try a couple of times to make sure it wasn't pure luck.

postolache, 2018-07-05:
[openScale4.txt](https://github.com/oliexdev/openScale/files/2165089/openScale4.txt)
Unfortunately you were right. This morning, the device version wasn't working .
What could be the problem?

erijo, 2018-07-05:
Did you have any measurements stored on the scale?

postolache, 2018-07-05:
Yes, I've weighted 2 times a few hours before, while the app was closed.
Not sure how to check on the scale if there are any measurements stored.

erijo, 2018-07-05:
Did the scale show your initials after weighing? I.e. was the weight within +/- 2 kg of the previous measurement?

postolache, 2018-07-05:
In settings I had "Smart user assignment" disabled. I've enabled now and
last 2 measurements were transferred.

I've noticed in the log that my birthday is 77-2-7 instead of 77-3-7 . In
app is set ok, March.

erijo, 2018-07-05:
The "Smart user assignment" doesn't affect reading the measurements from the scale. It is only used to assign a measurement to a user after it has been read from the scale. And as the scale already does this "smart" assignment it isn't really needed.

So if it works now, it was probably just pure luck. It would be nice to get some logs using the latest development version when it works.

As for the month, it is zero based so 0 is January and thus 2 is March.

postolache, 2018-07-06:
It seems that most of the time it gets something like this:

2018-07-06 18:50:22.684 Debug BluetoothBeurerSanitas: Ack Get scale status
2018-07-06 18:50:22.686 Debug BluetoothBeurerSanitas: BatteryLevel: 100,
weightThreshold: 2.00, BodyFatThreshold: 2.00, Unit: 1, userExists: true,
UserReference Weight Exists: true, UserMeasurementExists: false,
scaleVersion: 6
2018-07-06 18:50:42.142 Debug BluetoothCommunication$GattCallback:
onCharacteristicChanged 0000ffe1-0000-1000-8000-00805f9b34fb: E0 01
2018-07-06 18:50:42.147 Debug BluetoothBeurerSanitas: DataChanged - not
handled: *E0 01*

I will play more with it in the weekend and I will attach some logs.

postolache, 2018-07-08:
[openScale_2018-07-08_21-33.txt](https://github.com/oliexdev/openScale/files/2174042/openScale_2018-07-08_21-33.txt)
On that file there were 2 tries. First failed with something like : 
2018-07-08 21:34:30.326 Debug BluetoothBeurerSanitas: DataChanged - not handled: E0 01 

2nd try on same file was successful.
Unfortunately my phone doesn't seems to create the btsnoop log file to see if I get that E0 01 reply from original app.

oliexdev, 2018-08-18:
I don't know what the command `E0 01` from the scale is? Maybe an error code? Without more experimenting with the scale when and why this command happened it is hard to analyse. Therefore I will close this issue until we understand the command.

closed 2018-08-18 by oliexdev