oliexdev/openScale #301: [FR] Display every value in year view

breversa, 2018-07-03:
Hi,
I used to use a closed-source app before switching to openScale, but I've missed a feature ever since : the ability to display *every* weight value when in year view, and not only each month's first/average/last value (can't even tell which one it is).

Also, while while the graph's scaling (Ymin=min weight value, Ymax=max weight value) allows to more precisely observe weight variations, it makes it harder to see "the whole picture".
An option to NOT scale the graph would be highly appreciated (with Ymin =0, and maybe Ymax=max weight, rounded up to the next ten of kilograms).

oliexdev, 2018-07-04:
>  I used to use a closed-source app before switching to openScale, but I've missed a feature ever since : the ability to display every weight value when in year view, and not only each month's first/average/last value (can't even tell which one it is).

it is the average value which is displayed. You won't have every value in a year because then it looks like in #46.

> Also, while while the graph's scaling (Ymin=min weight value, Ymax=max weight value) allows to more precisely observe weight variations, it makes it harder to see "the whole picture".
> An option to [...]

True, we can disable the restriction for the viewport so you can zoom out to get a whole picture. I think we had already implemented it but I guess during a refactoring it has been vanished.

natrius, 2018-07-23:
It would be nice to implement a new line - median or something like that. That way it would be possible to show every weight in the graph while showing a smooth line simulatinously. 
An example what i mean can be found here:
https://www.belox.de/~oetjen/abnehmen.htm

I hope thats waht OP meant too.

breversa, 2018-07-23:
Yeah, that's not bad, though it's only getting useful once you've got lots of measure points (which I don't have yet).

Also, isn't there already a mean line for weight (but not for BMI) ?

natrius, 2018-07-23:
Ah, you will get more. I tend to weight at the beginning of every day so it does not change from what i eat and so on. Stable data. Always the same time and so on. Useful. 

Ouh, whoops, i just found out, the option for not showing the weight at each peak point is there already :D
And the line i meant seems to be the regression grade. Damn and thanks for the great work!

In this case just one thing: An overall diagram. From first input to last input. I could not find that one.

oliexdev, 2018-08-05:
I disabled the scale restriction so you can zoom out with commit https://github.com/oliexdev/openScale/commit/5e3fe90c6b924ef261faf44a40df72e077e9dd2a

@natrius No an overall diagram is not yet implemented, just a year overview.

closed 2018-08-05 by oliexdev