oliexdev/openScale #302: [FR] Display BMI in graph view

breversa, 2018-07-03:
Hi,
Why not display the BMI in the same graph as weight, along the Y-axis (and like #301, with an absolute rather than relative scale) ?

erijo, 2018-07-03:
Could you please elaborate a bit more on how you would like it to be? BMI can already be displayed in the graph, but I assume you want it in some other way?

breversa, 2018-07-04:
My bad : after double-checking th settings, I found that you can indeed display the BMI along the weight, so that's cool. :-)

Howewer, the BMI uses the same scale as the weight (= left Y-axis), which has the consequences that both plot lines look "compressed" to fit them in the same screen (I have a 5,5" display).

It would be better if the BMI used the right Y-axis with a different (but still absolute) scale.

breversa, 2018-07-04:
After seeing this : https://github.com/oliexdev/openScale/blob/master/doc/screens/screen_graph.png I come to the conclusion that there should be an option **to choose for each plot line to be scaled along the left or right axis**.

Currently, all the plot lines are scaled on the same lef Y-axis, be it weight (which can range up to 150 kg and more) or BMI (which can hardly go above 40-ish). IMO, it makes little sense to display both those plot lines on the same scale (they aren't directly related anyway).

I don't use the other measurements myself, but I believe that they may fit better on different scales : for instance, you could group the "high values" measurements (like weight) on the left Y-axis, and the "low values" measurements (like BMI) on the right Y-axis.

oliexdev, 2018-07-04:
@breversa the idea is not new but to have such an option needs some time. I need to think about it how to best implement such option during a bad weather day. ;)

oliexdev, 2018-12-26:
please test the latest openScale branch build of [MPAndroidChart](https://github.com/oliexdev/openScale/releases/tag/travis-dev_MPAndroidChart-build). You can select for each measurement if they are belong to the right or left graph axis.

closed 2019-02-10 by oliexdev