oliexdev/openScale #303: Measurements from Sanitas scale duplicated for both users

erijo, 2018-07-03:
From Google Play review:

> Very nice app, fully compatible with my sanitas scale. I don't give 5 stars 
just because of a little bug. I use it with two users and, every time we 
take a new measure, the app duplicates some measurements from one user to 
the another one. So, every time I have to delete those data from the list 
of past measurements. Hope you may fix it, the app is very good!

erijo, 2018-07-03:
It would be good if we could get a debug log from the [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) when this happens (enable debug log under settings -> about).

oliexdev, 2018-08-18:
seems he/she is the only one that have the problem. I will close it until more user reporting this behaviour.

erijo, 2018-09-30:
Reopen as requested on Google play.

giovanni-malara, 2018-09-30:
Thank you to have reopened it again. Apparently, the issue occurs randomly, so, some times, everything works well for many weeks. Anyway, if you can keep an eye on that, it would be great. I attach the last log file. I'm not very familiar with these technical files. Eventually, I can get a new one if I didn't do it properly.
[openScale_2018-09-15_09-12.txt](https://github.com/oliexdev/openScale/files/2431926/openScale_2018-09-15_09-12.txt)

erijo, 2018-10-06:
Do you by any chance have "Smart User assignment" enabled in the settings?

If not, can you try to reproduce the problem with the [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build)? Don't forget to enable the debug log.

giovanni-malara, 2018-10-07:
I have checked: no, it was not enabled. I enable it with the debug log and send you the file. I'll try to reproduce the problem both with the dev version and the standard one once I'm back home next week. Thanks!

oliexdev, 2018-10-28:
@giovanni-malara any progress? Otherwise I will close this issue. Btw you can still post on this issue even if it closed.

giovanni-malara, 2018-10-28:
Hi @oliexdev, no progress. I tried to reproduce the problem but no way to see it again. Maybe the smart user assignement option solved the issue? Anyway, no problem to close the post. Eventually, I'll post again if I see the issue again and I have the log file. Thank you!

closed 2018-11-01 by oliexdev