oliexdev/openScale #305: Add Yunmai Balance

jasimancas, 2018-07-05:
Hi!
Do you think if its possible add support to Yunmai Balance?

erijo, 2018-07-05:
We can try!

Start by installing the [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build). Then go to settings -> about and enable debug log. Then go to settings -> bluetooth and search for your scale. Once found, click on it. This should get a list of services etc. When it is done, attach the log here.

jasimancas, 2018-07-06:
> 2018-07-06 07:10:53.554 Debug [1] AboutPreferences: Debug log enabled, openScale (dev) v1.8.1-dev_269cbae1_2018-07-05 (1530831301), SDK 24, Xiaomi MI 5s Plus 2018-07-06 07:11:02.330 Debug [1] BluetoothPreferences: Start discovery 2018-07-06 07:11:02.419 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.adapter.action.DISCOVERY_STARTED 2018-07-06 07:11:02.421 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND 2018-07-06 07:11:02.436 Debug [1] BluetoothPreferences: Found unsupported device MIBOX3 [00:9E:C8:93:F1:83] (type: 3) 2018-07-06 07:11:02.449 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND 2018-07-06 07:11:02.457 Debug [1] BluetoothPreferences: Found unsupported device YUNMAI-ISM2-EN [64:CF:D9:0C:5C:00] (type: 2) 2018-07-06 07:11:03.266 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND 2018-07-06 07:11:08.037 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND 2018-07-06 07:11:08.052 Debug [1] BluetoothPreferences: Found unsupported device Flower care [C4:7C:8D:65:D8:0D] (type: 2) 2018-07-06 07:11:09.629 Debug [1] BluetoothPreferences: Stop discovery 2018-07-06 07:11:09.684 Debug [1] OpenScale: Trying to connect to bluetooth device [64:CF:D9:0C:5C:00] in debug mode 2018-07-06 07:11:09.694 Debug [1] BluetoothCommunication: BT is enabled, state=12, scan mode=23, is not discovering 2018-07-06 07:11:09.701 Debug [1] BluetoothCommunication: Connected GATT device: Amazfit Bip Watch [D3:0C:41:DD:D2:93] 2018-07-06 07:11:09.707 Debug [1] BluetoothCommunication: Connected GATT_SERVER device: Amazfit Bip Watch [D3:0C:41:DD:D2:93] 2018-07-06 07:11:09.712 Debug [1] BluetoothCommunication: Starting LE scan for device [64:CF:D9:0C:5C:00] 2018-07-06 07:11:09.800 Debug [1] BluetoothCommunication: Found LE device null [73:4C:93:5F:3B:DC] 2018-07-06 07:11:09.854 Debug [1] BluetoothCommunication: Found LE device null [14:BB:6E:23:64:93] 2018-07-06 07:11:10.022 Debug [1] BluetoothCommunication: Found LE device YUNMAI-ISM2-EN [64:CF:D9:0C:5C:00] 2018-07-06 07:11:10.023 Debug [1] BluetoothCommunication: Stopping LE scan 2018-07-06 07:11:10.029 Info [1] BluetoothCommunication: Connecting to [64:CF:D9:0C:5C:00] (driver: Debug) 2018-07-06 07:11:10.039 Debug [1] BluetoothCommunication: Found LE device Flower care [C4:7C:8D:65:D8:0D] 2018-07-06 07:11:10.762 Debug [3544] BluetoothCommunication$GattCallback: onConnectionStateChange: status=0, newState=2 2018-07-06 07:11:11.809 Debug [3544] BluetoothCommunication$GattCallback: onServicesDiscovered: status=0 (4 services) 2018-07-06 07:11:12.818 Debug [3544] BluetoothCommunication: INIT STATE: 0 2018-07-06 07:11:12.834 Debug [3544] BluetoothCommunication: Read characteristic 00002a00-0000-1000-8000-00805f9b34fb 2018-07-06 07:11:12.886 Debug [3544] BluetoothCommunication$GattCallback: onCharacteristicRead 00002a00-0000-1000-8000-00805f9b34fb (status=0): 59 55 4E 4D 41 49 2D 49 53 4D 32 2D 45 4E 2018-07-06 07:11:12.953 Debug [1] BluetoothCommunication: INIT STATE: 1 2018-07-06 07:11:12.962 Debug [1] BluetoothCommunication: Read characteristic 00002a01-0000-1000-8000-00805f9b34fb 2018-07-06 07:11:13.013 Debug [3544] BluetoothCommunication$GattCallback: onCharacteristicRead 00002a01-0000-1000-8000-00805f9b34fb (status=0): 00 00 2018-07-06 07:11:13.081 Debug [1] BluetoothCommunication: INIT STATE: 2 2018-07-06 07:11:13.091 Debug [1] BluetoothCommunication: Read characteristic 00002a02-0000-1000-8000-00805f9b34fb 2018-07-06 07:11:13.134 Debug [3544] BluetoothCommunication$GattCallback: onCharacteristicRead 00002a02-0000-1000-8000-00805f9b34fb (status=0): 00 2018-07-06 07:11:13.202 Debug [1] BluetoothCommunication: INIT STATE: 3 2018-07-06 07:11:13.210 Debug [1] BluetoothCommunication: Read characteristic 00002a04-0000-1000-8000-00805f9b34fb 2018-07-06 07:11:13.255 Debug [3544] BluetoothCommunication$GattCallback: onCharacteristicRead 00002a04-0000-1000-8000-00805f9b34fb (status=0): 50 00 A0 00 00 00 E8 03 2018-07-06 07:11:13.325 Debug [1] BluetoothCommunication: INIT STATE: 4 2018-07-06 07:11:13.335 Debug [1] BluetoothCommunication: Read descriptor 00002902-0000-1000-8000-00805f9b34fb 2018-07-06 07:11:13.382 Debug [3544] BluetoothCommunication$GattCallback: onDescriptorRead 00002902-0000-1000-8000-00805f9b34fb (status=0): 00 00 2018-07-06 07:11:13.454 Debug [1] BluetoothCommunication: INIT STATE: 5 2018-07-06 07:11:13.463 Debug [1] BluetoothCommunication: Read descriptor 00002902-0000-1000-8000-00805f9b34fb 2018-07-06 07:11:13.497 Debug [3544] BluetoothCommunication$GattCallback: onDescriptorRead 00002902-0000-1000-8000-00805f9b34fb (status=0): 00 00 2018-07-06 07:11:13.568 Debug [1] BluetoothCommunication: INIT STATE: 6 2018-07-06 07:11:13.577 Debug [1] BluetoothCommunication: Read descriptor 00002901-0000-1000-8000-00805f9b34fb 2018-07-06 07:11:13.639 Debug [3544] BluetoothCommunication$GattCallback: onDescriptorRead 00002901-0000-1000-8000-00805f9b34fb (status=0): 4B 65 79 20 50 72 65 73 73 20 53 74 61 74 65 2018-07-06 07:11:13.708 Debug [1] BluetoothCommunication: INIT STATE: 7 2018-07-06 07:11:13.717 Debug [1] BluetoothCommunication: Read descriptor 00002901-0000-1000-8000-00805f9b34fb 2018-07-06 07:11:13.753 Debug [3544] BluetoothCommunication$GattCallback: onDescriptorRead 00002901-0000-1000-8000-00805f9b34fb (status=0): 4B 65 79 20 50 72 65 73 73 20 53 74 61 74 65 2018-07-06 07:11:13.822 Debug [1] BluetoothCommunication: INIT STATE: 8 2018 [... cut ...] toothDebug: |- characteristic 0000ffea-0000-1000-8000-00805f9b34fb (instance 26): WRITE_NO_RESPONSE (permissions=0x0) 2018-07-06 07:11:14.002 Debug [1] BluetoothDebug: |--- descriptor 00002901-0000-1000-8000-00805f9b34fb (permissions=0x0) 2018-07-06 07:11:14.004 Debug [1] BluetoothDebug: |-----> value: 4B 65 79 20 50 72 65 73 73 20 53 74 61 74 65 2018-07-06 07:11:14.006 Info [1] BluetoothCommunication: Disconnecting 2018-07-06 07:11:14.009 Debug [1] OpenScale: Disconnecting from bluetooth device

jasimancas, 2018-07-06:
[openScale_2018-07-06_07-10 (1).txt](https://github.com/oliexdev/openScale/files/2169270/openScale_2018-07-06_07-10.1.txt)

erijo, 2018-07-06:
Please retest with latest development version and report back the status (with log).

jasimancas, 2018-07-07:
[openScale_2018-07-07_12-15.txt](https://github.com/oliexdev/openScale/files/2172654/openScale_2018-07-07_12-15.txt)

erijo, 2018-07-07:
By the looks of it it doesn't seem to work like the mini. To be able to add support for this scale I think we would need a bt snoop log (see [step 1 on the wiki](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale)).

oliexdev, 2018-07-22:
issue closed due inactivity

closed 2018-07-22 by oliexdev

Danix58, 2018-08-21:
the scale is supported #317