oliexdev/openScale #306: App crashes on start after removal of polynomial degree

LuccoJ, 2018-07-09:
I blanked the polynomialnfit degree in the settings thinking that would simply remove the polynomial line. Instead, now the app dumps a stack trace on start (also after wiping cache). The last open activity was the "all time" chart (neither Week nor Month selected).

Build version: 1.8.1 
Build date: 1979-11-30 01:00:00 
Current date: 2018-07-09 19:10:24 
Device: ZTE A2017U 
 
Stack trace:  
java.lang.RuntimeException: Unable to start activity ComponentInfo{com.health.openscale/com.health.openscale.gui.MainActivity}: java.lang.NumberFormatException: For input string: ""
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2778)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2856)
	at android.app.ActivityThread.-wrap11(Unknown Source:0)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1589)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loop(Looper.java:164)
	at android.app.ActivityThread.main(ActivityThread.java:6494)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:440)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)
Caused by: java.lang.NumberFormatException: For input string: ""
	at java.lang.Integer.parseInt(Integer.java:620)
	at java.lang.Integer.parseInt(Integer.java:643)
	at com.health.openscale.gui.fragments.GraphFragment.generateLineData(GraphFragment.java:323)
	at com.health.openscale.gui.fragments.GraphFragment.generateGraphs(GraphFragment.java:530)
	at com.health.openscale.gui.fragments.GraphFragment.updateOnView(GraphFragment.java:247)
	at com.health.openscale.core.OpenScale.registerFragment(OpenScale.java:617)
	at com.health.openscale.gui.fragments.GraphFragment.onCreateView(GraphFragment.java:225)
	at android.support.v4.app.Fragment.performCreateView(Fragment.java:2346)
	at android.support.v4.app.FragmentManagerImpl.moveToState(FragmentManager.java:1428)
	at android.support.v4.app.FragmentManagerImpl.moveFragmentToExpectedState(FragmentManager.java:1759)
	at android.support.v4.app.FragmentManagerImpl.moveToState(FragmentManager.java:1827)
	at android.support.v4.app.FragmentManagerImpl.dispatchStateChange(FragmentManager.java:3244)
	at android.support.v4.app.FragmentManagerImpl.dispatchActivityCreated(FragmentManager.java:3200)
	at android.support.v4.app.FragmentController.dispatchActivityCreated(FragmentController.java:195)
	at android.support.v4.app.FragmentActivity.onStart(FragmentActivity.java:597)
	at android.support.v7.app.AppCompatActivity.onStart(AppCompatActivity.java:177)
	at android.app.Instrumentation.callActivityOnStart(Instrumentation.java:1334)
	at android.app.Activity.performStart(Activity.java:7029)
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2741)
	... 9 more

User actions:

oliexdev, 2018-07-18:
thanks for your issue. Should be fixed with commit https://github.com/oliexdev/openScale/commit/fd0b5e6f9b13bb7407af1df4e54ab4ceb0091f2f

closed 2018-07-18 by oliexdev

LuccoJ, 2018-09-22:
I'm glad to know it's fixed, but do you think it will be long before the next release with the fix, so I can access my data again?