oliexdev/openScale #307: Import of csv files does not work.

rp-916, 2018-07-12:
I extracted weight details from my weightguru app into a csv file. I tried to import the same csv into a openscale app, but it failed with following error. 
> 
**Error importing: Problem parsing for type Date**
> 
I tried to search the format of the csv file compatible with openScale, but unfortunately didn't find. 
Appreciate, if somebody can update the CSV file format to be imported?

rp-916, 2018-07-19:
I actually did that and observed that the fields in file,I exported is completely different from what Openscale code expects. Reason I posted above question is for following:
  - Is there any standards for storing these kind of fields recommended by any community?  If there is standards, then integrations to different app would be easier and even end-user will have clarity without any additional documentation. 
  - As there is no documentation, it would be very difficult for non-technical people how to change the csv files. Documentation might help even non-technical people if importing does not work.

oliexdev, 2018-07-19:
> I actually did that and observed that the fields in file,I exported is completely different from what Openscale code expects.

what did you expect from the openScale source code?

> Is there any standards for storing these kind of fields recommended by any community? If there is standards, then integrations to different app would be easier and even end-user will have clarity without any additional documentation.

there is no standard which I aware of but CSV is the simplest format which I can imagine.

> As there is no documentation, it would be very difficult for non-technical people how to change the csv files. Documentation might help even non-technical people if importing does not work.

What is so hard to understand the CSV file? You have the header row which explains what is what and then you have for each row your data!? Ok, the date format have to been known but if you simple do a sample export and look into the exported file you should understand that too. 
But if you believe a documentation will help then you are welcome to write some information about the CSV file. If it helpful I can put it then on the wiki.