oliexdev/openScale #308: Request to make contributions to openScale.

HarryHeights, 2018-07-16:
Hello!.. 
I'll love to contribute to your project in documentation. I'll draft a privacy policy and also compile a list of Frequently asked questions(FAQs).
It'll be great thing if prospective users know what information they'll be parting away with while using the software and how that information is utilized. The FAQs will solve some usage problems for prospective users. 

I'll be anticipating your reply.

HarryHeights, 2018-07-18:
Alright. I already made a proposition though. If by proposals you mean compiling the document and submitting it for review, I'll do that right away.

oliexdev, 2018-07-23:
sure propose your documentation. I cannot promise to you I will use it but I will look into it.

HarryHeights, 2018-07-24:
Oh okay. I'll get back to you real soon.

oliexdev, 2018-08-18:
I will close this issue due inactivity.

closed 2018-08-18 by oliexdev