oliexdev/openScale #309: Support for Senssun Fat scales

freman, 2018-07-25:
Hi.

Scales are sold here in AU at Kmart, the original app worked beautifully just one time.

I can have a go at reverse engineering it

freman, 2018-07-30:
Turns out debugging this btle stuff is hard lol.

freman, 2018-07-30:
I have something that works some of the time.

Not made any easier by the fact that I don't think their scales work the way they think their scales work.

I copy/paste their gender code in and it still doesn't work for men, only women.

If it doesn't just ignore you outright.

oliexdev, 2018-08-01:
Could you send a PR with your first implementation?

dagbdagb, 2018-11-20:
@freman Did you get any further on this? Do you care to share your code? Looking to buy one of these dirt cheap scales locally, and having open source support for it is a major plus in my book.

gmarco, 2018-11-25:
Hi i have the same Scale. Initial Pull Request here #355.

some smaller Issues to fix (written in den PR)

hope this helps every Senssun User

dagbdagb, 2018-11-25:
Yay!
Thank you @gmarco !

gmarco, 2018-11-27:
Last update works very well IF the scale will connect.
i must rec/disconnect the scale very often, because it will not connect so fast (if connected, all works)

https://gist.github.com/gmarco/0678468278026dd0894cda978c87c745

here a log of the connecting tries. (i en-/disabled the BT button after nicht connecting after some seconds)

the status code 133 is also described (like in the sourcecode here written).
https://android.jlelse.eu/lessons-for-first-time-android-bluetooth-le-developers-i-learned-the-hard-way-fee07646624

where must be the delay added ??

2. can we change the API to updated the last measure ? (scale will get first , weight, later (but not everytime) it will get the other data.

freman, 2018-11-27:
Sorry, I have my source here somewhere but I had the same problem. Couldn't decide if crap scale was crap or if I sucked at Bluetooth

oliexdev, 2018-11-27:
thanks @gmarco as you already linked the Android Bluetooth stack is really corrupted. I consider on issue #357 to use a Bluetooth Android library to avoid those problems.

You could temporally store your measurement to get all measures.

erijo, 2018-11-27:
Try making doScanWhileConnecting return false for Senssun. See #353

gmarco, 2018-11-27:
with the doScanWhileConnecting return falseit works much much better :+1: 
did da time wait for new data (with fat and so on), but the timing is not every time the same.

do we have a "onDisconnect" Funtion to override ?

BTW: the openScale App is now :100: time better than the original

erijo, 2018-11-27:
Glad to hear that it works better now @gmarco!

Can you attach a log from openScale when you connect to the scale and fetch a measurement?

erijo, 2018-11-27:
Can you post a btsnoop log from the original app as well?

gmarco, 2018-11-27:
i would, but my phone is not saving this file, even if enabled in developer options

erijo, 2018-11-27:
You can try going to developer settings and click "Take bug report". The snoop log might be in there.

On my Nexus 5X phone i have to enable btsnoop, restart bluetooth and then the phone. After that I can get the log via the "take bug report" option.

freman, 2018-11-28:
This is what I wrote, not sure if it helps

https://gist.github.com/freman/803c3e7c2bbd40303ca4cd680da62721

Dellu, 2019-01-20:
Thanks for supporting Senssun scale. This is nice. I am having the following error: [screeshot](http://take.ms/1kUjY)

edent, 2020-07-25:
I have these scales, but I can't get the OpenScale to connect.  I've attached the debug log from the Dev version of the app.
[openScale_2020-07-25_08-35.txt](https://github.com/oliexdev/openScale/files/4975062/openScale_2020-07-25_08-35.txt)

I can connect with BLE Scanner. But OpenScale displays `IF_B6A` as not supported.

vger76, 2020-12-17:
Hi, there is a regression with the Senssun fat scale which wasn't remedied with the latest update. Should I open a new issue?

milpugeda, 2021-07-02:
Is this still being worked on? When will the scales be officially supported?

mptp24, 2021-07-02:
Is this still being worked on? When will the scales be officially supported?

My scale is also showing up as "device not supported" with the name "IF_B2".

Would be great to use openScale because the scale's official app is not good.

mptp24, 2021-07-18:
@oliexdev I know this issue has been opened for while and has been inactive but I'd like to give it a try and fix the issue. Any pointers on how I can support my Senssun scale with name "IF_B2"? Or are there other infos that I need to get instead of the name. Thanks.

czare, 2023-01-10:
Hi friends, any chances of having this implemented? What exactly is missing? unfortunately I'm not an android developer, but if there's anything I could do to help, I'm in.

judo16, 2024-12-26:
Hi there, 

I allay have the scales showing up as IF_B2 and would be keen to help out if I can. The app recommended for the scales is a bit flakey and there's no automated way to get the data out. Please let me know if there's anything I can assist with to get these scales supported. 

Kind regards,