oliexdev/openScale #310: Question BMR on Sanitas SBF70, strange values

new-on-github, 2018-07-28:
Who is the BMR calculated on the Sanitas SBF70?

The values in my App are totally different from the ones with the formulars on Wikipedia:
[https://en.wikipedia.org/wiki/Basal_metabolic_rate](https://en.wikipedia.org/wiki/Basal_metabolic_rate)

Can it be possible that the BMR depends on measurements and not only one the users given data? Seems odd to me.

Maybe someone can help me. Thanks a lot for the hole app!

oliexdev, 2018-07-29:
Which App do you mean? openScale or the Sanitas App?

I don't know how the Sanitas App does the calculation. You have to ask the manufacture and not me. 😉 

For openScale the BMR is calculated by 

> Mifflin, St Jeor et al: A new predictive equation for resting energy expenditure in healthy individuals

Furthermore the activity level is included with a factor taken from [this website](http://www.globalrph.com/harris-benedict-equation.htm).

new-on-github, 2018-07-30:
Thanks for your quick answer.

I don't know the Sanitas App. I mean the open scale app. 

If I understand you correctly, the value which is shown under BMR is not the BMR, but instead 
> A rough estimate of the number of calories required per day based on your current activity level
?

oliexdev, 2018-07-30:
yes, it is more like a RMR but they are closely related to BMR

> The Resting Metabolic Rate (RMR) is closely related to the basal metabolic rate (BMR) and it is the amount of energy required to maintain the body's normal metabolic activity, such as respiration, maintenance of body temperature (thermogenesis), and digestion. Specifically, it is the amount of energy required at rest with no additional activity. The energy consumed is sufficient only for the functioning of the vital organs such as the heart, lungs, nervous system, kidneys, liver, intestine, sex organs, muscles, and skin.

closed 2018-07-30 by oliexdev

new-on-github, 2018-07-31:
But would it be not better to write RMR in the App instead of BMR?

Because in my opinion it's strange that my BMR is around 3000kCal.