oliexdev/openScale #311: Zoetouch scale (model CF390BLE) not supported

JaimeGF, 2018-07-30:
Hi,

First of all, English is not my mother tongue. Please excuse any errors on my part.

I have three files "btsnoop_hci.log", with different weights.
I installed BLE Scanner app and I got the UUID numbers of characteristic and service.
I open the files in Wireshark but I am unable to find valid data. Who can help me better understand the files to find my weight and other data?

Regards,

oliexdev, 2018-07-30:
post your information here and attached your corresponded log files. If I have time, I will look into it or somebody else have the obligation to look into it.
In the mean time try to search for your weight in the hex log files via wireshark.

JaimeGF, 2018-07-30:
Files "btsnoop_hci.log" (renamed 01, 02 and 03) and weigth:

01.log = 91.6 kg
02.log = 97.4 kg
03.log = 102.1 kg

Also attached to this message are screenshots of BLE Scanner app.

Thanks in advance,

El software de antivirus Avast ha analizado este correo electrónico en busca de virus.
https://www.avast.com/antivirus

JaimeGF, 2018-08-04:
I keep trying but I can’t find useful data on any of the files. Maybe I don’t know how to look for it. I recreated the files but the result is the same, I can’t find anything. It is possible that the scale isn’t registering the data into the file, or maybe they’re under a different format? I’ll keep looking into it to see if I can find anything.

Regards,

oliexdev, 2018-08-05:
the weight is stored in big endian multiplied by 100 but in log 1 and 3 you only received the data via the advertising frame. Normally, the advertising frame are used for constant updates while your weighting. So did you get a error for log 1 and 3 from your scale? Or did something other happened from log 2?

In log 2 you get a typical weight notification on the characteristic `0000FFF4-0000-1000-8000-00805f9b34fb`with a value:

`cf:84:12:0c:26:81:85:c0:00:00:b7` weight: 0x260c (9740 / 100 = 97,4 kg)

Does you scale support only weight or other body measurement as well?
Next thing is to understand the other values in the notification value above. 
I guess last value `0xb7` could be the XOR checksum and `0xcf` the starting byte.
Rest could be the date and time but for that we need a log file with an exact date and time for a measurement.

JaimeGF, 2018-08-16:
Hi, 

I have tried to generate about 30 files "btsnoop_hci.log" and I only found weight data in one of them. It is quite complicated because my galaxy S8 does not generate this file and I do it with a galaxy S5 that I do not always have available. At the moment I leave it on stand-by until I have again available the Galaxy S5 to continue testing.

oliexdev, 2018-10-28:
@JaimeGF any progress in testing?

JaimeGF, 2018-11-10:
> 
> 
> @JaimeGF any progress in testing?

JaimeGF, 2018-11-10:
No, no progress in testing. I think it's better to close this issue. Thanks.

closed 2018-11-10 by JaimeGF