oliexdev/openScale #314: App crashes and don't start anymore showing a bug

logintonico, 2018-08-10:
Build version: 1.8.1 
Build date: 1979-11-30 00:00:00 
Current date: 2018-08-10 08:36:21 
Device: HUAWEI PRA-LX1 
 
Stack trace:  
java.lang.RuntimeException: Unable to start activity ComponentInfo{com.health.openscale/com.health.openscale.gui.MainActivity}: java.lang.NumberFormatException: For input string: ""
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3194)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3302)
	at android.app.ActivityThread.-wrap12(Unknown Source:0)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1891)
	at android.os.Handler.dispatchMessage(Handler.java:108)
	at android.os.Looper.loop(Looper.java:166)
	at android.app.ActivityThread.main(ActivityThread.java:7425)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:245)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:921)
Caused by: java.lang.NumberFormatException: For input string: ""
	at java.lang.Integer.parseInt(Integer.java:620)
	at java.lang.Integer.parseInt(Integer.java:643)
	at com.health.openscale.gui.fragments.GraphFragment.generateLineData(GraphFragment.java:323)
	at com.health.openscale.gui.fragments.GraphFragment.access$1100(GraphFragment.java:76)
	at com.health.openscale.gui.fragments.GraphFragment$chartTopValueTouchListener.onValueSelected(GraphFragment.java:544)
	at lecho.lib.hellocharts.view.ColumnChartView.callTouchListener(ColumnChartView.java:73)
	at lecho.lib.hellocharts.view.AbstractChartView.selectValue(AbstractChartView.java:449)
	at com.health.openscale.gui.fragments.GraphFragment.generateColumnData(GraphFragment.java:484)
	at com.health.openscale.gui.fragments.GraphFragment.generateGraphs(GraphFragment.java:519)
	at com.health.openscale.gui.fragments.GraphFragment.updateOnView(GraphFragment.java:247)
	at com.health.openscale.core.OpenScale.registerFragment(OpenScale.java:617)
	at com.health.openscale.gui.fragments.GraphFragment.onCreateView(GraphFragment.java:225)
	at android.support.v4.app.Fragment.performCreateView(Fragment.java:2346)
	at android.support.v4.app.FragmentManagerImpl.moveToState(FragmentManager.java:1428)
	at android.support.v4.app.FragmentManagerImpl.moveFragmentToExpectedState(FragmentManager.java:1759)
	at android.support.v4.app.FragmentManagerImpl.moveToState(FragmentManager.java:1827)
	at android.support.v4.app.FragmentManagerImpl.dispatchStateChange(FragmentManager.java:3244)
	at android.support.v4.app.FragmentManagerImpl.dispatchActivityCreated(FragmentManager.java:3200)
	at android.support.v4.app.FragmentController.dispatchActivityCreated(FragmentController.java:195)
	at android.support.v4.app.FragmentActivity.onStart(FragmentActivity.java:597)
	at android.support.v7.app.AppCompatActivity.onStart(AppCompatActivity.java:177)
	at android.app.Instrumentation.callActivityOnStart(Instrumentation.java:1339)
	at android.app.Activity.performStart(Activity.java:7392)
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3157)
	... 9 more

User actions: 
Changed the parameter 1 in configuration/graphic/tendencies to "blank"

What I must do to open the app and not loose all data? 
Please help

Thanks

oliexdev, 2018-08-12:
Did you have a backup or a CSV export?
This is already fixed in the latest dev build.

logintonico, 2018-08-13:
I don't have neither of both. If I reinstall the app, is it possible  
to recover the data that is anywhere in the memory phone? 

Thanks

Ligações:
---------
[1] https://github.com/oliexdev/openScale/issues/314#issuecomment-412329489
[2]  
https://github.com/

oliexdev, 2018-08-14:
Unfortunately, no if you reinstall the app the database is deleted.

closed 2018-08-14 by oliexdev