oliexdev/openScale #315: Add support for Crane scale (aldi)

MaUwJo, 2018-08-17:
It's this one:
https://www.cranesportsconnect.com/manuals/SCALE2/94283_20160722_AT_web.pdf
with this app:
https://play.google.com/store/apps/details?id=com.aldi_project&hl=de

I have the log files from connecting and syncing
(https://github.com/oliexdev/openScale/files/2298490/btsnoop_hci.log.connect.zip.zip)
(https://github.com/oliexdev/openScale/files/2298492/btsnoop_hci.log.wiegen.zip.zip)

But I don't have the skills to develop it. Maybe someone like to do it.

oliexdev, 2018-08-18:
thanks could you do the steps 1-3 of the [wiki](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale).
I need at least more logs with different weights and the BLE scan.

Thanks in advance.

MaUwJo, 2018-09-05:
Hi,

here are some more logs:

[btsnoop_hci.log.81.3.zip](https://github.com/oliexdev/openScale/files/2354582/btsnoop_hci.log.81.3.zip)
[btsnoop_hci.log](https://github.com/oliexdev/openScale/files/2354583/btsnoop_hci.log)
[log81.4.zip.zip](https://github.com/oliexdev/openScale/files/2354635/log81.4.zip.zip)

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[ble.txt](https://github.com/oliexdev/openScale/files/2354630/ble.txt)

MaUwJo, 2018-10-01:
Do you need anything more? Should I test something?

MaUwJo, 2018-10-01:
I have here a script to create the openscale csv file from the content of crane cloud ...
[waage_pw.pl.txt](https://github.com/oliexdev/openScale/files/2435600/waage_pw.pl.txt)

oliexdev, 2018-10-12:
@MaUwJo without the corresponding exact true date/time with all other information (e.g. weight, water percentage, bone mass and so on) the btsnoop_log files are useless.

I can't find the data in the btsnoop.log without knowing for what weight, fat, time and so on I am looking for.

So please create a btsnoop.log **and** write down the weight, fat, water and so on at the time your are weighting and recording the btsnoop.log (see "Acquiring some Bluetooth traffic" step 3 in the wiki)

MaUwJo, 2018-10-16:
Now with screenshots of the measurement: Do you need more?

[image]
[image]
[image]
[image]

oliexdev, 2018-10-16:
for which btsnoop_hci.log are the screenshots?

MaUwJo, 2018-10-16:
for:
btsnoop_hci.log.81.3.zip
log81.4.zip.zip
Comment from 09/05/18.

Thanks

MaUwJo, 2018-12-24:
Screenshot an log are attached.

[btsnoop_hci.log12 2018.zip](https://github.com/oliexdev/openScale/files/2706828/btsnoop_hci.log12.2018.zip)

Hopefully I will see your reply next time faster.

MaUwJo, 2018-12-24:
If you are from Germany, I can send you the device for finishing the development. This would increase the developing speed.
I would also pay the costs for sending the scale back after your development.

Thanks

Jantleing, 2025-05-17:
I own this scale..  decompiled the app and found commands for this scale:

[image]

it turned out that this Scale uses same protocol as Beurer and Sanitas Scales with start byte E7 (like SANITAS SBF70) as described here: https://github.com/oliexdev/openScale/wiki/Beurer-Sanitas

The scale identifies itself as ` AICDSCALE1` , so I added it to BluetoothFactory.java

[image]

and everything works as expected:

[image]

[image]

[image]

Since Servers were closed and App isn't available anymore, this Scale basically is unusable, so adding it to OpenScale would be great to revive it's full functionality.

oliexdev, 2025-05-22:
@Jantleing great discovery, would you be so kind to send a PR?

closed 2025-06-20 by oliexdev