oliexdev/openScale #316: Add support for Yunmai Premium Smart Scale

thoth291, 2018-08-18:
Yunmai app sucks - your one looks exactly what I need, but it doesn't seem to support my scale. Please help.

oliexdev, 2018-08-19:
I would like to support this scale but I guess it has the same problem as the yunmai mini scale, please see issue #71

oliexdev, 2018-08-20:
You can also manually enter your weight and estimate your body measurements with openScale ;)
I will close this issue until we can resolve the puzzle in issue #71.

closed 2018-08-20 by oliexdev

oliexdev, 2018-11-27:
Please enable the log under the settings-about debug and post it here (read the wiki for details)

OrcaXS, 2018-11-27:
[openScale_2018-11-27_18-57[1].txt](https://github.com/oliexdev/openScale/files/2619575/openScale_2018-11-27_18-57.1.txt)
Here it is.

oliexdev, 2018-11-27:
@OrcaXS try out the latest [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) and post the log file here.

OrcaXS, 2018-11-27:
[openScale_2018-11-27_18-57[1].txt](https://github.com/oliexdev/openScale/files/2619687/openScale_2018-11-27_18-57.1.txt)
Looks like it's working fine!

oliexdev, 2018-11-27:
Good but you posted the same log file again. ;)

OrcaXS, 2018-11-27:
[openScale_2018-11-27_19-30[1].txt](https://github.com/oliexdev/openScale/files/2619705/openScale_2018-11-27_19-30.1.txt)
Oops... that was the cached file from MTP on Windows