oliexdev/openScale #317: ADD SUPPORT YNUMAI Mini 2 SCALE (YNUMAI-ISM2-EN)

Danix58, 2018-08-20:
I find a Ynumai mini 2 scale that is not support. I'm very insteresting in openScale.

The device name identified by BLE scanner is: **YNUMAI-ISM2-EN**  (3C:A3:08:97:33:F0)

**20-08-2018 13.36**
[btsnoop_hci -10.log](https://github.com/oliexdev/openScale/files/2302437/btsnoop_hci.-10.log)
`0d:1f:14:02:00:5b:7a:a7:a2:23:d8:82:ea:1e:01:01:bb:05:28:36  // weight: 76.8 Kg 13.2 Body fat`
1e:01 = 76.81 kg
05:28 = 13.2 % body fat

The protocol is the same of BluetoothYunmaiSE_Mini.java. #71 
The scale image is the same of [image1](https://raw.githubusercontent.com/oliexdev/openScale/master/doc/scales/yunmai_mini.jpg)
Please help.

oliexdev, 2018-08-20:
please try out the latest [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build).

Danix58, 2018-08-20:
I try th new dev version. Now i can find my scale. Thanks

oliexdev, 2018-08-20:
and does it work?

Danix58, 2018-08-20:
Yes, it works

oliexdev, 2018-08-20:
great :smile: 
Does your scale support other body measurements like water and body fat percentage?
Could you post an image again but with the same camera angle like the one in out wiki gallery?

closed 2018-08-20 by oliexdev

Danix58, 2018-08-21:
[image]

The scale have these body mesaurements:
[image]

2018-08-21 09-17 (76,22 kg and 13.1 body fat)
`0d:1f:14:02:00:5b:7b:bc:83:23:d8:a0:99:1d:c6:02:09:05:1e:1f`
1d:c6 = 76.22 Kg
05:1e = 13.1 body fat

naor88, 2020-11-29:
I'm also try install the dev build (openScale-1.8.1.apk and latest) and at both can't find the device..
@Danix58 What type off android phone and OS version you use?