oliexdev/openScale #318: Mi Scale2 BLE Packets log

jinwook31, 2018-08-21:
I want to reverse engineer Mi scale 2 packets. I think the current app only request weight, but is there a way I can get packet logs of other features? (eg. water, skeleton.. etc )

+ someone working on this part??

oliexdev, 2018-08-21:
no like in #71 only the weight and the impedance value is send by the scale. Unfortunately, the calculation for the body measurements is done in the app via the .so library `libBodyfat.so`
If you are good in understanding disassembled code you can give it a try.

jason-song-dev, 2018-08-23:
[image]

I found this! App shows 21.5%

jinwook31, 2018-08-23:
[image]
I found this! App shows 21.5%