oliexdev/openScale #319: Problems with Sanitas SBF70

cherocee, 2018-08-27:
Hello,
Many thanks for developing openscale. This development was the reason for my buyig the SBF70. At the time of my purchase I did not notice the issues with the SBF 70 - I was happy that is labelled as supported.

Since issue [252](https://github.com/oliexdev/openScale/issues/252) was closed I am starting here a new issue. I hope this is fine.

I am experiencing problems with the Sanitas SBF70. I am using Android 7.0 Kernel 3.18.31

After going throrugh the different issues in this depository, I started with version 1.7.1 - hoping this would work with the SBF70. 
I could select the scale in the bluetooth settings - and weight readings were retrieved from the scale. No user was created in the balance. No further data were downloaded from the balance.  I stepped on the scale three time with an interval of 1 minute in between.

Then I registered a user with the healthcoach app (after uninstalling openscale)
weight and further reading were  read out from the balance.

I could not obtain any reading with openscale 1.7.1.

Then I changed to the dev version dated 2018-08-20 19:26:52 UTC (obtained from https://github.com/oliexdev/openScale/releases). I could choose the sbf 70 from the bluetooth settings. , When try to retrieve any data, I could not get any readings either. Resetting the balanced did not help. 
When trying to retrieve data I did obtain in German something like "connection established", and than something like "connection lost"
I tried in the dev version to create a bt log (https://forums.lenovo.com/t5/Moto-G5-Moto-G5-Plus/Bluetooth-BLE-problem-on-moto-G5/m-p/3961770/highlight/true#M18512) and subsequently an error log of openscale.

Can this be fixed?
best
che
[btsnoop_hci.log](https://github.com/oliexdev/openScale/files/2323536/btsnoop_hci.log)
[openScale_2018-08-27_12-37.txt](https://github.com/oliexdev/openScale/files/2323537/openScale_2018-08-27_12-37.txt)

cherocee, 2018-08-29:
I just realized from one of my previous attempts this morning, when the "Fetching info, please wait while we fetch extended info from you scale"  was terminated by the screenlog I could connect to the scale. The respective log file of this attempt is provided here. sorry for forgetting this previously. I hope it is not too messy this way.
I should mention, that currently there is no user "registered" on the scale. previous attempts with the sanitad healthcoach app failed.
thanks for your efforts.
[openScale_2018-08-29_12-32.txt](https://github.com/oliexdev/openScale/files/2333716/openScale_2018-08-29_12-32.txt)

cherocee, 2018-08-29:
under:
https://www.amazon.de/Sanitas-Diagnosewaage-Vernetzung-Smartphone-K%C3%B6rperdaten/product-reviews/B00NABN36E
(sorry about the German)
I read that one customer calls himself "Kunde" was able to use this balance with openscale.
That post is date Sept 13, 2017.
I tried to use openScale 1,5,4. that was released in Aug 19, 2017.
But that attempt did not succeed either.
Might it be possible that Sanitas ch

erijo, 2018-08-30:
BLE on Android is, as far as I can tell, unfortunately a mess. Different phones needs different work-arounds to make it work.

Please try restarting your phone and changing the batteries in the scale to see if that makes any difference. You can also try pulling the power from the Samsung TV you have around.

If it still doesn't work, try clicking the bluetooth icon on the main screen. I have in some cases seen it disappear as on your screenshot, but it still there even if it isn't visible. Try clicking to the left of the "+" and see if that triggers any bluetooth communication.

If it still doesn't work, please install the original app and try to capture a btsnoop log when you do a measurement that is successfully transferred to the phone.

cherocee, 2018-08-30:
Many thanks for you reply and your suggestion.
I can not unplug the Samsung TV - it belongs to my neighbor....
I did as you suggested. After uninstalling the dev version of openscale, I restarted my phone, removed during the booting of the phone the battery from the scale and even reset the scale.
After installation of the latest dev version of open scale, I created a user, selected the sanitas sbf70. this time the bluetooth icon was visible. However, not data were read out and no user was created on the scale. 

I can not connect to the sbf70 with the original app anymore. I tried after restarting the phone and resetting the balance after battery removal. The scale is found and coupled, but I can not assign a user to this device. Accordingly no data are assigned to a registered user and not data is transferred to the phone. So there is no point to produce any brssnoop.
This impairment of the original seems to be the case ever since I fist connected to the sbf 70 for the first time with openscale. 
I feel the balance is bricked - I guess  I should return it ASAP.

erijo, 2018-08-31:
You can also go to the Bluetooth settings on your phone and remove the scale from the list of paired devices (if it's listed there).

If you have access to another phone you could try openScale on that and see if that works.

cherocee, 2018-08-31:
Well, the scale is not listed under the aired bluetooth devices. For the pairng of the phone and the scale  a key is required. There are three different floating around on different amazon sites, but from what I can tell none of them works.
My old phone does not support the bluetooth required by openscale or healthcoach. I am reluctant of asking friends to install any apps on their phone they do not need. And ultimately I would appreciate the data on the current phone and do not have to be concerned that anything will not work with the next phone.   
I went through the issue 252 and found your reply [https://github.com/oliexdev/openScale/issues/252#issuecomment-392289201](url)
to ReidarHH, He seems to be using a Moto G5. I do use a Moto G5 Plus. An a dev version provided on or briefly before May 25 worked for ReidarHH. Do you still have that dev version or could re-create it easyly from a git or snapshot of your development?

I feel it might be worth giving this version a trial.

ReidarHH, 2018-09-06:
Hi Cherokee.
I had a version that was working. But after an Android security update it stopped working. 
I think openscale is a great program.
But I kind of gave up on using because of android and phone developers keep changing things so openscale stops working. 
I'm not able to help with the software (lack of knowledge about this area). 
If the app gets an update and works I definitely use it again.

oliexdev, 2018-09-18:
it seems a Moto G5 specific issue. Do you have problems with other Bluetooth connections with your smartphone? Are you asked for granting Bluetooth permissions when you have a fresh installed openScale? Some users on the internet reports problem after securtiy updates on the Moto G5 that they messed up the permissions on the smartphone.

ReidarHH, 2018-09-18:
Hi.
Just to clarify. My first problem was that the and phone only stay paired for 15 seconds and no data was retrieved. 
Then Erijo fixed the problem. 
And after the security updated I was back to the first problem. 
I have no other Bluetooth connection problems. 
Openscale connects with out any problems, it says connection established, but after 15 seconds it says connection lost.

I will attach the log from openscale below.

2018-09-18 09:02:09.031 Debug [1] AboutPreferences: Debug log enabled, openScale (dev) v1.8.1-dev_ef1632dc_2018-09-17 (1537198711), SDK 24, motorola Moto G (5)
2018-09-18 09:02:29.271 Debug [1] BluetoothPreferences: Start discovery
2018-09-18 09:02:29.306 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.adapter.action.DISCOVERY_STARTED
2018-09-18 09:02:34.828 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-09-18 09:02:37.651 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-09-18 09:02:37.678 Debug [1] BluetoothPreferences: Found supported device SANITAS SBF70 [C8:FD:19:9D:0B:68] (driver: Sanitas SBF70/SilverCrest SBF75, type: 2)
2018-09-18 09:02:37.719 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-09-18 09:02:37.867 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-09-18 09:02:37.906 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-09-18 09:02:37.921 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-09-18 09:02:37.976 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-09-18 09:02:38.033 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-09-18 09:02:38.086 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-09-18 09:02:38.149 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-09-18 09:02:38.157 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-09-18 09:02:38.200 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-09-18 09:02:38.315 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-09-18 09:02:38.474 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-09-18 09:02:38.530 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-09-18 09:02:38.588 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-09-18 09:02:38.614 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-09-18 09:02:38.646 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-09-18 09:02:38.767 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-09-18 09:02:38.918 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-09-18 09:02:38.970 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-09-18 09:02:40.351 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-09-18 09:02:41.721 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-09-18 09:02:41.777 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-09-18 09:02:43.148 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-09-18 09:02:43.187 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-09-18 09:02:48.589 Debug [1] BluetoothPreferences: Discovery: android.bluetooth.adapter.action.DISCOVERY_FINISHED
2018-09-18 09:02:48.592 Debug [1] BluetoothPreferences: Start LE scan
2018-09-18 09:02:49.861 Debug [1] BluetoothPreferences: Stop discovery
2018-09-18 09:02:49.871 Debug [1] BluetoothPreferences: Stop LE scan
2018-09-18 09:02:49.933 Debug [1] OpenScale: Trying to connect to bluetooth device [C8:FD:19:9D:0B:68] in debug mode
2018-09-18 09:02:49.947 Debug [1] BluetoothCommunication: BT is enabled, state=12, scan mode=21, is not discovering
2018-09-18 09:02:49.963 Debug [1] BluetoothCommunication: Starting LE scan for device [C8:FD:19:9D:0B:68]
2018-09-18 09:02:50.076 Debug [1] BluetoothCommunication: Found LE device SANITAS SBF70 [C8:FD:19:9D:0B:68]
2018-09-18 09:02:50.077 Debug [1] BluetoothCommunication: Stopping LE scan
2018-09-18 09:02:50.083 Info [1] BluetoothCommunication: Connecting to [C8:FD:19:9D:0B:68] (driver: Debug)
2018-09-18 09:02:50.513 Debug [10150] BluetoothCommunication$GattCallback: onConnectionStateChange: status=0, newState=2
2018-09-18 09:03:11.334 Debug [10151] BluetoothCommunication$GattCallback: onConnectionStateChange: status=8, newState=0
2018-09-18 09:03:11.337 Debug [1] OpenScale: Disconnecting from bluetooth device
2018-09-18 09:03:11.337 Info [10151] BluetoothCommunication: Disconnecting
2018-09-18 09:03:21.905 Debug [1] OpenScale: Trying to connect to bluetooth device [C8:FD:19:9D:0B:68] (SANITAS SBF70)
2018-09-18 09:03:21.917 Debug [1] BluetoothCommunication: BT is enabled, state=12, scan mode=21, is not discovering
2018-09-18 09:03:21.930 Debug [1] BluetoothCommunication: Starting LE scan for device [C8:FD:19:9D:0B:68]
2018-09-18 09:03:22.139 Debug [1] BluetoothCommunication: Found LE device SANITAS SBF70 [C8:FD:19:9D:0B:68]
[... cut ...]
2018-09-18 09:03:49.597 Debug [10150] BluetoothCommunication$GattCallback: onConnectionStateChange: status=0, newState=2
2018-09-18 09:03:49.607 Debug [1] MainActivity: Bluetooth connection successful established
2018-09-18 09:04:10.418 Debug [10159] BluetoothCommunication$GattCallback: onConnectionStateChange: status=8, newState=0
2018-09-18 09:04:10.421 Info [10159] BluetoothCommunication: Disconnecting
2018-09-18 09:04:10.432 Debug [1] MainActivity: Bluetooth connection lost
2018-09-18 09:04:20.656 Debug [1] AboutPreferences: Debug log disabled

oliexdev, 2018-09-18:
I don't know why its always disconnected and as I don't own this scale I can't reproduce it.
Additionally, we didn't change anything particular in the Bluetooth communication. So, since it began after a security update it guess it has to do something with the Android security system or timing problems with Motorola smartphones!? For example an older model has some [Bluetooth issues](https://wccftech.com/bluetooth-motorola-fix-play-protect/) or see [here](https://forums.republicwireless.com/t/bluetooth-does-not-detect-devices-on-my-moto-g5-plus/17832). You find a quite of similar problem reports with the Motorola smartphone. I don't know if this is the reason or not.

P.S.: please attach next time the log file on GitHub or paste the log file with code tags. Thanks in advance.

ReidarHH, 2018-09-18:
Maybe it's only a Motorola issue. I have tried the above fixes but still the same issue.
I don't have any other Bluetooth problem. 
Thank you for trying anyway.

oliexdev, 2018-09-24:
Please create a `btsnoop_hci.log` with the original app, see [step 1 in the wiki](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale) for how to do it.

Alejandro131, 2018-10-04:
Hello,

First I would like to thank you for the work you have done on this project and to note that for me it was the reason to buy SBF70 as well.

I'm using a Nexus 4 with android 8.1.0 - LineageOS version 15.1

I tried to do step 1 from the wiki, but since my only way to access the btsnoop_hci.log is through generating a bug report with `adb bugreport bugreport.zip` I can't seem to create 3 clean files so that it might be easier to look at them. I managed to get 1 file which includes the initial communication (from a reset scale) with the original app which creates a user, weighs it and returns the data, but the might be some trash in it.

What I gathered from the decimal site is somewhat inconclusive and I couldn't find the true date with the DCode tool since I wanted to make your job a little bit easier. This is a particular section that stood out to me with possible scale data:

```
0000   02 03 20 16 00 12 00 04 00 1b 22 00 e7 42 02 02   .. .......".çB..
0010   c4 01 4c 00 e4 05 e3 06 c5 00 ef                  Ä.L.ä.ã.Å.ï
```

The data that is used is:
Username - ALE (dunno if it is lowercase or uppercase)
gender - female
height - 170 cm
birth date - 4 oct 1993
activity level - probably 1 (but I didn't see if it was 3 since that was the default one on the app)
weight - 69.2 kg - which I could find as `b4 02` in 3 packets in the file
bmi - 23.9 - which I found as `ef` at the end of the packet
body fat - 29.4 - `26 01` which I couldn't find
water - 45.2 - which I found as `c4 01` in the upper packet
muscle - 33.2 - `4c 01` which I couldn't find, but found `4c 00` in the packet above
bone - 25.1 - `fb` which I couldn't find as well
bmr - 1,507 cal
amr - 1,733 kcal
supposed true date of weighing - 10.04.2018. 02:29 pm utc +2:00

Hopefully this helps you resolve the issue and I can try to supply more data if needed, but can't guarantee that it would be a clean file.

[btsnoop_hci.log](https://github.com/oliexdev/openScale/files/2446380/btsnoop_hci.log)

erijo, 2018-10-07:
Can some of you please try the steps in #335?

kuleszdl, 2018-10-09:
Hi folks,

first of all - many thanks for creating this great app!

I bought a SBF70 as well and wanted to report my experience with it.

My setup:

- Nexus 5X with self-compiled AOSP 8.1, no gapps/Playstore
- I did not install the vendor app at all.
- I resetted the scale before use.
- Pairing via Android Bluetooth was asking for a PIN (0000 and 1234 not working), but pairing in Openscale was successful.

Issues discovered so far:

- I can only see weigth, BMI and BMR in Openscale, the other values are all zero.
- I have also seen the issue of the bluetooth icon disappearing as reported in this thread.
- My 4-digit user name does not show up on the scale.
- The measurements only get transfered when I measure them "directly" with the app running. I have the impression that measurements stored on the scale are not being synced/transmitted.

Olangu, 2018-11-24:
I'm  still facing this issue even in the dev version.

First it seem like we have a off by one issue in month. My user always has February as birthday, but scale says month 1.

Tried to change my openscale user but it still won't fetch the saved weights.

I'll attach one log from ordinary version and one from dev.

Olangu, 2018-11-24:
’
2018-11-24 13:02:49.598 Debug [2] AboutPreferences: Debug log enabled, openScale v1.8.3 (31), SDK 27, samsung SM-G960F
2018-11-24 13:02:50.109 Debug [2] BluetoothCommunication: Device not found in LE scan, connecting directly
2018-11-24 13:02:50.110 Info [2] BluetoothCommunication: Connecting to [50:8C:B1:57:E5:16] (driver: Sanitas SBF70/SilverCrest SBF75)
2018-11-24 13:02:54.147 Debug [558] BluetoothCommunication$GattCallback: onConnectionStateChange: status=0, newState=2
2018-11-24 13:02:54.149 Debug [2] BluetoothCommunication: Stopping LE scan
2018-11-24 13:02:54.171 Debug [2] MainActivity: Bluetooth connection successful established
2018-11-24 13:02:55.161 Debug [558] BluetoothCommunication$GattCallback: onServicesDiscovered: status=0 (5 services)
2018-11-24 13:02:56.166 Debug [2] BluetoothCommunication: INIT STATE: 0
2018-11-24 13:02:56.167 Debug [2] BluetoothCommunication: Set notification on for 0000ffe1-0000-1000-8000-00805f9b34fb
2018-11-24 13:02:56.170 Debug [2] BluetoothCommunication: Write descriptor 00002902-0000-1000-8000-00805f9b34fb: 01 00 (queue: 0 0)
2018-11-24 13:02:56.307 Debug [2] BluetoothCommunication: INIT STATE: 1
2018-11-24 13:02:56.311 Debug [2] BluetoothCommunication: Write characteristic 0000ffe1-0000-1000-8000-00805f9b34fb: E6 01 (type: 2; queue: 0 0)
2018-11-24 13:02:56.391 Debug [558] BluetoothCommunication$GattCallback: onCharacteristicChanged 0000ffe1-0000-1000-8000-00805f9b34fb: E6 00 20
2018-11-24 13:02:56.393 Debug [2] BluetoothBeurerSanitas: ACK Scale is ready
2018-11-24 13:02:56.396 Debug [2] BluetoothCommunication: INIT STATE: 2
2018-11-24 13:02:56.451 Debug [2] BluetoothCommunication: INIT STATE: 3
2018-11-24 13:02:56.453 Debug [2] BluetoothBeurerSanitas: Write new Date/Time: 1543060976 (5B F9 3D F0)
2018-11-24 13:02:56.455 Debug [2] BluetoothCommunication: Write characteristic 0000ffe1-0000-1000-8000-00805f9b34fb: E9 5B F9 3D F0 (type: 2; queue: 0 0)
2018-11-24 13:02:56.598 Debug [2] BluetoothCommunication: INIT STATE: 4
2018-11-24 13:02:56.600 Debug [2] BluetoothBeurerSanitas: Setting unit kg
2018-11-24 13:02:56.603 Debug [2] BluetoothCommunication: Write characteristic 0000ffe1-0000-1000-8000-00805f9b34fb: E7 4D 01 (type: 2; queue: 0 0)
2018-11-24 13:02:56.745 Debug [2] BluetoothCommunication: INIT STATE: 5
2018-11-24 13:02:57.074 Debug [558] BluetoothCommunication$GattCallback: onCharacteristicChanged 0000ffe1-0000-1000-8000-00805f9b34fb: E7 F0 4D 00
2018-11-24 13:02:57.075 Debug [2] BluetoothBeurerSanitas: ACK Unit set
2018-11-24 13:02:57.076 Debug [2] BluetoothCommunication: INIT STATE: 6
2018-11-24 13:02:57.077 Debug [2] BluetoothCommunication: Write characteristic 0000ffe1-0000-1000-8000-00805f9b34fb: E7 33 (type: 2; queue: 0 0)
2018-11-24 13:02:57.233 Debug [2] BluetoothCommunication: INIT STATE: 7
2018-11-24 13:02:57.367 Debug [558] BluetoothCommunication$GattCallback: onCharacteristicChanged 0000ffe1-0000-1000-8000-00805f9b34fb: E7 F0 33 00 02 08
2018-11-24 13:02:57.368 Debug [2] BluetoothBeurerSanitas: ACK Got general user information
2018-11-24 13:02:57.369 Debug [2] BluetoothBeurerSanitas: Count: 2, maxUsers: 8
2018-11-24 13:02:57.369 Debug [2] BluetoothCommunication: INIT STATE: 7
2018-11-24 13:02:57.416 Debug [558] BluetoothCommunication$GattCallback: onCharacteristicChanged 0000ffe1-0000-1000-8000-00805f9b34fb: E7 34 02 01 00 00 00 00 00 12 91 12 4C 49 4E 5B
2018-11-24 13:02:57.418 Debug [2] BluetoothBeurerSanitas: Ack Get UUIDSs List of Users
2018-11-24 13:02:57.418 Debug [2] BluetoothBeurerSanitas: Send ack gotUser
2018-11-24 13:02:57.419 Debug [2] BluetoothCommunication: Write characteristic 0000ffe1-0000-1000-8000-00805f9b34fb: E7 F1 34 02 01 (type: 2; queue: 0 0)
2018-11-24 13:02:57.562 Debug [558] BluetoothCommunication$GattCallback: onCharacteristicChanged 0000ffe1-0000-1000-8000-00805f9b34fb: E7 34 02 02 00 00 00 00 00 00 00 13 41 4E 44 52
2018-11-24 13:02:57.564 Debug [2] BluetoothBeurerSanitas: Ack Get UUIDSs List of Users
2018-11-24 13:02:57.564 Debug [2] BluetoothBeurerSanitas: Send ack gotUser
2018-11-24 13:02:57.565 Debug [2] BluetoothCommunication: Request pending (queue 0 1)
2018-11-24 13:02:57.573 Debug [2] BluetoothCommunication: Write characteristic 0000ffe1-0000-1000-8000-00805f9b34fb: E7 F1 34 02 02 (type: 2; queue: 0 0)
2018-11-24 13:02:57.720 Debug [2] BluetoothCommunication: INIT STATE: 7
2018-11-24 13:02:57.721 Debug [2] BluetoothBeurerSanitas: Request getUserInfo 19
2018-11-24 13:02:57.722 Debug [2] BluetoothCommunication: Write characteristic 0000ffe1-0000-1000-8000-00805f9b34fb: E7 36 00 00 00 00 00 00 00 13 (type: 2; queue: 0 0)
2018-11-24 13:02:57.724 Debug [2] BluetoothBeurerSanitas: scaleuserid: 19, registered users: 2, extracted users: 2
2018-11-24 13:02:57.855 Debug [558] BluetoothCommunication$GattCallback: onCharacteristicChanged 0000ffe1-0000-1000-8000-00805f9b34fb: E7 F0 36 00 41 4E 44 52 01 1A B7 82
2018-11-24 13:02:57.856 Debug [2] BluetoothBeurerSanitas: Ack Get User Info Initials
2018-11-24 13:02:57.857 Debug [2] BluetoothBeurerSanitas: Name: AND, YY-MM-DD: 82-1-26, Height: 183, Sex: male, activity: 2
2018-11-24 13:02:57.858 Debug [2] BluetoothCommunication: Request pending (queue 0 1)
2018-11-24 13:02:57.867 Debug [2] BluetoothCommunication: Write characteristic 0000ffe1-0000-1000-8000-00805f9b34fb: E7 4F 00 00 00 00 00 00 00 13 (type: 2; queue: 0 0)
[... cut ...]
2018-11-24 13:02:58.002 Debug [2] BluetoothBeurerSanitas: Ack Get scale status
2018-11-24 13:02:58.004 Debug [2] BluetoothBeurerSanitas: BatteryLevel: 100, weightThreshold: 2,00, BodyFatThreshold: 22,00, Unit: 1, userExists: true, UserReference Weight Exists: true, UserMeasurementExists: true, scaleVersion: 5
2018-11-24 13:02:58.012 Debug [2] BluetoothCommunication: INIT STATE: 8
2018-11-24 13:03:44.993 Debug [2] OpenScale: Select user 1
2018-11-24 13:04:57.792 Debug [2] AboutPreferences: Debug log disabled
’

Olangu, 2018-11-24:
2018-11-24 13:27:59.578 Debug [2] AboutPreferences: Debug log enabled, openScale (dev) v1.8.3-dev_4d09351f_2018-11-21 (1542794810), SDK 27, samsung SM-G960F
2018-11-24 13:28:04.301 Debug [2] BluetoothPreferences: Start discovery
2018-11-24 13:28:04.367 Debug [2] BluetoothPreferences: Discovery: android.bluetooth.adapter.action.DISCOVERY_STARTED
2018-11-24 13:28:11.010 Debug [2] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-11-24 13:28:11.059 Debug [2] BluetoothPreferences: Found supported device SANITAS SBF70 [50:8C:B1:57:E5:16] (driver: Sanitas SBF70/SilverCrest SBF75, type: 2)
2018-11-24 13:28:13.134 Debug [2] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-11-24 13:28:17.147 Debug [2] BluetoothPreferences: Discovery: android.bluetooth.adapter.action.DISCOVERY_FINISHED
2018-11-24 13:28:17.153 Debug [2] BluetoothPreferences: Start LE scan
2018-11-24 13:28:19.567 Debug [2] BluetoothPreferences: Stop discovery
2018-11-24 13:28:19.571 Debug [2] BluetoothPreferences: Stop LE scan
2018-11-24 13:28:19.622 Debug [2] OpenScale: Trying to connect to bluetooth device [50:8C:B1:57:E5:16] in debug mode
2018-11-24 13:28:19.628 Debug [2] BluetoothCommunication: BT is enabled, state=12, scan mode=21, is not discovering
2018-11-24 13:28:19.633 Debug [2] BluetoothCommunication: Starting LE scan for device [50:8C:B1:57:E5:16]
2018-11-24 13:28:29.643 Debug [2] BluetoothCommunication: Device not found in LE scan, connecting directly
2018-11-24 13:28:29.650 Info [2] BluetoothCommunication: Connecting to [50:8C:B1:57:E5:16] (driver: Debug)
2018-11-24 13:28:31.478 Debug [643] BluetoothCommunication$GattCallback: onConnectionStateChange: status=0, newState=2
2018-11-24 13:28:31.486 Debug [2] BluetoothCommunication: Stopping LE scan
2018-11-24 13:28:32.516 Debug [643] BluetoothCommunication$GattCallback: onServicesDiscovered: status=0 (5 services)
2018-11-24 13:28:33.528 Debug [2] BluetoothCommunication: INIT STATE: 0
2018-11-24 13:28:33.539 Debug [2] BluetoothCommunication: Read characteristic 0x2a00 "device name"
2018-11-24 13:28:33.639 Debug [643] BluetoothCommunication$GattCallback: onCharacteristicRead 0x2a00 "device name" (status=0): 53 41 4E 49 54 41 53 20 53 42 46 37 30
2018-11-24 13:28:33.708 Debug [2] BluetoothCommunication: INIT STATE: 1
2018-11-24 13:28:33.719 Debug [2] BluetoothCommunication: Read characteristic 0x2a01 "appearance"
2018-11-24 13:28:33.828 Debug [643] BluetoothCommunication$GattCallback: onCharacteristicRead 0x2a01 "appearance" (status=0): 00 00
2018-11-24 13:28:33.898 Debug [2] BluetoothCommunication: INIT STATE: 2
2018-11-24 13:28:33.910 Debug [2] BluetoothCommunication: Read characteristic 0x2a02 "peripheral privacy flag"
2018-11-24 13:28:34.023 Debug [643] BluetoothCommunication$GattCallback: onCharacteristicRead 0x2a02 "peripheral privacy flag" (status=0): 00
2018-11-24 13:28:34.092 Debug [2] BluetoothCommunication: INIT STATE: 3
2018-11-24 13:28:34.107 Debug [2] BluetoothCommunication: Read characteristic 0x2a04 "peripheral preferred connection parameters"
2018-11-24 13:28:34.220 Debug [643] BluetoothCommunication$GattCallback: onCharacteristicRead 0x2a04 "peripheral preferred connection parameters" (status=0): 50 00 A0 00 00 00 E8 03
2018-11-24 13:28:34.289 Debug [2] BluetoothCommunication: INIT STATE: 4
2018-11-24 13:28:34.302 Debug [2] BluetoothCommunication: Read descriptor 0x2902 "client characteristic configuration"
2018-11-24 13:28:34.415 Debug [643] BluetoothCommunication$GattCallback: onDescriptorRead 0x2902 "client characteristic configuration" (status=0): 00 00
2018-11-24 13:28:34.484 Debug [2] BluetoothCommunication: INIT STATE: 5
2018-11-24 13:28:34.499 Debug [2] BluetoothCommunication: Read characteristic 0x2a23 "system id"
2018-11-24 13:28:34.611 Debug [643] BluetoothCommunication$GattCallback: onCharacteristicRead 0x2a23 "system id" (status=0): 16 E5 57 00 00 B1 8C 50
2018-11-24 13:28:34.679 Debug [2] BluetoothCommunication: INIT STATE: 6
2018-11-24 13:28:34.691 Debug [2] BluetoothCommunication: Read characteristic 0x2a24 "model number string"
2018-11-24 13:28:34.807 Debug [643] BluetoothCommunication$GattCallback: onCharacteristicRead 0x2a24 "model number string" (status=0): 4D 6F 64 65 6C 20 4E 75 6D 62 65 72
2018-11-24 13:28:34.874 Debug [2] BluetoothCommunication: INIT STATE: 7
2018-11-24 13:28:34.891 Debug [2] BluetoothCommunication: Read characteristic 0x2a25 "serial number string"
2018-11-24 13:28:35.002 Debug [643] BluetoothCommunication$GattCallback: onCharacteristicRead 0x2a25 "serial number string" (status=0): 53 65 72 69 61 6C 20 4E 75 6D 62 65 72
2018-11-24 13:28:35.069 Debug [2] BluetoothCommunication: INIT STATE: 8
2018-11-24 13:28:35.083 Debug [2] BluetoothCommunication: Read characteristic 0x2a26 "firmware revision string"
2018-11-24 13:28:35.198 Debug [643] BluetoothCommunication$GattCallback: onCharacteristicRead 0x2a26 "firmware revision string" (status=0): 46 69 72 6D 77 61 72 65 20 52 65 76 69 73 69 6F 6E
2018-11-24 13:28:35.266 Debug [2] BluetoothCommunication: INIT STATE: 9
2018-11-24 13:28:35.283 Debug [2] BluetoothCommunication: Read characteristic 0x2a27 "hardware revision string"
2018-11-24 13:28:35.393 Debug [643] BluetoothCommunication$GattCallback: onCharacteristicRead 0x2a27 "hardware revision string" (status=0): 48 61 72 64 77 61 72 65 20 52 65 76 69 73 69 6F 6E
2018-11-24 13:28:35.460 Debug [2] BluetoothCommunication: INIT STATE: 10
[... cut ...]
2018-11-24 13:29:57.822 Debug [643] BluetoothCommunication$GattCallback: onCharacteristicChanged 0xffe1: E7 F0 4F 00 64 14 DC 01 00 00 00 05
2018-11-24 13:29:57.824 Debug [2] BluetoothBeurerSanitas: Ack Get scale status
2018-11-24 13:29:57.825 Debug [2] BluetoothBeurerSanitas: BatteryLevel: 100, weightThreshold: 2,00, BodyFatThreshold: 22,00, Unit: 1, userExists: true, UserReference Weight Exists: true, UserMeasurementExists: true, scaleVersion: 5
2018-11-24 13:29:57.835 Debug [2] BluetoothCommunication: INIT STATE: 8
2018-11-24 13:30:05.900 Debug [2] AboutPreferences: Debug log disabled

erijo, 2018-11-24:
I have some ideas about this. Once I have something to show I'll ask you to test it.

erijo, 2018-11-25:
@Olangu: please try [this dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev_beurer-build) and attach the log (i.e. please don't paste it in the comment field).

Olangu, 2018-11-26:
> @Olangu: please try [this dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev_beurer-build) and attach the log (i.e. please don't paste it in the comment field).

Yes, this version does work.
I can now see both users stored in the scale in the log.

Yeah, sorry for pasting the log like that. But I'm yet to find a way to upload files from the phone.
I'll attach the new log when I get to a computer.

erijo, 2018-11-26:
You can paste the log here https://gist.github.com/ and post a link.

Olangu, 2018-11-26:
[openScale_2018-11-26_08-33[1].txt](https://github.com/oliexdev/openScale/files/2616572/openScale_2018-11-26_08-33.1.txt)

erijo, 2018-11-26:
Thanks @Olangu! Looks like expected, and glad to hear that it worked.

Please try it a few more times and report back if you encounter any issues/strange things. I will do some more testing myself before merging it back to master.

Olangu, 2018-11-27:
> Thanks @Olangu! Looks like expected, and glad to hear that it worked.
> 
> Please try it a few more times and report back if you encounter any issues/strange things. I will do some more testing myself before merging it back to master.

The dev version won't connect in low energy mode, even if the scale is on. It always fall back to device scanning. The stable version connects right away.

erijo, 2018-11-27:
It's not that you haven't granted the stable version the location permission? If that is missing, openScale will try to connect without scanning first (which works for some scales but not all).