oliexdev/openScale #324: Xiaomi Mi scale v2

itge13, 2018-09-17:
Please make the rest of the data also available

oliexdev, 2018-09-18:
see issue #150

closed 2018-09-18 by oliexdev

itge13, 2018-09-18:
So no chance in near future? Is there a way to copy the data from the Mi Fit app?

oliexdev, 2018-09-18:
sure there is a chance but it takes some time. You could enter manually the measurement from the mi fit app :wink: no seriously you can't copy the data from the mi fit app or do they have an export function? Then maybe you could import it via the CSV import by openScale.

itge13, 2018-09-18:
Nope, no export function but the data is stored anywhere on the phone (shows the data even in flight mode) so there is maybe a way.

oliexdev, 2018-09-18:
yeah in the database but you don't have access to another database if the app doesn't allow it. Would be a security nightmare if this would be general possible.