oliexdev/openScale #325: Import/sync with Sanitas SBF70 does not work

studioart, 2018-09-20:
Hi,

I searched the FAQ and the issues as well as the App in Playstore - but I haven't found a solution to import the data from the Sanitas SBF70 scale. 

- I have first created a user in openScale, then with the same name a user in the vendor app (HealthCoach).

- When starting openScale a message appears "connected to SBF70", so far, so good.

But how to get the measured data in openScale automatically please?
It works fine with the vendor app, but I would prefer using openScale for a better overview over all the data.

Thanks.

oliexdev, 2018-09-21:
Maybe delete all users and then create the user with the vendors app, do a measurement, create a user with the same in openScale and try to do a measurement.

Please use code tags for the log file or attach it.

oliexdev, 2018-09-24:
Please create a `btsnoop_hci.log` with the original app, see [step 1 in the wiki](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale) for how to do it

erijo, 2018-10-06:
Can you try the [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) and attach log from that?