oliexdev/openScale #327: SBF 70 not working with Android Oreo

landrea2, 2018-09-24:
Hi all,
I bought SBF 70 this morning at Lidl. I was very curious to see how it worked.
Unfortunately, I can't pair it with Openscale. I'm on Honor 6x, Android 8.0, openkirin rom (Lineage OS).
I must say the scale doesn't work with vendor app neither, so the issue may be bigger than Openscale.
Thank you in advance, i'm ready to provide logs but please tell me how to do it :)

oliexdev, 2018-09-24:
see issue #325 and #319

closed 2018-09-24 by oliexdev

erijo, 2018-10-06:
FYI: you can provide logs by going to openScale settings and then under about til you'll find an option to enable debug log. Preferably use the [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) to do so.