oliexdev/openScale #330: Support for Android Pie and Xiaomi Mi Scale

diamant-x, 2018-10-02:
After upgrading my OnePlus 6 to Android Pie, the app is no longer receiving measures from a Xiaomi Mi scale.
The Bluetooth icon appears as connected but the records don't get loaded.
The same was working before the upgrade.
I tried deleting app data and cache for a fresh install and same issues persist.

Thanks.

erijo, 2018-10-02:
Can you please test the [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) and see how it works? If it doesn't, please ensur enable log under settings, about and attach it here.

diamant-x, 2018-10-02:
Just installed.
Old measures aren't being retrieved. Will try new measures later.
When exiting the app though it crashes with following trace:

Build version: 1.8.1-dev_8f6c7c4c_2018-10-02 
Current date: 2018-10-02 23:11:35 
Device: OnePlus ONEPLUS A6003 
 
Stack trace:  
java.lang.RuntimeException: Unable to destroy activity {com.health.openscale.dev/com.health.openscale.gui.MainActivity}: java.lang.NullPointerException: Attempt to invoke virtual method 'android.bluetooth.BluetoothGattCharacteristic android.bluetooth.BluetoothGattService.getCharacteristic(java.util.UUID)' on a null object reference
	at android.app.ActivityThread.performDestroyActivity(ActivityThread.java:4594)
	at android.app.ActivityThread.handleDestroyActivity(ActivityThread.java:4612)
	at android.app.servertransaction.DestroyActivityItem.execute(DestroyActivityItem.java:39)
	at android.app.servertransaction.TransactionExecutor.executeLifecycleState(TransactionExecutor.java:145)
	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:70)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1906)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loop(Looper.java:193)
	at android.app.ActivityThread.main(ActivityThread.java:6863)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:537)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)
Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'android.bluetooth.BluetoothGattCharacteristic android.bluetooth.BluetoothGattService.getCharacteristic(java.util.UUID)' on a null object reference
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:260)
	at com.health.openscale.core.bluetooth.BluetoothMiScale.nextCleanUpCmd(BluetoothMiScale.java:183)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStateStep(BluetoothCommunication.java:577)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.handleRequests(BluetoothCommunication.java:625)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.setBtMachineState(BluetoothCommunication.java:245)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.disconnect(BluetoothCommunication.java:533)
	at com.health.openscale.core.OpenScale.disconnectFromBluetoothDevice(OpenScale.java:576)
	at com.health.openscale.gui.MainActivity.onDestroy(MainActivity.java:191)
	at android.app.Activity.performDestroy(Activity.java:7417)
	at android.app.Instrumentation.callActivityOnDestroy(Instrumentation.java:1323)
	at android.app.ActivityThread.performDestroyActivity(ActivityThread.java:4579)
	... 11 more

User actions: 
2018-10-02 23:10:01: MainActivity resumed
2018-10-02 23:10:01: MainActivity paused
2018-10-02 23:10:01: UserSettingsActivity created
2018-10-02 23:10:01: UserSettingsActivity resumed
2018-10-02 23:10:16: UserSettingsActivity paused
2018-10-02 23:10:16: MainActivity resumed
2018-10-02 23:10:16: UserSettingsActivity destroyed
2018-10-02 23:10:20: MainActivity paused
2018-10-02 23:10:34: MainActivity resumed
2018-10-02 23:10:37: MainActivity paused
2018-10-02 23:10:37: SettingsActivity created
2018-10-02 23:10:37: SettingsActivity resumed
2018-10-02 23:10:39: MainActivity destroyed
2018-10-02 23:10:39: MainActivity created
2018-10-02 23:10:39: MainActivity resumed
2018-10-02 23:10:39: MainActivity paused
2018-10-02 23:10:39: MainActivity destroyed
2018-10-02 23:10:39: MainActivity created
2018-10-02 23:10:39: MainActivity resumed
2018-10-02 23:10:39: MainActivity paused
2018-10-02 23:10:39: MainActivity destroyed
2018-10-02 23:10:39: MainActivity created
2018-10-02 23:10:39: MainActivity resumed
2018-10-02 23:10:39: MainActivity paused
2018-10-02 23:10:39: MainActivity destroyed
2018-10-02 23:10:39: MainActivity created
2018-10-02 23:10:39: MainActivity resumed
2018-10-02 23:10:39: MainActivity paused
2018-10-02 23:10:42: SettingsActivity paused
2018-10-02 23:10:43: SettingsActivity resumed
2018-10-02 23:11:02: SettingsActivity paused
2018-10-02 23:11:02: MainActivity resumed
2018-10-02 23:11:02: SettingsActivity destroyed
2018-10-02 23:11:24: MainActivity paused

diamant-x, 2018-10-03:
Nothing.
Dev version is not retrieving new measures either.
Attaching log.
After connecting I stepped on the scale and got a new measure.
After checking it wasn't appearing in the app I exited it and attaching next.

2018-10-03 10:33:30.781 Debug [2] AboutPreferences: Debug log enabled, openScale (dev) v1.8.1-dev_8f6c7c4c_2018-10-02 (1538465646), SDK 28, OnePlus ONEPLUS A6003
2018-10-03 10:33:42.013 Debug [2] OpenScale: Disconnecting from bluetooth device
2018-10-03 10:33:43.696 Debug [2] OpenScale: Trying to connect to bluetooth device [88:0F:10:A8:58:17] (MI_SCALE)
2018-10-03 10:33:43.706 Debug [2] BluetoothCommunication: BT is enabled, state=12, scan mode=21, is not discovering
2018-10-03 10:33:43.715 Debug [2] BluetoothCommunication: Starting LE scan for device [88:0F:10:A8:58:17]
2018-10-03 10:33:43.833 Debug [2] BluetoothCommunication: Found LE device [TV] Samsung 6 Series (40) [7C:64:56:18:66:F2]
2018-10-03 10:33:43.983 Debug [2] BluetoothCommunication: Found LE device null [63:BE:56:3A:F0:DA]
2018-10-03 10:33:43.990 Debug [2] BluetoothCommunication: Found LE device [TV] Samsung 6 Series (40) [7C:64:56:18:66:F2]
2018-10-03 10:33:44.042 Debug [2] BluetoothCommunication: Found LE device null [3F:C6:99:52:B4:44]
2018-10-03 10:33:44.251 Debug [2] BluetoothCommunication: Found LE device null [3F:C6:99:52:B4:44]
2018-10-03 10:33:44.262 Debug [2] BluetoothCommunication: Found LE device null [63:BE:56:3A:F0:DA]
2018-10-03 10:33:44.355 Debug [2] BluetoothCommunication: Found LE device null [3F:C6:99:52:B4:44]
2018-10-03 10:33:44.454 Debug [2] BluetoothCommunication: Found LE device [TV] Samsung 6 Series (40) [7C:64:56:18:66:F2]
2018-10-03 10:33:44.520 Debug [2] BluetoothCommunication: Found LE device null [B8:78:2E:42:B9:23]
2018-10-03 10:33:44.530 Debug [2] BluetoothCommunication: Found LE device null [63:BE:56:3A:F0:DA]
2018-10-03 10:33:44.557 Debug [2] BluetoothCommunication: Found LE device null [3F:C6:99:52:B4:44]
2018-10-03 10:33:44.569 Debug [2] BluetoothCommunication: Found LE device null [7C:9E:EB:99:A5:3E]
2018-10-03 10:33:44.636 Debug [2] BluetoothCommunication: Found LE device null [64:81:7E:2E:50:68]
2018-10-03 10:33:44.660 Debug [2] BluetoothCommunication: Found LE device null [79:A1:8D:B5:A9:DD]
2018-10-03 10:33:44.684 Debug [2] BluetoothCommunication: Found LE device null [4C:BC:16:94:2B:4F]
2018-10-03 10:33:44.701 Debug [2] BluetoothCommunication: Found LE device null [B8:78:2E:42:B9:23]
2018-10-03 10:33:44.859 Debug [2] BluetoothCommunication: Found LE device null [04:69:F8:AB:1F:A3]
2018-10-03 10:33:44.886 Debug [2] BluetoothCommunication: Found LE device null [B8:78:2E:42:B9:23]
2018-10-03 10:33:44.918 Debug [2] BluetoothCommunication: Found LE device null [79:A1:8D:B5:A9:DD]
2018-10-03 10:33:45.008 Debug [2] BluetoothCommunication: Found LE device MI_SCALE [88:0F:10:A8:58:17]
2018-10-03 10:33:45.014 Debug [2] BluetoothCommunication: Stopping LE scan
2018-10-03 10:33:45.026 Info [2] BluetoothCommunication: Connecting to [88:0F:10:A8:58:17] (driver: Xiaomi Mi Scale v1)
2018-10-03 10:33:47.241 Debug [2392] BluetoothCommunication$GattCallback: onConnectionStateChange: status=0, newState=2
2018-10-03 10:33:47.263 Debug [2] MainActivity: Bluetooth connection successful established
2018-10-03 10:33:48.255 Debug [2392] BluetoothCommunication$GattCallback: onServicesDiscovered: status=0 (0 services)
2018-10-03 10:33:49.264 Debug [2392] BluetoothCommunication: INIT STATE: 0
2018-10-03 10:34:17.130 Debug [2388] BluetoothCommunication$GattCallback: onConnectionStateChange: status=19, newState=0
2018-10-03 10:34:17.163 Info [2388] BluetoothCommunication: Disconnecting
2018-10-03 10:34:17.163 Debug [2] MainActivity: Bluetooth connection lost
2018-10-03 10:34:19.956 Debug [2] OpenScale: Disconnecting from bluetooth device

erijo, 2018-10-06:
Can you please get a new log with the latest [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build)?

oliexdev, 2018-10-28:
please try out the [ble-main-dev](https://github.com/oliexdev/openScale/releases/tag/travis-dev_ble-main-build) version

diamant-x, 2018-10-28:
Still not working. 
Same errors.

2018-10-28 10:02:35.893 Debug [2] AboutPreferences: Debug log enabled, openScale (dev_ble-main) v1.8.1-dev_4ace7319_2018-10-15 (1539634634), SDK 28, OnePlus ONEPLUS A6003
2018-10-28 10:02:40.618 Debug [2] OpenScale: Disconnecting from bluetooth device
2018-10-28 10:02:43.122 Debug [2] OpenScale: Trying to connect to bluetooth device [88:0F:10:A8:58:17] (MI_SCALE)
2018-10-28 10:02:43.132 Debug [2] BluetoothCommunication: BT is enabled, state=12, scan mode=21, is not discovering
2018-10-28 10:02:43.139 Debug [2] BluetoothCommunication: Starting LE scan for device [#Removed for confidentiality]
2018-10-28 10:02:44.072 Info [2] BluetoothCommunication: Connecting to [#removed for confidentiality] (driver: Xiaomi Mi Scale v1)

2018-10-28 10:02:45.287 Debug [2540] BluetoothCommunication$GattCallback: onConnectionStateChange: status=0, newState=2
2018-10-28 10:02:45.296 Debug [2] BluetoothCommunication: Stopping LE scan
2018-10-28 10:02:45.321 Debug [2] MainActivity: Bluetooth connection successful established
2018-10-28 10:02:46.308 Debug [2540] BluetoothCommunication$GattCallback: onServicesDiscovered: status=0 (0 services)
2018-10-28 10:02:46.315 Info [2] BluetoothCommunication: Disconnecting
2018-10-28 10:02:46.343 Error [2] MainActivity: Bluetooth unexpected error: No services found
2018-10-28 10:02:51.969 Debug [2] OpenScale: Disconnecting from bluetooth device

erijo, 2018-10-30:
I'm missing lines in that log that I would expect to see. Can you please try with the latest [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) (ble-main was merged to that yesterday)?

diamant-x, 2018-10-30:
> I'm missing lines in that log that I would expect to see. Can you please try with the latest [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) (ble-main was merged to that yesterday)?

Here the log. I removed references to other BLE devices found & MAC ofuscated.
I tried several times to resync with the balance with the same error as the beginning.

```
2018-10-30 19:57:50.182 Debug [2] AboutPreferences: Debug log enabled, openScale (dev) v1.8.2-dev_f81c25fb_2018-10-29 (1540846175), SDK 28, OnePlus ONEPLUS A6003
2018-10-30 19:58:05.101 Debug [2] BluetoothPreferences: Start discovery
2018-10-30 19:58:05.150 Debug [2] BluetoothPreferences: Discovery: android.bluetooth.adapter.action.DISCOVERY_STARTED
2018-10-30 19:58:05.200 Debug [2] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-10-30 19:58:05.205 Debug [2] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-10-30 19:58:05.302 Debug [2] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-10-30 19:58:06.794 Debug [2] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-10-30 19:58:06.798 Debug [2] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-10-30 19:58:06.857 Debug [2] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-10-30 19:58:06.956 Debug [2] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-10-30 19:58:06.968 Debug [2] BluetoothPreferences: Found unsupported device RE Camera [20:OFUSCATED:E2] (type: 2)
2018-10-30 19:58:09.628 Debug [2] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-10-30 19:58:09.674 Debug [2] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-10-30 19:58:09.693 Debug [2] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-10-30 19:58:09.703 Debug [2] BluetoothPreferences: Found unsupported device RECAM [20:OFUSCATED:FD] (type: 2)
2018-10-30 19:58:09.858 Debug [2] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-10-30 19:58:11.211 Debug [2] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-10-30 19:58:11.232 Debug [2] BluetoothPreferences: Found supported device MI_SCALE [88:OFUSCATED:17] (driver: Xiaomi Mi Scale v1, type: 2)
2018-10-30 19:58:13.034 Debug [2] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-10-30 19:58:15.392 Debug [2] BluetoothPreferences: Stop discovery
2018-10-30 19:58:15.464 Debug [2] OpenScale: Trying to connect to bluetooth device [88:OFUSCATED:17] in debug mode
2018-10-30 19:58:15.469 Debug [2] BluetoothCommunication: BT is enabled, state=12, scan mode=21, is not discovering
2018-10-30 19:58:15.473 Debug [2] BluetoothCommunication: Starting LE scan for device [88:OFUSCATED:17]
2018-10-30 19:58:17.362 Info [2] BluetoothCommunication: Connecting to [88:OFUSCATED:17] (driver: Debug)
2018-10-30 19:58:19.592 Debug [8331] BluetoothCommunication$GattCallback: onConnectionStateChange: status=0, newState=2
2018-10-30 19:58:19.596 Debug [2] BluetoothCommunication: Stopping LE scan
2018-10-30 19:58:20.624 Debug [8331] BluetoothCommunication$GattCallback: onServicesDiscovered: status=0 (0 services)
2018-10-30 19:58:20.629 Info [2] BluetoothCommunication: Disconnecting
2018-10-30 19:59:36.664 Debug [2] OpenScale: Disconnecting from bluetooth device
2018-10-30 19:59:37.834 Debug [2] BluetoothPreferences: Start discovery
2018-10-30 19:59:37.982 Debug [2] BluetoothPreferences: Discovery: android.bluetooth.adapter.action.DISCOVERY_STARTED
2018-10-30 19:59:38.035 Debug [2] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-10-30 19:59:38.180 Debug [2] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-10-30 19:59:39.755 Debug [2] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-10-30 19:59:39.769 Debug [2] BluetoothPreferences: Found supported device MI_SCALE [88:OFUSCATED:17] (driver: Xiaomi Mi Scale v1, type: 2)
2018-10-30 19:59:39.778 Debug [2] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-10-30 19:59:41.187 Debug [2] BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-10-30 19:59:41.217 Debug [2] BluetoothPreferences: Stop discovery
2018-10-30 19:59:41.276 Debug [2] OpenScale: Trying to connect to bluetooth device [88:OFUSCATED:17] in debug mode
2018-10-30 19:59:41.279 Debug [2] BluetoothCommunication: BT is enabled, state=12, scan mode=21, is not discovering
2018-10-30 19:59:41.284 Debug [2] BluetoothCommunication: Starting LE scan for device [88:OFUSCATED:17]
2018-10-30 19:59:41.793 Debug [2] BluetoothCommunication: Found LE device MI_SCALE [88:OFUSCATED:17]
2018-10-30 19:59:41.795 Info [2] BluetoothCommunication: Connecting to [88:OFUSCATED:17] (driver: Debug)
2018-10-30 19:59:42.986 Debug [8331] BluetoothCommunication$GattCallback: onConnectionStateChange: status=0, newState=2
2018-10-30 19:59:43.001 Debug [2] BluetoothCommunication: Stopping LE scan
2018-10-30 19:59:44.005 Debug [8331] BluetoothCommunication$GattCallback: onServicesDiscovered: status=0 (0 services)
2018-10-30 19:59:44.009 Info [2] BluetoothCommunication: Disconnecting
2018-10-30 20:00:03.747 Debug [2] OpenScale: Disconnecting from bluetooth device
2018-10-30 20:00:03.750 Debug [2] BluetoothPreferences: Start discovery
[... cut ...]
2018-10-30 20:00:59.986 Debug [2] BluetoothCommunication: Stopping LE scan
2018-10-30 20:01:00.001 Debug [2] MainActivity: Bluetooth connection successful established
2018-10-30 20:01:01.001 Debug [8346] BluetoothCommunication$GattCallback: onServicesDiscovered: status=0 (0 services)
2018-10-30 20:01:01.004 Info [2] BluetoothCommunication: Disconnecting
2018-10-30 20:01:01.019 Error [2] MainActivity: Bluetooth unexpected error: No services found
2018-10-30 20:01:18.793 Debug [2] OpenScale: Disconnecting from bluetooth device

```

diamant-x, 2018-10-30:
> If you download "BLE Scanner" from the play store, can it find the scale and list available services?
> 
> If that works, can you capture a btsnoop log when connecting with openScale?

Ok, I first downloaded the app BLE Scanner and the balance appeared and services appeared. Then I googled how to create a btsnoop log, enabled the toggle in the Developer options. Went to the Dev version of openscale and.... It connected! All scale measures satrted syncing.
Then i went to the Play Store version and tried again and.... Again it worked!

So not sure what the toggle for the HCI log did but it fixed the issue. Maybe it reseted some internal bluetooth logs or something? Anyways now is working fine in OP6 in Android Pie with both GP version and Dev version.

Thanks for the support!

PD: the app work greats thanks! In my case i would like to be able to sync with Google Fit the data retrieved. As an option, so that only those who want may share the data.

closed 2018-10-30 by diamant-x

erijo, 2018-10-30:
Interesting! Does it continue working if you disable the Bluetooth snooping?