oliexdev/openScale #332: Problem with data on 2nd account with Beurer BF 700

jguillouf, 2018-10-06:
Hi,
I've just bought a Beurer BF 700.
It works well for me. The scale recognizes me and the data are syncing.
I have also tried to add another account for my girl-friend in openscale but the scale doesn't recognize my girlfriend. So the scale only prints the weight and sync it ( and the  BMI )  to openscale in my girlfriend account before halting.
So I don't know if the issue is on the scale side or on the openscale side. It's very strange because the data seem to be syncing on the good accounts.
But I'm sure it's not working because I can't see the name of my girlfriend on the scale and there is a lot of data missing. 
Is there anybody here with this scale who have another account on openscale ? 
Is it working for you ? What procedure did you follow?

erijo, 2018-10-06:
Enable the debug log under settings, about. Then reproduce the issue and attach the log here.

sik0vny, 2018-10-07:
Wow. Amazing! I have EXACTLY the same problem. I have Beurer BF800. I also added my girlfriend. A I also can only see her weight and BMI. After each usage of the scale she constantly asks me: "do I have any water today?"

jguillouf, 2018-10-08:
Ok I succeed adding a 2nc account. Here is what I deed : 
- I deleted all my accounts in OpenScale
- I reset all data in the Scale ( Push Reset button during 3 seconds )
- I created my girlfriend account in OpenScale
- I powered ON the scale ( I put on foot on it during 1 second )
- I clicked on the Bluetooth button in OpenScale
- My girlfriend went on the scale and the data synced.
- Then I created my account and did the same
I think something went wrong when associating the first time with my girlfriend...
@sik0vny : I hope it could help you
But now I have some problems syncing data on my smartphone. The scale recognizes us but it only syncs 2 measurements ( instead of 3 ) for me. And only 1 ( instead of 2 )  for my GF.
i didn't have the time to investigate. I will try to do that this evening or tomorrow.

erijo, 2018-10-08:
@jguillouf: enable debug log under settings -> about and post that here after performing the sync.

jguillouf, 2018-10-10:
It works that way : 
- My phone is near the scale with BT enabled
- My girlfriend is going to the scale
- This result prints in OpenScale but not the others results when the BT was not enabled. 

But the results when  the BT was not enabled don't seem to sync for my GF.

The problem is : I am not at home when my GF weighs herself on the morning.
So I want to sync the data when I come back at home. 

@erijo Do you know how can I force a sync ? I mean if I see the BT button on the scale does it mean it's syncing or I have to click on the BT button in OpenScale just before going on the scale ?

sik0vny, 2018-10-12:
Got logs!

https://pastebin.com/33CVkYb7

erijo, 2018-10-13:
@jguillouf: I need to see the log to be able to help you.

For the scale to store measurements, it needs to be able to identify the user standing on the scale. With the default settings that means that the weight must be within 2 kg of the previous weight. This can be change to 4 kg in the original app. If the scale shows --- instead of name, then it's not able to assign the measurement to an user and openScale can't get it either.

@sik0vny: is that the full log? I'm missing the bluetooth initialization steps. Can you enable logging before connecting?

sik0vny, 2018-10-13:
Yeah. I will

jguillouf, 2018-10-15:
@erijo Here are [my logs](https://framabin.org/p/?56d8f785d86778f1#gaRi5M26hi1K4GTDaXxSCgP/LaFeQjYtmEu2bOmFW40=)

The scale identifies me and my girlfriend.
We can see our name ( in 3 letters ) on the scale.
It works well when my phone has the BT activated but I can't get the data ( mine and my gf ) later if the BT isn't activated when we weigh ourselves (or the same for my GF ).

erijo, 2018-10-15:
@jguillouf: can you run the original app and capture a btsnoop log when syncing? I would need to see how the original communicates with the scale.

sik0vny, 2018-10-16:
Here are the logs enabled before connecting:
https://pastebin.com/zui45NUM

sik0vny, 2018-10-18:
Good logs this time?

sik0vny, 2018-10-21:
In that log i dont have a second user on the scale. So you have multiuser log in first link and just the BT init in the second log.

oliexdev, 2018-10-28:
I am not sure if you can get the history data for every user on the scale later on. That might be the problem. Bluetooth must be always activated to communicate to the scale. But the procedure which are described in https://github.com/oliexdev/openScale/issues/332#issuecomment-427826006 is normal.

@sik0vny do you have still a problem? Otherwise I will close this issue.

sik0vny, 2018-10-29:
Yes I still have the same problem. Second is not identified on the scale, and the openScale receives only weight and BMI data for that second user.

jguillouf, 2018-10-29:
The Beurer BF700 scale can get the history data for a second user if OpenScale and BT are started. 
But last two days, OpenScale couldn't get the data ( mine and my gf ) with these conditions.
So I decided to bring back the scale to the seller.
Sorry I couldn't help you more. 
I think I'm going to buy a Sanitas SBF 70.

oliexdev, 2018-11-01:
@sik0vny sorry I don't know if we can still help you. 
@jguillouf the Sanitas SBF 70 and the Beurer 700 has the same firmware. You will get the same problem :wink:

closed 2018-11-01 by oliexdev