oliexdev/openScale #335: BLE connectivity issues debugging

erijo, 2018-10-07:
To help debug BLE issues, please:

1. Install the [dev build of ble-main](https://github.com/oliexdev/openScale/releases/tag/travis-dev_ble-main-build).
2. Create an user if one doesn't exists.
3. Go to Settings -> About and enable debug log.
4. Go to Settings -> Bluetooth and scan for your scale. Once found, click on it. Wait for openScale to fetch extended info from the scale. If it takes more than 30 seconds you can abort it. 
5. Go back to the main screen and click the Bluetooth button and try to fetch measurements from the scale.

Then report your result as a comment here. Remember to attach (i.e. don't copy paste it as a comment) the debug log. Also describe what you did and what happened.

Dododappere, 2018-10-08:
What issues are you trying to solve exactly?

At had a look at BluetoothCommunication.java and I have a couple of ideas you can try:

- Everytime you do a call to the BluetoothGatt object, do it on your Handler thread. Now you often call Gatt methods in callback code that runs on Binder threads. For example, you call discoverServices() while being in the onConnectionStateChange callback. You basically need to get off Binder threads as soon as you can. Do this for ALL gatt calls, so including connect, disconnect, close etc...
- You do Thread.sleep() in your code. Replace that by handler.postDelayed() calls. You don't want to do a sleep() while being on Binder threads... 
- Stop scanning before connecting to a device. I think it is a 'myth' that it helps connecting to devices. From my own experience I only found it to make connection issues worse.
- You now always use scanning to reconnect to devices. But if the devices is already known you can also use connect() with autoconnect = true. This connect call will never time out and simply create a connection when the device is seen by the OS. There were some issues with it in older Android versions but in the the recent versions it is working fine.

Here is what I do in my own BLE apps when the device is getting connected and need to discover the services:
```
...
final int delayWhenBonded = 1000;
final int delay = bondstate == BOND_BONDED ? delayWhenBonded : 0;

handler.postDelayed(new Runnable() {
    @Override
    public void run() {
        boolean result = gatt.discoverServices();
        if (!result) {
            disconnect();
        }
    }
}, delay);
...
```

Alejandro131, 2018-10-08:
Hello,

I followed your steps on a scale which had a pre-configured user from the original app as I mentioned in the other issue and just added it in this app.

The extended info (during "pairing") from the scale was fetched for around 10-15 seconds, then I saw that the table wasn't synced and I couldn't see my last weighing so I decided to step on the scale, but it couldn't weigh me as I was on a carpet floor and moved to a hard surface floor. It weighed me and showed me all my stats on the scale display, but nothing came to the app and after that I downloaded the log which I am attaching now.

[openScale_2018-10-08_10-37 (1).txt](https://github.com/oliexdev/openScale/files/2455070/openScale_2018-10-08_10-37.1.txt)

ReidarHH, 2018-10-08:
Here one from moto g5. 
Nothing transfered from the scale to the phone. Connected for about 20-30 seconds and then disconnects. 

https://drive.google.com/file/d/1pBNgj_PXoKX3TzLYzRKFlmQDJ-u0_7_7/view?usp=drivesdk

erijo, 2018-10-08:
@Alejandro131: the problem in your log I think was that the wrong descriptor was used to enable notifications. Please try with the latest version and report how it works for you.

erijo, 2018-10-08:
@ReidarHH: your case is harder. I added an extra delay just now. Please test with the latest build (once done) and let me know how it went. If that doesn't work, try to revoke the coarse location permission and try connecting to the scale. In that case you can't do the search for scales under settings -> bluetooth so just try to connect to it after enabling the debug log.

erijo, 2018-10-08:
@Dododappere: Thanks for your suggestions! I appreciate all points and hints. We've seen a number of different BLE problems, and this is my attempt at structuring the reporting a bit.

Your first two suggestions I've already implemented on the ble-main branch which is used here.

For the third, we've seen at least one scale (Medisana BS444) where it seems like connecting always fails if it isn't discovered first. There are also some phones where scanning improves the success rate. I have had the scanning during connect removed, but had to add it to fix connectivity issues.

I will let @ReidarHH try a build with autoconnect = true as a next step (after the one in the previous comment) to see if that helps in that case.

ReidarHH, 2018-10-08:
Here's the result of the debug log. 
Still same result. It connects to the scale and no data is transferred. After 20-30 seconds it disconnects. 
I try to figure the coarse settings tomorrow. 

https://drive.google.com/file/d/13ps4qoWRACtD43brBicbpuUeyAfmIA5z/view?usp=drivesdk

bandita00, 2018-10-09:
Here's mine (SBF70): 
https://pastebin.com/RXMFfJVp

Alejandro131, 2018-10-09:
@erijo Thanks for the quick response and fix. (SBF70 and Nexus 4)

I downloaded the app and fired it under 2 conditions - with a scale which has a user from the original app and with a reset scale.

With a scale which has a user from the original app and I added the user to openScale I got everything - the intermediate weight and all the body data, but there is one strange thing - the scale reports AMR (1736 kcal) and BMR (1510 kcal), but the app shows only BMR by default (and there is no option to activate AMR) and also it seems that the BMR value read by the app is actually the AMR value - 1736 kcal.

This is the log for that condition: 
[openScale_2018-10-09_09-21.txt](https://github.com/oliexdev/openScale/files/2459213/openScale_2018-10-09_09-21.txt)

With the other conditing - reset scale and reset app, I only get the weight measurement and nothing else, so it seems that openScale can't create users on the scale so that I can get all the other body data.

This is the log for it: 
[openScale_2018-10-09_09-48.txt](https://github.com/oliexdev/openScale/files/2459222/openScale_2018-10-09_09-48.txt)

On a side note, not every time I weighed myself the app registered it, but that is probably because of the merge readings settings and the fact that I weighed myself at close intervals (less than a minute), but worked with larger time intervals, so I guess it is how it works.

erijo, 2018-10-09:
@bandita00: you seem to have the same problem as @Alejandro131 had. That problem is fixed in 2f1b7663b768af4174fa128ffaba2e56a8380092. But you need to use the build in the issue description and not from the play store.

erijo, 2018-10-09:
There's been discussions about BMR, RMR and AMR elsewhere. The value is calculated by the app and different apps may use different formulas. Which is right or wrong I don't know, but we'll leave that discussion for another issue.

Regarding the new user problem I did an untested "fix". Please try the latest build after resetting the scale.

erijo, 2018-10-09:
@ReidarHH: added some code that will toggle the autoConnect flag if connection fails (to test @Dododappere idea about using autoConnect = true). After trying the other test (without the coarse location permission), please try with the latest version. If the first connection fails, try to connect again. It should then use autoConnect = true.

Alejandro131, 2018-10-10:
@erijo Thanks for the quick response and fix.

I tested it on a reset scale with 2 different users (both times the scale was reset, so I could be sure that the problem wasn't from me). The scale gets the information from the app and now my name is shown on the scale and all values are calculated, but the moment this happens, the app loses connection and no information is sent back to it, so I can't see the values in the app, only on the scale display.

Here are the logs:
[openScale_2018-10-10_10-37.txt](https://github.com/oliexdev/openScale/files/2463670/openScale_2018-10-10_10-37.txt)
[openScale_2018-10-10_10-44.txt](https://github.com/oliexdev/openScale/files/2463671/openScale_2018-10-10_10-44.txt)

P.S. About the BMR/AMR should I open another issue or it isn't a priority currently?

Alejandro131, 2018-10-11:
I decided to test the scale again and here are the results.

This time I tested on a reset scale but after that I tried to measure myself again and also tried adding a 2nd user as well. Here are my steps:
1) reset scale, install app, create new user
2) app prompts me to step barefoot on scale and measures me
3) the app loses connection but scale gives out all data and has the correct username on its display
4) I wait some minutes and connect to the scale again with the same user
5) this time the scale doesn't prompt me to step on the scale, so I just do it anyway
6) this time both the scale and the app have to correct readings, so probably it is just a problem in the first time user creation/measurement which I can live with
7) create 2nd user and go through 2) - 4) but this time even before stepping on the scale, the 2nd user gets the information from the scale from the first measurement, but it only has the weight and bmi/bmr (the scale didn't have that data as well so maybe it if it did, it would've sent everything)
8) 5) and 6) for the second user

I'm attaching 2 logs, one is from 1) to 6) and the second is from 7) to 8)

[openScale_2018-10-11_09-44.txt](https://github.com/oliexdev/openScale/files/2467719/openScale_2018-10-11_09-44.txt)
[openScale_2018-10-11_09-54.txt](https://github.com/oliexdev/openScale/files/2467721/openScale_2018-10-11_09-54.txt)

All in all it seems that even with some quirks I can successfully use openScale with SBF70 - keeping in mind that 1st time user creation readings aren't available and that I have to read the BMR/AMR from the scale display it is a working solution.

oliexdev, 2018-10-28:
@Alejandro131 you have to do it like in https://github.com/oliexdev/openScale/issues/332#issuecomment-427826006 described to create a 2nd account.

For the BMR/AMR issue you don't have to create a new issue. If you want to be exactly the RMR is calculated in openScale see #310 

@erijo Do you think that the Bluetooth connection problem is resolved with the ble branch? Should we merge it into the master rep? Do you know what the problem was? A timing problem?

erijo, 2018-10-29:
I think the connectivity is improved at least. I've merged it to master so we can perhaps close this for now. Don't have much time at the moment to look into this more.

closed 2018-10-29 by erijo

ReidarHH, 2018-11-09:
Hello Erik
I installed the the new main and tested it without success. 
Then today I got the new oreo update. 
And now it works importing all previous weight logs.
At first it didn't import all data, but after making a new measurement openscale imported all data. 
Perfect. Happy to use the app again. 
Thank you for all the work you put into the app.

erijo, 2018-11-25:
@Alejandro131 : if you want to you can test [this dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev_beurer-build) which should hopefully work better with the first user registration.

Alejandro131, 2018-12-05:
Just tested it - works perfectly - 1st user doesn't get disconnected on first measurement and everything is transferred to the app. 2nd user registration works flawlessly as well - thanks for the update and fix :)