oliexdev/openScale #336: [Feature] Waistline integration

yarons, 2018-10-08:
Hey,
I came across Waistline by @davidhealey.

This app is using data from OpenFoodFacts to make better dietary decisions based on facts.
I was thinking about transferring data to (and from?) this app in order to have the entire picture when it comes to weight.

I also opened a feature request in Gadgetbridge with the same request (I was told something about ContentProvider).

What would be the best way to achieve this kind of data exchange automagically?

Corresponding issues:
https://github.com/davidhealey/waistline/issues/71
https://github.com/Freeyourgadget/Gadgetbridge/issues/1291

oliexdev, 2018-10-08:
@yarons I like this idea and I will support it. We have already a simple content provider in the latest release of openScale v1.8.2 (released yesterday).
So actually you can get the weight if you just grab the data via the content provider. 
@erijo wrote also a simple example app on https://github.com/erijo/openScale/tree/connector how to grab the weight data. So I guess on the openScale it is already ready for the connection between the three open source apps. If you have a question don't hesitate to ask.

yarons, 2018-10-08:
What can I say, you are simply awesome! Thank you so much for your enthusiasm about this cooperation, hoping we'll get to keep a better health using only open source software 😄

closed 2020-06-18 by oliexdev

yarons, 2020-06-19:
So what's the final resolution then?

oliexdev, 2020-06-19:
nothing no effort from the other projects.

yarons, 2020-06-19:
So sad :( I'm sorry to hear that, thank you so much for all you've done!

yarons, 2024-04-08:
I've reignited the conversation at Energize:
https://codeberg.org/epinez/Energize
https://codeberg.org/epinez/Energize/issues/91
Currently it's mostly about GadgetBridge but I have mentioned openScale, thank you again :)