oliexdev/openScale #338: Feature request: Add automatic data export (not backup)

sik0vny, 2018-10-10:
Hi.
I would really love to see a feature that will automatically export all measurements data to csv file:
- done after there is a new data added/measured
- separate file for each user
- written always to a specified directory
- always overwrite the old files
- without any confirmations, just "Always export"

Without this, I can't pass the data in readable format to any other application.

oliexdev, 2018-10-12:
Currently, you can use the content provider or you can import the exported backup database. If we have time we can in the future add an option for the backup as a CSV file, see also #236.

sik0vny, 2018-10-13:
Thanks for answer. Database file is not for imports/exports. Its for backup. I will have to write some harder stuff to decode it/open it in other application. Ofcourse I will probably be able to do that, but its not its purpose. Thats why you have export option in openScale. To export the data in standardized, easily readable format. I would just like to have automatic export:)
So thumbs up for introducing this feature:)
Thank

JetSetDoritos, 2020-07-20:
My use case for this is auto importing my data into a google sheet, I could use some sync app to auto upload the .csv to google drive then have some Apps Script code to add new data to the spreadsheet.  This is still possible with the Database file just heavier/more involved, pretty sure I cant use Apps Script.

tjaderxyz, 2020-11-19:
I use the daily backup function as a "auto data export" feature, and it seems other people also do similar things. Perhaps a simple change that would be useful for us is to have a "after data is added" backup schedule option. Currently my backup file is on average half a day behind, it would be nice if it was always updated after data is added.

thusser, 2023-01-31:
I know, this is old, but I'd like to add something. I think a general "export new data to <database>" would be nice after each new measurement, with <database> being either something like a simple CSV, JSON, etc, or even a full blown database. Most people have their scale at home and are connected to their home WiFi when they use it. So why not have an option to export new data to a SQL/Influx/whatever?