oliexdev/openScale #339: Add support for Wii balance board

XavierTolza, 2018-10-10:
Hi! could you add support for the wii balance board? Its a well known hardware, I tried to implement it myself but I could get the android studio running.
Do you have some piece of information on how to get the dev env running for android newbies? I tried opening the `android_app` folder using android studio but I got nothing.
It may be a stupid question, but I'm asking anyway ;)

oliexdev, 2018-10-12:
Why doesn't work your android studio? See https://developer.android.com/studio/intro/

oliexdev, 2018-10-28:
I did a little research. The calculation is not the problem, as you wrote the hardware is well known.
There is even an open-source Android App by paulburton https://github.com/paulburton/fitscales
The problem is that the wii balance board is using Bluetooth 3.x with the l2cap protocol. openScale uses only the newer Bluetooth 4.x which is not downwards compatible. We have to implement the Bluetooth 3.x routines into openScale to do the communication. Unfortunately, in Android version > 4.2 Google removes the l2cap protocol in Android which is needed. :/
All in all it is possible if anybody are willing to take the effort to implement it for a new Android version.

tomassenabre, 2018-11-06:
Thank you Oliex for your response and your interest. Unfortunately I do not have the necessary knowledge to make the implementation that you comment on. We will wait in case someone wants to help us

SamSnead, 2019-02-20:
Hoffe, ich kann auch in deutsch etwas schreiben.

Ich suche eine App, die mit dem WII Balance Board kommuniziert, aber nicht nur das Gesamtgewicht anzeigt, sondern die jeweilige Belastung des linken und rechten Fusses.  Bin auch bereit, für die Entwicklung zu zahlen.

oliexdev, 2019-02-20:
Also ich verstehe auch Deutsch aber wahrscheinlich die anderen nicht ;) 
Eigentlich wurde bereits alles in https://github.com/oliexdev/openScale/issues/339#issuecomment-433680792 gesagt. Für was willst du die Wii Balance denn genau benutzen?

SamSnead, 2019-02-21:
Ich brauche eine Waage für medizinische Zwecke, die die Belastung (Gewicht) einer Person aufteilt in linker und rechter Fuss. Das Wii Board wäre perfekt dafür plus eben eine App, die das Gewicht links/rechts getrennt anzeigt.

oliexdev, 2019-02-21:
Muss es eine App sein? Sonst würde ich sagen benutze https://github.com/schellingb/WiiBalanceScale

Michuelnik, 2021-08-30:
I just read that Android Q (aka Android 10) brought back L2CAP...?

Rudy-MS, 2023-04-12:
I would like to also voice my support in getting the Wii Fit board to work with OpenScale. 

The hardware is pretty much everywhere(Thrift Stores, Flea Markets & Garage Salea). 

Its also pretty cheap these days.

Liriel-666, 2024-04-03:
I would be for support of it. 
The old android app doesnt work anymore and tge only otger way is windows

LittleTale, 2024-07-18:
Any news on integrating the Wii Balance Board ? 
The windows app works great. But it would be super useful to just step on it and check everything on your phone ! 🤳🏼

Lord-Memester, 2025-09-18:
Update: got the balance board to pair to my phone and maintain connection, courtesy of https://github.com/gaberilde/PairWiimote-Improved! 

[image]

I still haven't managed to get it to show up in the list of devices that can be connected to by openScale.

@oliexdev: What file do I need to modify to get the board to show up in the list of connectable devices? As a newcomer to Java, it's been a little difficult for me to determine the structure of the repository. On that note, I'd also really like to know what I do and don't need in the file that defines the protocol. A template would be helpful, and I'm not quite sure what to reference for that purpose.

Liriel-666, 2025-10-20:
> Any news on integrating the Wii Balance Board ? The windows app works great. But it would be super useful to just step on it and check everything on your phone ! 🤳🏼

What do you mean with  windows app?