oliexdev/openScale #341: last version broke sbf75 bluetooth connection

Somsy, 2018-10-12:
It seems that since last version rolled out on my phone I can no longer synchronize data with my sbf75 phone.
I tried to cleared cache partition on the phone, rebooted, turned off. Nothing works.
whenever I try to connect to the scale, It'd connect and within a second bluetooth message error.
I really love this app and have been using it for the last 4 months flawlessly.
I don't know if there is somewhere I could get previous apk so that I could try previous version and see if it solves my issue

erijo, 2018-10-12:
Please reproduce the problem with logging enabled (under settings, about) and attach the log here.

Somsy, 2018-10-13:
Here's an attempt...
[openScale_2018-10-13_09-22.txt](https://github.com/oliexdev/openScale/files/2475201/openScale_2018-10-13_09-22.txt)

erijo, 2018-10-13:
Can you try the build in #335 and attach the log here afterwards?

Somsy, 2018-10-13:
Shit installing dev version delete weight history?

Somsy, 2018-10-13:
[openScale_2018-10-13_14-26.txt](https://github.com/oliexdev/openScale/files/2475447/openScale_2018-10-13_14-26.txt)
there it is

erijo, 2018-10-13:
As you hopefully noticed the history of still there in the original app. The dev version is installed alongside to not cause that kind of problem.

Somsy, 2018-10-13:
It connects and stall...
Most of the time though I get com.android.Bluetooth stopped.

Somsy, 2018-10-13:
[openScale_2018-10-13_20-12.txt](https://github.com/oliexdev/openScale/files/2475714/openScale_2018-10-13_20-12.txt)

Somsy, 2018-10-13:
I just installed 1.8.1 release version. It works and communicates normally. So it's indeed something with changes that have been done since then.

erijo, 2018-10-13:
Can you post a log from 1.8.1 as well?

Somsy, 2018-10-13:
[openScale_2018-10-13_20-32.txt](https://github.com/oliexdev/openScale/files/2475724/openScale_2018-10-13_20-32.txt)

erijo, 2018-10-14:
@Somsy: can you install the Android SDK and run `adb logcat` while trying to connect with the dev version? I would like to see the error that should be logged when discoverServices() return false. Unfortunately Android logs can't be included in the openScale log.

erijo, 2018-10-14:
You did. From the log:
```10-14 20:21:04.757 17917 17917 I crash_dump64: performing dump of process 28945 (target tid = 28982)
10-14 20:21:04.758 17917 17917 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
10-14 20:21:04.758 17917 17917 F DEBUG   : Build fingerprint: 'asus/WW_Phone/ASUS_Z017D_1:8.0.0/OPR1.170623.026/15.0410.1807.75-0:user/release-keys'
10-14 20:21:04.758 17917 17917 F DEBUG   : Revision: '0'
10-14 20:21:04.758 17917 17917 F DEBUG   : ABI: 'arm64'
10-14 20:21:04.758 17917 17917 F DEBUG   : pid: 28945, tid: 28982, name: bt_workqueue  >>> com.android.bluetooth <<<
10-14 20:21:04.758 17917 17917 F DEBUG   : signal 6 (SIGABRT), code -6 (SI_TKILL), fault addr --------
10-14 20:21:04.763 17917 17917 F DEBUG   : Abort message: '[FATAL:gatt_api.cc(701)] Assert failed: len == 5. Received bad response length: 1
10-14 20:21:04.763 17917 17917 F DEBUG   : '
10-14 20:21:04.763 17917 17917 F DEBUG   :     x0   0000000000000000  x1   0000000000007136  x2   0000000000000006  x3   0000000000000008
10-14 20:21:04.763 17917 17917 F DEBUG   :     x4   673a4c415441465b  x5   673a4c415441465b  x6   673a4c415441465b  x7   2e6970615f747461
10-14 20:21:04.763 17917 17917 F DEBUG   :     x8   0000000000000083  x9   7870f0e54f0b789d  x10  0000000000000000  x11  0000000000000001
10-14 20:21:04.763 17917 17917 F DEBUG   :     x12  ffffffffffffffff  x13  7220646162206465  x14  0000000000000000  x15  0000007e076ba000
10-14 20:21:04.763 17917 17917 F DEBUG   :     x16  0000007e043a82e8  x17  0000007e04349d00  x18  0000007de998be50  x19  0000000000007111
10-14 20:21:04.763 17917 17917 F DEBUG   :     x20  0000000000007136  x21  0000000000000000  x22  0000007de96fb000  x23  0000007de85c1321
10-14 20:21:04.763 17917 17917 F DEBUG   :     x24  00000000000af59e  x25  0000000000002019  x26  0000000000000000  x27  0000000000000001
10-14 20:21:04.763 17917 17917 F DEBUG   :     x28  0000007de85c1c48  x29  0000007de85c12b0  x30  0000007e042fde14
10-14 20:21:04.763 17917 17917 F DEBUG   :     sp   0000007de85c1270  pc   0000007e04349d08  pstate 0000000060000000
10-14 20:21:04.807 17917 17917 F DEBUG   : 
10-14 20:21:04.807 17917 17917 F DEBUG   : backtrace:
10-14 20:21:04.807 17917 17917 F DEBUG   :     #00 pc 0000000000069d08  /system/lib64/libc.so (tgkill+8)
10-14 20:21:04.807 17917 17917 F DEBUG   :     #01 pc 000000000001de10  /system/lib64/libc.so (abort+88)
10-14 20:21:04.807 17917 17917 F DEBUG   :     #02 pc 00000000000824c4  /system/lib64/libchrome.so (_ZN4base5debug13BreakDebuggerEv+20)
10-14 20:21:04.807 17917 17917 F DEBUG   :     #03 pc 0000000000099104  /system/lib64/libchrome.so (_ZN7logging10LogMessageD2Ev+928)
10-14 20:21:04.807 17917 17917 F DEBUG   :     #04 pc 0000000000152f00  /system/lib64/hw/bluetooth.default.so (_Z11read_phy_cbN4base8CallbackIFvhhhELNS_8internal8CopyModeE1EEEPht+156)
10-14 20:21:04.807 17917 17917 F DEBUG   :     #05 pc 00000000001548ec  /system/lib64/hw/bluetooth.default.so (_ZN4base8internal7InvokerINS0_9BindStateIPFvNS_8CallbackIFvhhhELNS0_8CopyModeE1EEEPhtEJS6_EEEFvS7_tEE3RunEPNS0_13BindStateBase
EOS7_Ot+68)
10-14 20:21:04.807 17917 17917 F DEBUG   :     #06 pc 000000000014dce0  /system/lib64/hw/bluetooth.default.so (_ZL45btu_hcif_command_complete_evt_with_cb_on_taskP6BT_HDR+180)
10-14 20:21:04.807 17917 17917 F DEBUG   :     #07 pc 00000000001ab3c4  /system/lib64/hw/bluetooth.default.so (_ZL22internal_dequeue_readyPv+80)
10-14 20:21:04.807 17917 17917 F DEBUG   :     #08 pc 00000000001afc0c  /system/lib64/hw/bluetooth.default.so (_ZL11run_reactorP9reactor_ti+312)
10-14 20:21:04.807 17917 17917 F DEBUG   :     #09 pc 00000000001afaa8  /system/lib64/hw/bluetooth.default.so (_Z13reactor_startP9reactor_t+80)
10-14 20:21:04.807 17917 17917 F DEBUG   :     #10 pc 00000000001b143c  /system/lib64/hw/bluetooth.default.so (_ZL10run_threadPv+256)
10-14 20:21:04.808 17917 17917 F DEBUG   :     #11 pc 00000000000666a0  /system/lib64/libc.so (_ZL15__pthread_startPv+36)
10-14 20:21:04.808 17917 17917 F DEBUG   :     #12 pc 000000000001f1a4  /system/lib64/libc.so (__start_thread+68)
```

An assert fails in the bluetooth stack, making the call to discoverServices fail later on:
```10-14 20:21:06.176 17857 17857 E BluetoothGatt: 
10-14 20:21:06.176 17857 17857 E BluetoothGatt: android.os.DeadObjectException
10-14 20:21:06.176 17857 17857 E BluetoothGatt:         at android.os.BinderProxy.transactNative(Native Method)
10-14 20:21:06.176 17857 17857 E BluetoothGatt:         at android.os.BinderProxy.transact(Binder.java:748)
10-14 20:21:06.176 17857 17857 E BluetoothGatt:         at android.bluetooth.IBluetoothGatt$Stub$Proxy.discoverServices(IBluetoothGatt.java:1403)
10-14 20:21:06.176 17857 17857 E BluetoothGatt:         at android.bluetooth.BluetoothGatt.discoverServices(BluetoothGatt.java:919)
10-14 20:21:06.176 17857 17857 E BluetoothGatt:         at com.health.openscale.core.bluetooth.BluetoothCommunication$GattCallback$2.run(BluetoothCommunication.java:661)
10-14 20:21:06.176 17857 17857 E BluetoothGatt:         at android.os.Handler.handleCallback(Handler.java:789)
10-14 20:21:06.176 17857 17857 E BluetoothGatt:         at android.os.Handler.dispatchMessage(Handler.java:98)
10-14 20:21:06.176 17857 17857 E BluetoothGatt:         at android.os.Looper.loop(Looper.java:169)
10-14 20:21:06.176 17857 17857 E BluetoothGatt:         at android.app.ActivityThread.main(ActivityThread.java:6578)
[... cut ...] oke(Native Method)
10-14 20:21:06.176 17857 17857 E BluetoothGatt:         at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
10-14 20:21:06.176 17857 17857 E BluetoothGatt:         at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:767)
10-14 20:21:06.176 17857 17857 E BluetoothCommunication$GattCallback: Could not start service discovery
```

I've done some changes to see if it can avoid triggering the assert. Please try with the latest build (once it is built).

Somsy, 2018-10-15:
Yes you did it to this time dev version is working. It can communicates without crashing

erijo, 2018-10-15:
Nice!

Can you please attach the log from openScale when you do all the steps in #335?

lenucksi, 2018-10-15:
I've also got a similar issue, it appeared with the release version 1.8.2 with other versions working flawlessly before. Also on a SBF75 scale.

Effect: 
- Open the app
- OpenScale reports that it connects successfully
- No stored measurements are imported anymore
- If I put load on the scale I get a toast message with a weight reading.

erijo, 2018-10-15:
@lenucksi: please try the version in #335 and post the log here.

Somsy, 2018-10-15:
[openScale_2018-10-15_21-45.txt](https://github.com/oliexdev/openScale/files/2480319/openScale_2018-10-15_21-45.txt)

Somsy, 2018-10-15:
Everything went well...

closed 2018-11-01 by oliexdev