oliexdev/openScale #343: Bluetooth connection broken for Beurer BF700

CodeFinder2, 2018-10-24:
This issue appears to be very similar (if not equal to) #341.

openScale says "connected" and immediately afterwards "disconnected". No data is synced. The scale appears to work with the original (Beurer) app (which is crap). Here's the log:

2018-10-24 20:34:57.291 Debug [2] AboutPreferences: Debug log enabled, openScale v1.8.2 (29), SDK 26, OnePlus ONEPLUS A3003
2018-10-24 20:35:54.086 Debug [2] OpenScale: Trying to connect to bluetooth device [00:13:04:1A:D4:20] (Beurer BF700)
2018-10-24 20:35:54.106 Debug [2] BluetoothCommunication: BT is enabled, state=12, scan mode=21, is not discovering
2018-10-24 20:35:54.126 Debug [2] BluetoothCommunication: Starting LE scan for device [00:13:04:1A:D4:20]
2018-10-24 20:35:54.522 Debug [2] OpenScale: Disconnecting from bluetooth device
2018-10-24 20:35:54.524 Debug [2] BluetoothCommunication: Stopping LE scan
2018-10-24 20:35:55.137 Debug [2] OpenScale: Trying to connect to bluetooth device [00:13:04:1A:D4:20] (Beurer BF700)
2018-10-24 20:35:55.154 Debug [2] BluetoothCommunication: BT is enabled, state=12, scan mode=21, is not discovering
2018-10-24 20:35:55.163 Debug [2] BluetoothCommunication: Starting LE scan for device [00:13:04:1A:D4:20]
2018-10-24 20:36:05.183 Debug [2] BluetoothCommunication: Device not found in LE scan, connecting directly
2018-10-24 20:36:05.189 Debug [2] BluetoothCommunication: Stopping LE scan
2018-10-24 20:36:05.209 Info [2] BluetoothCommunication: Connecting to [00:13:04:1A:D4:20] (driver: Beurer BF700/800 / Runtastic Libra)
2018-10-24 20:36:05.744 Debug [3901] BluetoothCommunication$GattCallback: onConnectionStateChange: status=0, newState=2
2018-10-24 20:36:05.764 Debug [2] MainActivity: Bluetooth connection successful established
2018-10-24 20:36:06.789 Error [3901] BluetoothCommunication$GattCallback: Could not start service discovery
2018-10-24 20:36:06.791 Info [3901] BluetoothCommunication: Disconnecting
2018-10-24 20:36:06.816 Debug [2] MainActivity: Bluetooth connection lost

Appears to be related to:
2018-10-24 20:36:06.789 Error [3901] BluetoothCommunication$GattCallback: Could not start service discovery

If you need more information, please let me know. Any ideas?

I've tried the latest dev version from 13/10/2018 as well the version on Google Play.

erijo, 2018-10-26:
Can you try the version in #335?

CodeFinder2, 2018-10-27:
Thanks for the hint! I did that, see the **attached debug log file**: [openScale_2018-10-27_11-08.txt](https://github.com/oliexdev/openScale/files/2521630/openScale_2018-10-27_11-08.txt)

What happened is that openScale was **able to connect** to my scale and to retrieve measurements. However, the scale **didn't measured other values**, only the weight. Also, openScale only showed the weight. I made 5 weight measurements in total. Version 1.6.1 was able to retrieve other values (like body fat percentage). Note that fetching the 'extended info' from the scale terminated automatically (i.e., it didn't take longer than 30s).

If you need more information, please let me know! 😎

By the way, the version in **#335 crashed with this stack trace** when i switched back to the app and pressing the 'Info' button in settings again (IIRC). Unfortunately, I was not able to reproduce the problem once more:
Build version: 1.8.1-dev_4ace7319_2018-10-15 
Current date: 2018-10-27 11:07:47 
Device: OnePlus ONEPLUS A3003 
 
Stack trace:  
java.lang.IllegalArgumentException: Invalid fragment for this activity: com.health.openscale.gui.preferences.AboutPreferences
	at android.preference.PreferenceActivity.switchToHeaderInner(PreferenceActivity.java:1250)
	at android.preference.PreferenceActivity.switchToHeader(PreferenceActivity.java:1305)
	at android.preference.PreferenceActivity.onHeaderClick(PreferenceActivity.java:1102)
	at android.preference.PreferenceActivity.onListItemClick(PreferenceActivity.java:1087)
	at android.app.ListActivity$2.onItemClick(ListActivity.java:319)
	at android.widget.AdapterView.performItemClick(AdapterView.java:318)
	at android.widget.AbsListView.performItemClick(AbsListView.java:1197)
	at android.widget.AbsListView$PerformClick.run(AbsListView.java:3166)
	at android.widget.AbsListView$3.run(AbsListView.java:4119)
	at android.os.Handler.handleCallback(Handler.java:789)
	at android.os.Handler.dispatchMessage(Handler.java:98)
	at android.os.Looper.loop(Looper.java:164)
	at android.app.ActivityThread.main(ActivityThread.java:6798)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:767)

oliexdev, 2018-11-04:
maybe reset the scale and create a new user!?

I can't reproduce the error too.

closed 2018-11-04 by oliexdev

ccoenen, 2018-11-08:
I had (what sounds like) the same error. It started with the prevoous update a few weeks ago. It's gone with the most recent update from 4 days ago. All good now :-)