oliexdev/openScale #346: [Yunmai Mini 2 Balance] Historical Data not transferred

bpi5ckuesv, 2018-10-28:
Application works but historical data is not transferred, i.e. when i weight myself once or several times and try to get this data afterwards connecting to openScale it does not work. openScale is only available to catch data which was recorded after connection establishment. Same holds for Yunmai app. Any hints, is there a hidden bluetooth setting for this?

oliexdev, 2018-10-28:
no the scale doesn't support it, see https://github.com/oliexdev/openScale/wiki/Supported-scales-in-openScale

This has nothing to do with Bluetooth. If the vendor doesn't implement a memory on the scale then it can't store a measurement.

closed 2018-10-28 by oliexdev

bpi5ckuesv, 2018-10-28:
@oliexdev Hello, my scale doesn't appear on the wiki of supported scales at all. For the older version Yunmai Mini (v1) the FAQs on the support page http://www.iyunmai.com/uk/light-mini/specs.html says

_11. How often do I need to bring the Bluetooth device (smartphone) near 
the scale to retain history? Every time you weigh in? Every week? Every 
day?
There are two ways for data to transfer between the scale and smartphone. If 
you have the smartphone close to the scale when you weigh yourself, the app 
will automatically connect with the scale through Bluetooth, and will transfer 
the body metrics into the smartphone immediately. If you do not have a smart
-phone with you when you weigh in, the data will be recorded into the scale, 
and will be sent to the app when the scale is later connected to the smart
-phone. The scale will recognize up to 16 users and store their data separately. 
The scale can store up to 1000 sets of data. Please connect the smartphone 
to the scale before it reaches the 1000 limit._

So is assumed that data is stored on the device and can be synced later and i was pretty sure that it worked with openScale before. After it stopped working i installed the vendors app and checked there, reset did not helped. I will check with the vendor. Thanks for your support.