oliexdev/openScale #348: Oneplus3 + Mi Scale (Rev.1) Bluetooth connection lost

schneekluth, 2018-10-31:
I'm currently running latest Version of OpenScale (1.8.2) on my Oneplus 3 with latest OxygenOs (5.0.6). The phone can no longer retrieve values from the Mi Scale. Bluetooth connection is enabled but lost seconds after ist. Whereas my old Nexus 5 retrieves the data just fine. Is this an OnePlus issue or is OpenScale the faulty part?

Here is the debug log from OpenScale:
```
2018-10-28 11:40:09.626 Debug [2] OpenScale: Trying to connect to bluetooth device [XX:XX:XX:XX:XX:XX] (MI_SCALE)
2018-10-28 11:40:09.640 Debug [2] BluetoothCommunication: BT is enabled, state=12, scan mode=21, is not discovering
2018-10-28 11:40:09.648 Debug [2] BluetoothCommunication: Starting LE scan for device [XX:XX:XX:XX:XX:XX]
2018-10-28 11:40:13.084 Debug [2] BluetoothCommunication: Stopping LE scan
2018-10-28 11:40:13.093 Info [2] BluetoothCommunication: Connecting to [XX:XX:XX:XX:XX:XX] (driver: Xiaomi Mi Scale v1)
2018-10-28 11:40:17.342 Debug [3979] BluetoothCommunication$GattCallback: onConnectionStateChange: status=0, newState=2
2018-10-28 11:40:17.363 Debug [2] MainActivity: Bluetooth connection successful established
2018-10-28 11:40:18.593 Error [3979] BluetoothCommunication$GattCallback: Could not start service discovery
2018-10-28 11:40:18.594 Info [3979] BluetoothCommunication: Disconnecting
2018-10-28 11:40:18.603 Debug [2] MainActivity: Bluetooth connection lost
```

oliexdev, 2018-11-01:
please try out the latest [dev build](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build)

schneekluth, 2018-11-01:
Dev build works as expected and reads the values perfectly fine.

closed 2018-11-02 by oliexdev