oliexdev/openScale #350: Beurer BF700 listed as "not supported"

rsmeral, 2018-11-11:
**Describe the bug**
The app sees the scale, but shows it as not supported.
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Open Bluetooth scanning menu
2. See BF700 in the list – shown as "device not supported"

Reproduced with v1.8.3

**Expected behavior**
BF700 listed as supported (since the README claims it's supported)

**Debug log**
[openScale_2018-11-11_15-26.txt](https://github.com/oliexdev/openScale/files/2569452/openScale_2018-11-11_15-26.txt)

erijo, 2018-11-11:
Can you please enable debug logging, go search for the device and then click on it? This will fetch some more info from the scale. Then attach that log here.

rsmeral, 2018-11-12:
Attaching detailed log after tapping on the BF700 entry.
[openScale_2018-11-11_23-04.txt](https://github.com/oliexdev/openScale/files/2571322/openScale_2018-11-11_23-04.txt)

erijo, 2018-11-12:
Please try the [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) and report back the result.

LosD, 2018-11-14:
That's odd. I just added my BF-700 today, and it works like a charm (and much, _MUCH_ better than Beurers own app!).

rsmeral, 2018-11-14:
Tried the dev version – still no luck. The app now showed the scale as supported, but couldn't connect.
Attaching log: [openScale_2018-11-12_23-55.txt](https://github.com/oliexdev/openScale/files/2582304/openScale_2018-11-12_23-55.txt)

@LosD Exactly, the Beurer app sucks big time. Really hoping I can get openScale to work.

LosD, 2018-11-14:
Did you try removing the batteries and/or a reset? Mine did something similar (though with Beurers own app), and the manual's advice is to remove the batteries for a moment, then try again, which worked for me.

erijo, 2018-11-14:
@rsmeral's scale identifies itself with "BF700". We have support for "BEURER BF700" and "BF-700" and I assume your scale has one of those names @LosD.

@rsmeral: I did a new change just now that I would like you to test. Please install the latest dev version and try again.

LosD, 2018-11-14:
@erijo Right you are. BEURER BF700

rsmeral, 2018-11-14:
@erijo This worked! Apparently, my BF700 has an identity crisis.
Thanks for the quick fix!

erijo, 2018-11-15:
Excellent @rsmeral. Closing this issue then.

closed 2018-11-15 by erijo