oliexdev/openScale #351: Error when load data

brunobps, 2018-11-14:
Build version: 1.8.3 
Build date: 1979-11-30 00:00:00 
Current date: 2018-11-13 22:56:38 
Device: Motorola Moto G (5) Plus 
 
Stack trace:  
java.lang.NullPointerException: Attempt to invoke virtual method 'void com.health.openscale.core.datatypes.ScaleMeasurement.setMuscle(float)' on a null object reference
	at com.health.openscale.core.bluetooth.BluetoothMGB.onBluetoothDataChange(BluetoothMGB.java:182)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$GattCallback$8.run(BluetoothCommunication.java:759)
	at android.os.Handler.handleCallback(Handler.java:790)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:164)
	at android.app.ActivityThread.main(ActivityThread.java:6626)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:811)

erijo, 2018-11-14:
Please attach a log (enable under settings, about) when the error happens.

brunobps, 2018-11-15:
[openScale_2018-11-14_23-06.txt](https://github.com/oliexdev/openScale/files/2583235/openScale_2018-11-14_23-06.txt)

brunobps, 2018-11-15:
[openScale_2018-11-14_23-12.txt](https://github.com/oliexdev/openScale/files/2583245/openScale_2018-11-14_23-12.txt)

erijo, 2018-11-17:
Please try with the latest [dev version|https://github.com/oliexdev/openScale/releases/tag/travis-dev-build].

brunobps, 2018-11-18:
It's solved. TKS!

closed 2018-11-18 by erijo