oliexdev/openScale #352: Inlife scale support, it looks like the Accuway scale on your page (clone?)

hyakki-, 2018-11-20:
Hi can you add support for the InLife scale it appears to have the same design as the Accuway scale, might just be a clone, 
[btsnoop_hci-1.log](https://github.com/oliexdev/openScale/files/2597563/btsnoop_hci-1.log)

oliexdev, 2018-11-20:
please try out the latest [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build)

erijo, 2018-11-20:
I'm not so sure it's the same protocol as the Accuway scale. Here's a notification from the btsnoop:
`02 dd 02 c0 00 b6 d4 01 c2 3c 3d 00 bf aa`

* `02 dd` seems to mean that measurement is done.
* `02 c0` = 704 = 70.4 kg = 155.2 lb (weight)
* `00 b6 d4` = 46804 = 46.804 kg (LBM, scaled slightly based on activity level)
* `01 c2` = 450 (base for visceral calculations)
* `3c 3d` = 15421 = 1542.1 (BMR)
* `00` = battery level (?)
* `bf` = ?
* `aa` = all commands and notifications end with this

Can you provide the values for fat, water, bone, etc. for the measurement in the attached btsnoop? Or provide a new?

Also provide your age, sex and activity level as set in the app.

erijo, 2018-11-20:
@oliexdev: I reverted your test commit as the inlife scale doesn't have the service used in BluetoothExingtechY1.

@hyakki-: I added a skeleton for the scale that, if it works, should get the weight. Please try the dev version as @oliexdev wrote.

hyakki-, 2018-11-20:
I tried the new dev build and it seems to connect to the scale, but I don't see any output in the app
here are some new logs

[image]

[image]
[btsnoop_hci-new1.log](https://github.com/oliexdev/openScale/files/2599176/btsnoop_hci-new1.log)
-
[image]
[btsnoop_hci-new2.log](https://github.com/oliexdev/openScale/files/2599177/btsnoop_hci-new2.log)

erijo, 2018-11-20:
Please attach a log from openscale as well.

hyakki-, 2018-11-20:
sure here is the openscale log
[openScale_2018-11-20_02-40.txt](https://github.com/oliexdev/openScale/files/2599282/openScale_2018-11-20_02-40.txt)

erijo, 2018-11-20:
It seems like you only did the discovery step. After selecting your scale in the settings, go back to the main screen and click the Bluetooth icon at the top.

erijo, 2018-11-20:
Did you step on the scale after openScale connected to it?

hyakki-, 2018-11-20:
> 
> 
> Did you step on the scale after openScale connected to it?

yeah that's the only way it turns on, I stood on it until it turns off.

erijo, 2018-11-20:
Please try with the latest version and post log.

hyakki-, 2018-11-20:
ok this time I got some values, it took 2 tries though, maybe just a glitch in bt even with stock app sometimes it doesn't connect fast enough

(first try, no values)
[openScale_2018-11-20_04-06.txt](https://github.com/oliexdev/openScale/files/2599627/openScale_2018-11-20_04-06.txt)

(second try, app showed all values) 
[openScale_2018-11-20_04-06 (1).txt](https://github.com/oliexdev/openScale/files/2599630/openScale_2018-11-20_04-06.1.txt)

[image]

hyakki-, 2018-11-20:
btw there is some loop bug under Table 
[image]

erijo, 2018-11-20:
Please try again with the latest version and upload a log.

hyakki-, 2018-11-20:
[openScale_2018-11-20_04-59.txt](https://github.com/oliexdev/openScale/files/2599802/openScale_2018-11-20_04-59.txt)

almost there!, values seem more accurate now, it still adds a bunch of duplicate data to the table list, one scale read adds like 13 columns, sometimes it takes a try or two for values to display.

I will do some more testing later today, off to work now

erijo, 2018-11-20:
It would be good if you could test the latest version and compare the values you get in the original app with the ones you get in openScale. Make sure you have the same settings (age, gender, height and athlete/activity level (pick amateur in original app and moderate in openScale)) when you do the measurements. You can ignore metabolism since that's calculated in openScale and does not use the value from the scale. But all the other values should be reported in openScale as well. Visceral fat must be enabled in openScale settings as it is hidden by default.

Please run the test with gender set to male and height set to 195 cm and 160 cm. Then redo the test with gender set to female (for both heights). I.e. four tests in both apps.

hyakki-, 2018-11-21:
Tested using dev a3e4cdb , working excellent, now connects first try and loop seems to be fixed.

```

Height 195 - amateur - male (orig app)
Weight 70.9kg , Bone 3.2kg , Water 64.4 % , Muscle  48.4 %, Fat 11.8% , Visceral fat 1.0 , BMI 18.6 , Metabolism 1738

Height 195 - moderate - male (Openscale app)
Weight 70.90 kg , Bone Mass 3.18kg , Water 63.41 % , Muscle  47.60 %, Body Fat 13.14% , Visceral fat 18.49 , BMI 18.65 , BMR 2749

Height 160 - amateur - male (orig app)
Weight 70.9kg , Bone 2.6 kg , Water 52.6 % , Muscle  39.5 %, Fat 27.9% , Visceral fat 5.8 , BMI 27.7 , Metabolism 1563

Height 160 - moderate - male (Openscale app)
Weight 70.90 kg , Bone Mass 2.59 kg , Water 51.65 % , Muscle  38.77 %, Body Fat 29.25 % , Visceral fat 6.67 , BMI 27.70 , BMR 2421
--

Height 195 - amateur - female (orig app)
Weight 70.9kg , Bone 3.1kg , Water 62.3 % , Muscle  46.7 %, Fat 14.7% , Visceral fat 1.0 , BMI 18.6 , Metabolism 1716

Height 195 - moderate - female (Openscale app)
Weight 70.90 kg , Bone Mass 3.11kg , Water 62.05 % , Muscle  46.58 %, Body Fat 15.00% , Visceral fat 25.45 , BMI 18.65 , BMR 2500

-
Height 160 - amateur - female (orig app)
Weight 70.9kg , Bone 2.5kg , Water 50.5 % , Muscle  37.9 %, Fat 30.9% , Visceral fat 3.5 , BMI 27.7 , Metabolism 1650

Height 160 - moderate - female (Openscale app)
Weight 70.90 kg , Bone Mass 2.52kg , Water 50.24 % , Muscle  37.71 %, Body Fat 31.18% , Visceral fat 4.92 , BMI 27.70 , BMR 2172

```

erijo, 2018-11-21:
Thanks for the detailed test result!

I think I've fixed calculation of visceral fat for the height = 195 case. Can you please retest that case in openScale (for both male and female)? Please attach the openScale log as well. I'd like to see how things work now.

Then we just need a picture of the scale as well.

hyakki-, 2018-11-21:
195 male
[openScale_2018-11-21_01-01.txt](https://github.com/oliexdev/openScale/files/2603495/openScale_2018-11-21_01-01.txt)

195 female
[openScale_2018-11-21_00-59.txt](https://github.com/oliexdev/openScale/files/2603494/openScale_2018-11-21_00-59.txt)

Visceral fat shows 1.00 for both

here is picture of scale

[IMG_20181121_010801455.zip](https://github.com/oliexdev/openScale/files/2603528/IMG_20181121_010801455.zip)

erijo, 2018-11-21:
Excellent. Guess we can close this one then?!

erijo, 2018-11-21:
If you have time it would be interesting to see a log where you've done a measurement but with one foot of the metal plates (so that it can't measure body fat etc.).

Also, can you see if the log message "Scale disconnecting" correlates to you stepping of the scale or the measurement being done? (Note the time on the phone and then compare it with the time in the log.)

hyakki-, 2018-11-21:
yea there seems to be a bug when using 1 foot or no feet on the metal plates
if only one foot on one side it seems to get caught in infinite loop displaying toast messages of different weights (even when off scale, waited a minute to but never seemed to finalize, closed the app).

if no feet on any metal plates it shows the weight in a toast message but doesn't fill it out.

the disconnect message "lost bluetooth connection" toast message appears after i step off the scale or continuing to stand on it when the LED display goes off (about 5-10 seconds after), seems correct.
2018-11-21 02:37:23.953 Debug [1] OpenScale: Disconnecting from bluetooth device
same time I logged on device.

(standing on left side metal plates one foot, toast message bug / loop ) 
[openScale_2018-11-21_02-32.txt](https://github.com/oliexdev/openScale/files/2603889/openScale_2018-11-21_02-32.txt)

(standing in middle)
[openScale_2018-11-21_02-34.txt](https://github.com/oliexdev/openScale/files/2603890/openScale_2018-11-21_02-34.txt)

(probably not important, using normal logging the disconnect time)
[openScale_2018-11-21_02-36.txt](https://github.com/oliexdev/openScale/files/2603891/openScale_2018-11-21_02-36.txt)

erijo, 2018-11-21:
Ok, thanks!

It seems like it worked as expected. The multiple toast messages needs to be fixed, but not right now.

Thanks for all the testing!

closed 2018-11-21 by erijo