oliexdev/openScale #353: Bluetooth unreliable for Medisana BS 444 since 1.8.3

hirnsushi, 2018-11-22:
**Describe the bug**
Bluetooth connects but the connection drops immediately. Sync only happens for about 20% of my connection attempts. 

**To Reproduce**
Steps to reproduce the behavior:
1. Activate Bluetooth through Bluetooth icon
2. Wait for connection to establish
3. Connection drops
4. Repeat till the sync happens

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): Yes

**Expected behavior**
Previous version had a success rate of at least 95%.

**Debug log**
[openScale_2018-11-22_09-50.txt](https://github.com/oliexdev/openScale/files/2609360/openScale_2018-11-22_09-50.txt)

erijo, 2018-11-23:
We fixed this problem earlier in a4a27d2, but it seems I reverted that in 9ef17a9 to fix another issue. I'll look into making it work in both cases.

erijo, 2018-11-24:
Added a special case for Medisana. Please test the latest version @hirnsushi and let me know how it goes.

closed 2018-11-24 by erijo

hirnsushi, 2018-11-25:
Works great :+1:  Thank you very much for the quick fix :smiley:

erijo, 2018-11-25:
Hopefully I don't break it again in the next version this time 😊

commit 559d1767 (2018-11-24): Stop scan before connecting to Medisana

Fixes #353 (hopefully)