oliexdev/openScale #354: Bug at sync between silvercrest sbf75 &sony xperia xz1 compact

ravezed, 2018-11-24:
**Describe the bug**
Application crashes when receiving data from silvercrest sbf75 scale

**To Reproduce**
Steps to reproduce the behavior:
1. Start app
2. Start scale by stepping on it
3. Measure begins, App connects to scale
4. Unexpected error

**Expected behavior**
The App should sync with the scale and begin receiving data

**Debug log**
Build version: 1.8.3 
Current date: 2018-11-24 09:05:32 
Device: Sony G8441 
 
Stack trace:  
java.lang.StringIndexOutOfBoundsException: length=4; regionStart=4; regionLength=3
	at java.lang.StringFactory.newStringFromBytes(StringFactory.java:71)
	at java.lang.StringFactory.newStringFromBytes(StringFactory.java:54)
	at com.health.openscale.core.bluetooth.BluetoothBeurerSanitas.onBluetoothDataChange(BluetoothBeurerSanitas.java:359)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$GattCallback$8.run(BluetoothCommunication.java:759)
	at android.os.Handler.handleCallback(Handler.java:789)
	at android.os.Handler.dispatchMessage(Handler.java:98)
	at android.os.Looper.loop(Looper.java:251)
	at android.app.ActivityThread.main(ActivityThread.java:6572)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:767)

erijo, 2018-11-24:
Please attach a log from when it happens (settings, about).

ravezed, 2018-11-24:
2018-11-24 10:55:17.661 Debug [2] AboutPreferences: Debug log enabled, openScale v1.8.3 (31), SDK 26, Sony G8441
2018-11-24 10:55:17.810 Debug [2] BluetoothCommunication: INIT STATE: 8
2018-11-24 10:55:17.941 Debug [2035] BluetoothCommunication$GattCallback: onCharacteristicChanged 0000ffe1-0000-1000-8000-00805f9b34fb: E7 F0 36 01
2018-11-24 10:55:17.942 Debug [2] BluetoothBeurerSanitas: Ack Get User Info Initials

erijo, 2018-11-24:
Please disconnect Bluetooth first by clicking on the Bluetooth icon, then enable logging and connect Bluetooth.

ravezed, 2018-11-24:
2018-11-24 11:55:34.565 Debug [2] AboutPreferences: Debug log enabled, openScale v1.8.3 (31), SDK 26, Sony G8441
2018-11-24 11:56:05.791 Debug [2] OpenScale: Trying to connect to bluetooth device [18:93:D7:5F:89:15] (SBF75)
2018-11-24 11:56:05.798 Debug [2] BluetoothCommunication: BT is enabled, state=12, scan mode=21, is not discovering
2018-11-24 11:56:05.803 Debug [2] BluetoothCommunication: Starting LE scan for device [18:93:D7:5F:89:15]
2018-11-24 11:56:07.357 Debug [2] BluetoothCommunication: Found LE device null [78:BD:BC:90:E2:CA]
2018-11-24 11:56:07.362 Debug [2] BluetoothCommunication: Found LE device null [9C:8C:6E:E7:EF:9B]
2018-11-24 11:56:07.515 Debug [2] BluetoothCommunication: Found LE device null [78:BD:BC:90:E2:CA]
2018-11-24 11:56:09.224 Debug [2] BluetoothCommunication: Found LE device null [9C:8C:6E:E7:EF:9B]
2018-11-24 11:56:09.242 Debug [2] BluetoothCommunication: Found LE device null [78:BD:BC:90:E2:CA]
2018-11-24 11:56:09.386 Debug [2] BluetoothCommunication: Found LE device null [9C:8C:6E:E7:EF:9B]
2018-11-24 11:56:09.395 Debug [2] BluetoothCommunication: Found LE device null [78:BD:BC:90:E2:CA]
2018-11-24 11:56:12.635 Debug [2] BluetoothCommunication: Found LE device null [9C:8C:6E:E7:EF:9B]
2018-11-24 11:56:12.789 Debug [2] BluetoothCommunication: Found LE device null [9C:8C:6E:E7:EF:9B]
2018-11-24 11:56:14.630 Debug [2] BluetoothCommunication: Found LE device null [9C:8C:6E:E7:EF:9B]
2018-11-24 11:56:14.642 Debug [2] BluetoothCommunication: Found LE device null [78:BD:BC:90:E2:CA]
2018-11-24 11:56:15.804 Debug [2] BluetoothCommunication: Device not found in LE scan, connecting directly
2018-11-24 11:56:15.804 Info [2] BluetoothCommunication: Connecting to [18:93:D7:5F:89:15] (driver: Sanitas SBF70/SilverCrest SBF75)
2018-11-24 11:56:18.038 Debug [2] BluetoothCommunication: Found LE device null [78:BD:BC:90:E2:CA]
2018-11-24 11:56:18.053 Debug [2] BluetoothCommunication: Found LE device null [9C:8C:6E:E7:EF:9B]
2018-11-24 11:56:18.187 Debug [2] BluetoothCommunication: Found LE device null [78:BD:BC:90:E2:CA]
2018-11-24 11:56:20.025 Debug [2] BluetoothCommunication: Found LE device null [78:BD:BC:90:E2:CA]
2018-11-24 11:56:20.088 Debug [2] BluetoothCommunication: Found LE device null [9C:8C:6E:E7:EF:9B]
2018-11-24 11:56:23.489 Debug [2] BluetoothCommunication: Found LE device null [9C:8C:6E:E7:EF:9B]
2018-11-24 11:56:23.586 Debug [2] BluetoothCommunication: Found LE device null [78:BD:BC:90:E2:CA]
2018-11-24 11:56:23.634 Debug [2] BluetoothCommunication: Found LE device null [9C:8C:6E:E7:EF:9B]
2018-11-24 11:56:25.277 Debug [2] BluetoothCommunication: Found LE device null [78:BD:BC:90:E2:CA]
2018-11-24 11:56:25.318 Debug [2] BluetoothCommunication: Found LE device null [9C:8C:6E:E7:EF:9B]
2018-11-24 11:56:25.427 Debug [2] BluetoothCommunication: Found LE device null [78:BD:BC:90:E2:CA]
2018-11-24 11:56:25.489 Debug [2] BluetoothCommunication: Found LE device null [9C:8C:6E:E7:EF:9B]
2018-11-24 11:56:28.747 Debug [2] BluetoothCommunication: Found LE device null [9C:8C:6E:E7:EF:9B]
2018-11-24 11:56:28.826 Debug [2] BluetoothCommunication: Found LE device null [78:BD:BC:90:E2:CA]
2018-11-24 11:56:30.675 Debug [2] BluetoothCommunication: Found LE device null [78:BD:BC:90:E2:CA]
2018-11-24 11:56:30.764 Debug [2] BluetoothCommunication: Found LE device null [9C:8C:6E:E7:EF:9B]
2018-11-24 11:56:31.277 Debug [2186] BluetoothCommunication$GattCallback: onConnectionStateChange: status=0, newState=2
2018-11-24 11:56:31.290 Debug [2] BluetoothCommunication: Stopping LE scan
2018-11-24 11:56:31.306 Debug [2] MainActivity: Bluetooth connection successful established
2018-11-24 11:56:32.449 Debug [2186] BluetoothCommunication$GattCallback: onServicesDiscovered: status=0 (5 services)
2018-11-24 11:56:33.453 Debug [2] BluetoothCommunication: INIT STATE: 0
2018-11-24 11:56:33.454 Debug [2] BluetoothCommunication: Set notification on for 0000ffe1-0000-1000-8000-00805f9b34fb
2018-11-24 11:56:33.457 Debug [2] BluetoothCommunication: Write descriptor 00002902-0000-1000-8000-00805f9b34fb: 01 00 (queue: 0 0)
2018-11-24 11:56:33.585 Debug [2] BluetoothCommunication: INIT STATE: 1
2018-11-24 11:56:33.593 Debug [2] BluetoothCommunication: Write characteristic 0000ffe1-0000-1000-8000-00805f9b34fb: E6 01 (type: 2; queue: 0 0)
2018-11-24 11:56:33.674 Debug [2186] BluetoothCommunication$GattCallback: onCharacteristicChanged 0000ffe1-0000-1000-8000-00805f9b34fb: E6 00 20
2018-11-24 11:56:33.678 Debug [2] BluetoothBeurerSanitas: ACK Scale is ready
2018-11-24 11:56:33.680 Debug [2] BluetoothCommunication: INIT STATE: 2
2018-11-24 11:56:33.730 Debug [2] BluetoothCommunication: INIT STATE: 3
2018-11-24 11:56:33.732 Debug [2] BluetoothBeurerSanitas: Write new Date/Time: 1543056993 (5B F9 2E 61)
2018-11-24 11:56:33.736 Debug [2] BluetoothCommunication: Write characteristic 0000ffe1-0000-1000-8000-00805f9b34fb: E9 5B F9 2E 61 (type: 2; queue: 0 0)
2018-11-24 11:56:33.877 Debug [2] BluetoothCommunication: INIT STATE: 4
2018-11-24 11:56:33.883 Debug [2] BluetoothBeurerSanitas: Setting unit kg
2018-11-24 11:56:33.887 Debug [2] BluetoothCommunication: Write characteristic 0000ffe1-0000-1000-8000-00805f9b34fb: E7 4D 01 (type: 2; queue: 0 0)
2018-11-24 11:56:34.010 Debug [2] BluetoothCommunication: INIT STATE: 5
2018-11-24 11:56:34.333 Debug [2186] BluetoothCommunication$GattCallback: onCharacteristicChanged 0000ffe1-0000-1000-8000-00805f9b34fb: E7 F0 4D 00
[... cut ...]
2018-11-24 11:56:34.820 Debug [2] BluetoothCommunication: Write characteristic 0000ffe1-0000-1000-8000-00805f9b34fb: E7 36 00 00 00 00 00 00 00 4C (type: 2; queue: 0 0)
2018-11-24 11:56:34.824 Debug [2] BluetoothBeurerSanitas: scaleuserid: 76, registered users: 1, extracted users: 1
2018-11-24 11:56:34.964 Debug [2] BluetoothCommunication: INIT STATE: 8
2018-11-24 11:56:34.997 Debug [2186] BluetoothCommunication$GattCallback: onCharacteristicChanged 0000ffe1-0000-1000-8000-00805f9b34fb: E7 F0 36 01
2018-11-24 11:56:34.998 Debug [2] BluetoothBeurerSanitas: Ack Get User Info Initials

erijo, 2018-11-24:
I think the problem here is that openScale only uses the last byte in the user ID, but you seem to have an id that requires two bytes. I'm in the process of refactoring that code a bit, but it will take some time before it's done. If you don't want to wait for that I'd recommend that you reset your scale. Then you'll get a user ID.

erijo, 2018-11-25:
If you haven't reset your scale you can try [this dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev_beurer-build) where it should hopefully work.

erijo, 2018-12-04:
@ravezed: This should be fixed on master now (try the [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) to verify).

closed 2018-12-04 by erijo