oliexdev/openScale #357: SweetBlue Android Bluetooth library

oliexdev, 2018-11-27:
With all the Bluetooth hacks and problems around Android, maybe we should consider using a Bluetooth library like [SweetBlue](https://github.com/iDevicesInc/SweetBlue)

I will create a experimental branch for the experimental development but does anybody know a better open-source library?

erijo, 2018-11-27:
I've thought about sweetblue as well, but as it will no longer be maintained I'm not sure it's the best package to use.

oliexdev, 2018-11-27:
Yes, I share your concerns but currently I didn't found a better library which solves (hopefully) our ongoing problems with the Android Bluetooth stack. The reconnection loops and continuous scanning mode are nice features of SweetBlue which I didn't see in other libraries!? But on the long run this library will be dead at least for the open-source community. :(

oliexdev, 2018-11-27:
I watched the [talk ](https://www.youtube.com/watch?v=0aKfUGCxUDM) about RxAndroidBLE which seems they have also some kind of fallback mechanism and reconnection loops.
Since it is still maintained as an open-source project I guess it is better to try this out. The drawback is that they are using RxJava which I have never used before. Have you some experiences of RxJava @erijo ?

erijo, 2018-11-27:
No, I've never heard of RxJava, but then I'm generally no Java person so that doesn't say much ;)

closed 2018-12-21 by oliexdev

gmarco, 2019-02-10:
i updated now the new BT Library on my development phone. 
Connecting is now without "hack" fast.
it receives also the first weight fast BUT it sends data for next Measurements (fat...) but the Scale will not get it (sometimes never, sometimes after a looooong time)
i dont knwo what the problem is, (only thing i saw is that WEIGHT_MEASUREMENT_SERVICE for Senssuns Scales is not needed anymore to get the data ??

(logs from before and after is present, app is sending correct data, but the scale seems not to receive it)

oliexdev, 2019-02-10:
@gmarco please try out the latest dev build version

gmarco, 2019-02-10:
it was compiled from git master

oliexdev, 2019-02-10:
I don't know what you mean?

gmarco, 2019-02-10:
tested version was checkout from git/master and installed after compile
i tried now all available versions
1.9 and 1.9.1 works well (fast fat scanning).
with 1.9.2 and later it takes very long or will not get more data as only the weight.

so the cause for the wait must be within 1.9.2

oliexdev, 2019-02-10:
please attach the log file by openScale (settings->about)

gmarco, 2019-02-10:
log can be attached it that helps 
[openScale_2019-02-10_16-52.txt](https://github.com/oliexdev/openScale/files/2849173/openScale_2019-02-10_16-52.txt)

i looked for the change between 1.9.1 and 1.9.2

removing the .delay(... ) from BluetoothCommunication.java which was added there, let it waor as before fast and reliable

oliexdev, 2019-02-10:
you are using the wrong version. Please use the [latest dev build](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) and post again the log file!

gmarco, 2019-02-10:
the latest dev build is now working (with your patch made) :)
but now it will save the data only if all data where recorded.
before there was the fallback, that if only the weight was measured and no fat data, the weight was saved on disconnect (for that was overriding the disconnect method).

now is will not save the weigth, if more cannot be measured.

oliexdev, 2019-02-10:
could you send a PR without overwriting the disconnect method?

gmarco, 2019-02-10:
i tried. i hope in the cleanupCmd is the ridght place.
can you tell why it is not working https://github.com/gmarco/openScale/blob/new_try/android_app/app/src/main/java/com/health/openscale/core/bluetooth/BluetoothSenssun.java

the scale is now Saving on every Data got from scacle (even with and without fat)
the savedUserData variable seems not to prevent that.

gmarco, 2019-02-10:
i changed now everythng (only wanted to support new features)

to avoid disconnect override i tried to use this, to save only 1 time:

    private void saveUserData(){
      if ( isBitSet(WeightFatMus,2)  && !savedUserData ) {
          addScaleData(measurement);
          WeightFatMus=0;
          savedUserData = true;
      }
    }
but is is saving every second (not matter w/o fat). 1 weight measurement -> tens of measures

oliexdev, 2019-02-10:
attach the log file