oliexdev/openScale #360: BoneMass value 1byone is incorrect

Dododappere, 2018-12-01:
I received a 1byone scale and have been playing around with Openscale and comparing numbers with the 1byone app.

All body composition values are correct except the **Bone Mass** value. As a result the MuscleMass value is also wrong but if I manually insert the correct bonemass value in the muscleMass calculation, the correct number comes out so it is correct.

In order to compare the numbers I yanked out the library from the 1byone executable  so I could run them side by side.

Really amazing how you managed to get these formulas correct from the assembly code! My compliments and I take my hat off!

Regarding solving this bug I have some thoughts:
- You use the 'impedanceValue' in the bonemass function (scaleBtData.setBone(oneByoneLib.getBoneMass(weight, impedanceValue));) but I think this is wrong. You calculated the impedanceValue by adding the values of 3 bytes but I don't believe this is correct and I also don't see this being done in the 1byoneapp.
- In the 1byone app I see that they use the BMR as the parameter in the bonemass function (this.boneKg = new Bone(peopleModel).getValue(this.BMR);)
- However, manually inserting my BMR value also doesn't lead to the right number so there must still be something else wrong...

Here is a comparison I did where I used my BMR (1764) for the boneMass function. The bonemass value is a bit wrong:

Got data <cfbc116c250803ac00008c> from <0000fff4-0000-1000-8000-00805f9b34fb>
Weight 95.8, impedance 4540, unit 0

OPENSCALE: Bodyfat 27.71
OPENSCALE: BodyWater 49.59
OPENSCALE: BoneMass **2.98**
OPENSCALE: VisceralFat 13.74
OPENSCALE: MuscleMassPercentage 66.28
OPENSCALE: BMI 26.5
OPENSCALE: BMR 1917.5

HOLTEK: Bodyfat 27.71
HOLTEK: BodyWater 49.59
HOLTEK: BoneMass 3.44
HOLTEK: VisceralFat 13
HOLTEK: MuscleMassPercentage 65.82
HOLTEK: BMI 26.5
HOLTEK: BMR 1764

oliexdev, 2018-12-02:
no the BMR is not used for the calculation.The parameters mismatch with the internal calculation.
Can you play a bit with the activity level in openScale? Maybe that is wrong!?
Please note that the age, activity level, height, weight, sex, impedance value are used for the calculation. Are those parameters the same as in the original app? Please also attach the log file while your weighting (you can enable the log in openScale under the Settings->About)

Dododappere, 2018-12-02:
I did some more digging and I think I solved it...

In BluetoothOneByone.java you now say:
int impedanceValue = weightBytes[5] + weightBytes[6] + weightBytes[7];

..but if you change that to:
int impedanceValue = (int) ((((weightBytes[2] & 0xFF) << 8) + (weightBytes[1] & 0xFF)) * 0.1f);

..then BoneMass and MuscleMass are totally correct up to the second digit! Those 2 bytes also appear in a notification that the scale sends for the BodyCompositionMeasurement characteristics and it appears exactly on the spot of the impedance value. So I assume those two bytes are the impedance in Ohms (after dividing by 10).

Will experiment a bit more but it looks like this solves the issue!

Dododappere, 2018-12-02:
Turns out I was cheering too quickly...I asked my wife to stand on the scale and the numbers were wrong again. The bodyFat value goes through the roof and then the rest is also messed up.

My Wife (45 years, 175, skinny, activity level = 0)
Got data <cfbe1430169696870000c4> from <0000fff4-0000-1000-8000-00805f9b34fb>
Weight 56.8, impedance 531, unit 0

OPENSCALE: Bodyfat 45.00
OPENSCALE: BodyWater 39.27
OPENSCALE: BoneMass 2.45
OPENSCALE: VisceralFat 3.00
OPENSCALE: MuscleMassPercentage 28.79

HOLTEK: Bodyfat 22.31
HOLTEK: BodyWater 53.29
HOLTEK: BoneMass 2.45
HOLTEK: VisceralFat 3
HOLTEK: MuscleMassPercentage 41.68

Got data <cfc814f415b3b6ff000008> from <0000fff4-0000-1000-8000-00805f9b34fb>
Weight 56.2, impedance 532, unit 0

OPENSCALE: Bodyfat 45.00
OPENSCALE: BodyWater 39.27
OPENSCALE: BoneMass 2.44
OPENSCALE: VisceralFat 3.00
OPENSCALE: MuscleMassPercentage 28.47

HOLTEK: Bodyfat 21.85
HOLTEK: BodyWater 53.61
HOLTEK: BoneMass 2.44
HOLTEK: VisceralFat 3
HOLTEK: MuscleMassPercentage 41.49

My wife again but now saying that she is a male

Got data <cffa14e0156d65ec000030cffa14e0156d65ec00> from <0000fff4-0000-1000-8000-00805f9b34fb>
Weight 56.0, impedance 537, unit 0

OPENSCALE: Bodyfat 45.00
OPENSCALE: BodyWater 39.27
OPENSCALE: BoneMass 2.50
OPENSCALE: VisceralFat 4.00
OPENSCALE: MuscleMassPercentage 28.30

HOLTEK: Bodyfat 10.76
HOLTEK: BodyWater 61.22
HOLTEK: BoneMass 2.50
HOLTEK: VisceralFat 4
HOLTEK: MuscleMassPercentage 47.48

erijo, 2018-12-05:
I don't understand how you can get any data at all with openScale. Can you please post an openScale log when measuring using the [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build)?

erijo, 2018-12-05:
Please try [this version](https://github.com/oliexdev/openScale/releases/tag/travis-dev_1byone-build) instead @Dododappere.

Dododappere, 2018-12-09:
@erijo I will try your version but I don't understand what difference that will make. I am not using OpenScale to read the data...I used my own app to connect to the 1byone scale and I simply added both your calculations and the original one (with the libBodyfat.so library) from the 1byoneapp so that I could compare them easily...I just wanted to doublecheck if the formula was correct.

When I have time in the coming days I'll create a simple app with both calculations so it is easy to compare and you can try it yourself....

oliexdev, 2019-01-29:
Is this still a problem? I tested it with the data provided in issue #159 

Did you find a mistake in the formula? For females it was not quite tested because I didn't have any female data!?

closed 2019-03-09 by oliexdev

lbdroid, 2021-06-17:
@Dododappere ; in case you're still monitoring this issue, I believe that you were correct about the 2 bytes used for impedance value. The issue you were experiencing for female measurements is due to error here:
https://github.com/oliexdev/openScale/blob/master/android_app/app/src/main/java/com/health/openscale/core/bluetooth/lib/OneByoneLib.java#L251

That (along with other adjustments to bodyVar in that block) should read as ```bodyVar *= xx.xxf;```
I.e., multiply by, rather than obliterate.

The prior use of 3 bytes at offset 5 was, I believe, entirely bogus. Those 3 bytes weren't even being used consistently, in one place, they were summed together, and in another place, a complex formula that made no sense;
https://github.com/oliexdev/openScale/blob/master/android_app/app/src/main/java/com/health/openscale/core/bluetooth/lib/OneByoneLib.java#L183

I've also observed that the values of those bytes swing wildly between measurements, even if the measurements are only 5 seconds apart on the same person. So I don't know what those 3 bytes are, but they're not impedance. The 2 bytes you found for impedance, however, provide good consistency between measurements, and impedance values within a range that makes sense.